espressif/arduino-esp32 #9768: Exception on BLERemoteService::getCharacteristic

bobemoe, 2024-06-03:
### Board

ESP32 DevKitC V4

### Device Description

ESP32 DevKitC V4

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

arduino-cli

### Operating System

Linux Devuan

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

`arduino-cli compile --fqbn "esp32:esp32:esp32" --board-options "DebugLevel=verbose" --warnings "all" --verbose --clean -e`

```
0x400decd7: String::isSSO() const at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:344
 (inlined by) String::buffer() const at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:378
 (inlined by) String::compareTo(String const&) const at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.cpp:526
0x400ded99: String::operator<(String const&) const at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.cpp:553
0x400d8619: std::less<String>::operator()(String const&, String const&) const at /home/bob/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_function.h:408
 (inlined by) std::_Rb_tree<String, std::pair<String const, BLERemoteDescriptor*>, std::_Select1st<std::pair<String const, BLERemoteDescriptor*> >, std::less<String>, std::allocator<std::pair<String const, BLERemoteDescriptor*> > >::_M_lower_bound(std::_Rb_tree_node<std::pair<String const, BLERemoteDescriptor*> >*, std::_Rb_tree_node_base*, String const&) at /home/bob/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_tree.h:1951
 (inlined by) std::_Rb_tree<String, std::pair<String const, BLERemoteDescriptor*>, std::_Select1st<std::pair<String const, BLERemoteDescriptor*> >, std::less<String>, std::allocator<std::pair<String const, BLERemoteDescriptor*> > >::lower_bound(String const&) at /home/bob/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_tree.h:1270
 (inlined by) std::map<String, BLERemoteDescriptor*, std::less<String>, std::allocator<std::pair<String const, BLERemoteDescriptor*> > >::lower_bound(String const&) at /home/bob/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_map.h:1307
 (inlined by) std::enable_if<std::is_constructible<std::pair<String const, BLERemoteDescriptor*>, std::pair<String, BLERemoteDescriptor*> >::value, std::pair<std::_Rb_tree_iterator<std::pair<String const, BLERemoteDescriptor*> >, bool> >::type std::map<String, BLERemoteDescriptor*, std::less<String>, std::allocator<std::pair<String const, BLERemoteDescriptor*> > >::insert<std::pair<String, BLERemoteDescriptor*> >(std::pair<String, BLERemoteDescriptor*>&&) at /home/bob/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_map.h:855
 (inlined by) BLERemoteCharacteristic::retrieveDescriptors() at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/BLE/src/BLERemoteCharacteristic.cpp:300
0x400d87f9: BLERemoteCharacteristic::BLERemoteCharacteristic(unsigned short, BLEUUID, unsigned char, BLERemoteService*) at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/BLE/src/BLERemoteCharacteristic.cpp:43
0x400d958a: BLERemoteService::retrieveCharacteristics() at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/BLE/src/BLERemoteService.cpp:180 (discriminator 3)
0x400d96b6: BLERemoteService::getCharacteristic(BLEUUID) at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/BLE/src/BLERemoteService.cpp:135
0x400d2aa9: connectToServer() at /home/bob/Projects/ESPBMS/ESPBMS.ino:64
0x400d2cba: loop() at /home/bob/Projects/ESPBMS/ESPBMS.ino:133
0x400e09e4: loopTask(void*) at /home/bob/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/main.cpp:73

```

### Sketch

```cpp
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

[... 450 lines cut ...]
[  4714][D][BLERemoteCharacteristic.cpp:295] retrieveDescriptors(): Found a descriptor: Handle: 18, UUID: 00002902-0000-1000-8000-00805f9b34fb
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400decda  PS      : 0x00060430  A0      : 0x800ded9c  A1      : 0x3ffc8410  
A2      : 0xa5a5a5b5  A3      : 0x3ffc84b4  A4      : 0x0000002f  A5      : 0x3ffbd5b0  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x0000007c  A9      : 0x3ffc83f0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0000001c  A13     : 0x00000024  
A14     : 0x3ffdeb68  A15     : 0xff000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa5a5a5c4  LBEG    : 0x40090e14  LEND    : 0x40090e1f  LCOUNT  : 0x00000000  

Backtrace: 0x400decd7:0x3ffc8410 0x400ded99:0x3ffc8430 0x400d8619:0x3ffc8450 0x400d87f9:0x3ffc8500 0x400d958a:0x3ffc8570 0x400d96b6:0x3ffc8620 0x400d2aa9:0x3ffc8680 0x400d2cba:0x3ffc86c0 0x400e09e4:0x3ffc8720
```

### Other Steps to Reproduce

The sketch above is the exact example provided in the 3.0.0 release repo, except the UUIDs changed to match my BLE device.

This was working on 2.x

I am no longer using this project but just copying the details from the now closed [3.0.0 version Migration related issues](https://github.com/espressif/arduino-esp32/issues/8796) as it was never resolved there.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-07:
@bobemoe - I have teste Arduino Core 3.0.0 with ESP32 using the BLE Client + BLE Server example. It works fine.
I'm not able to reprodue this issue as I have no `xiaoxiang bms` device here.

It seems related to the way the BMS BLE server provides the information of the characteristic.
ESP32 is able to connect, retrieve the list of characteristics, but when it tries to read the descriptor, it crashes.

It seems to fail when it tries to create the String for the Descriptor data.
My guess is that it could be some Heap Allocation failure... Memory Fault?!

SuGlider, 2024-06-07:
The difference I see from the BLE Client/Server example is that it has no Descriptors... just Characteristics.
```
[  3619][V][BLERemoteCharacteristic.cpp:268] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: beb5483e-36e1-4688-b7f5-ea07361b26a8
[  3632][E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: ESP_GATT_NOT_FOUND
[  3642][V][BLERemoteCharacteristic.cpp:305] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
```

While the BMS has some descriptor, which causes the crash.
```
[  4701][V][BLERemoteCharacteristic.cpp:268] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 0000ff01-0000-1000-8000-00805f9b34fb
[  4714][D][BLERemoteCharacteristic.cpp:295] retrieveDescriptors(): Found a descriptor: Handle: 18, UUID: 00002902-0000-1000-8000-00805f9b34fb
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

```

SuGlider, 2024-06-07:
I'll try to emulate such configuration and test it. I'll keep you posted.

bobemoe, 2024-06-08:
Would it help for me to capture a debug log from a successful connection to the same BMS on the working 2.x branch?

SuGlider, 2024-06-08:
> Would it help for me to capture a debug log from a successful connection to the same BMS on the working 2.x branch?

Yes, please. This way I will be able to see all the Characteristics and Descriptors of the BMS.

bobemoe, 2024-06-09:
@SuGlider Here it is:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[    23][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Starting Arduino BLE Client application...
[   730][V][BLEScan.cpp:385] start(): >> start(duration=5)
[   731][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffdc658), owner: <N/A> for start
[   734][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffdc658), owner: start
[   743][V][BLEScan.cpp:417] start(): << start()
[   747][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ScanEnd (0x3ffdc658), owner: start for start
[   745][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT
[   767][V][BLEUtils.cpp:1156] dumpGapEvent(): [status: 0]
[   772][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
[   781][V][BLEUtils.cpp:1221] dumpGapEvent(): [status: 0]
[   786][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[   795][V][BLEUtils.cpp:1189] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 70:3e:97:d2:b6:1e, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -87, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 16, scan_rsp_len: 11
[   826][D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -87
[   832][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06
[   843][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x02 (ESP_BLE_AD_TYPE_16SRV_PART), length: 2, data: 00ff
[   854][D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000ff00-0000-1000-8000-00805f9b34fb
[   865][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 7, data: 424d5320312e33
[   877][D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: BMS 1.3
[   884][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0xff (ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE), length: 6, data: 703e97d2b61e
[   897][D][BLEAdvertisedDevice.cpp:449] setManufacturerData(): - manufacturer data: 703e97d2b61e
[   905][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (ESP_BLE_AD_TYPE_TX_PWR), length: 1, data: 3f
[   915][D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 63
BLE Advertised Device found: Name: BMS 1.3, Address: 70:3e:97:d2:b6:1e, manufacturer data: 703e97d2b61e, serviceUUID: 0000ff00-0000-1000-8000-00805f9b34fb, txPower: 63, rssi: -87
[   943][V][BLEScan.cpp:440] stop(): >> stop()
[   943][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ScanEnd (0x3ffdc658), owner: start
[   950][V][BLEScan.cpp:452] stop(): << stop()
[   950][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: ScanEnd (0x3ffdc658), owner: <N/A>
[   954][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
Forming a connection to [   972][V][BLEUtils.cpp:1189] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 70:3e:97:d2:b0:df, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -84, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 16, scan_rsp_len: 11
70:3e:97:d2:b6:1e
[  1005][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[  1015][V][BLEDevice.cpp:60] createClient(): >> createClient
[  1015][V][BLEUtils.cpp:1189] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 70:3e:97:d2:b1:75, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -86, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 16, scan_rsp_len: 11
[  1021][V][BLEDevice.cpp:66] createClient(): << createClient
 - Created client
[  1052][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
[  1068][V][BLEClient.cpp:98] connect(): >> connect(70:3e:97:d2:b6:1e)
[  1068][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[  1072][I][BLEDevice.cpp:622] addPeerDevice(): add conn_id: 0, GATT role: client
[  1081][V][BLEUtils.cpp:1189] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 70:3e:97:d2:af:43, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -80, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 16, scan_rsp_len: 11
[  1088][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3ffdca98), owner: <N/A> for connect
[  1119][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
[  1129][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3ffdca98), owner: connect
[  1137][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
[... cut ...]
[  5389][V][BLEUtils.cpp:1506] dumpGattClientEvent(): [status: ESP_GATT_WRITE_NOT_PERMIT, conn_id: 0, handle: 17 0x11, offset: 0]
[  5401][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_WRITE_CHAR_EVT
[  5412][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: WriteCharEvt (0x3ffdf800), owner: writeValue
[  5421][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: WriteCharEvt (0x3ffdf800), owner: <N/A>
[  5431][V][BLERemoteCharacteristic.cpp:604] writeValue(): << writeValue
Setting new characteristic value to "Time since boot: 6"

```

SuGlider, 2024-06-14:
Issue confirmed. I can reproduce it using another ESP32 as Server with 0x2902 and 0x2901 for a characteristic.

SuGlider, 2024-06-14:
Issue Log Output:
```
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ESP32, Address: 40:4c:ca:50:20:72, serviceUUID: 0000ff00-0000-1000-8000-00805f9b34fb, txPower: 9, rssi: -42
Forming a connection to 40:4c:ca:50:20:72
 - Created client
 - Connected to server
 - Found our service

Stack smashing protect failure!
```

Apparently some problem with Array space.

SuGlider, 2024-06-16:
This is a C++ issue. BLE classes were changed to use Arduino `String` class instead of `std::string`.
This change was done every place with no distinction about how it was used.
There is a side effect when using `std::map`  template, which demands 4 specific classes.
But Arduino `String` is not compatible with it, therefore it crashes internally in execution time.

template<
    class Key,
    class T,
    class Compare = [std::less](http://en.cppreference.com/w/cpp/utility/functional/less)<Key>,
    class Allocator = [std::allocator](http://en.cppreference.com/w/cpp/memory/allocator)<[std::pair](http://en.cppreference.com/w/cpp/utility/pair)<const Key, T>>
\> [class map](https://en.cppreference.com/w/cpp/container/map);

The fix demands a deep review of all the changes done in https://github.com/espressif/arduino-esp32/pull/8724
All `std::map` must be used as `std::map<std:string, T*>` instead of `std::map<String, T*>`

SuGlider, 2024-06-16:
The issue affects several cases: 
- BLE servers with more than 1 characteristic per Service 
- BLE servers with more than 1 descriptor per Characteristic
- BLE servers with more than 1 service per device
- BLE Scan device list  (erase, insert)

SuGlider, 2024-06-17:
@bobemoe - Please test the fix from PR #9875 within Arduino Core 3.0.0+ 
This shall fix the issue you have repoted here. Thanks!

closed 2024-06-17 by me-no-dev