espressif/arduino-esp32 #9771: A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x01): Possible serial noise or corruption. For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html Failed uploading: uploading error: exit status 2

Shinjan-saha, 2024-06-04:
### Board

ESP32 Cam 

### Device Description

I have connected the esp32 Cam with Arduino UNO R3  7 want to upload an code but i am getting an error "A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x01): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2"

### Hardware Configuration

camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.frame_size = FRAMESIZE_UXGA;
  config.pixel_format = PIXFORMAT_JPEG; // for streaming
  //config.pixel_format = PIXFORMAT_RGB565; // for face detection/recognition
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 12;
  config.fb_count = 1;

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x01): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2

Want to resolve this errror get my ESP 32C Am running

### Sketch

```cpp
#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
//            Ensure ESP32 Wrover Module or other board with PSRAM is selected
//            Partial images will be transmitted if image exceeds buffer size
//
//            You must select partition scheme from the board menu that has at least 3MB APP space.
//            Face Recognition is DISABLED for ESP32 and ESP32-S2, because it takes up from 15 
//            seconds to process single frame. Face Detection is ENABLED if PSRAM is enabled as well

// ===================
// Select camera model
// ===================
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
//#define CAMERA_MODEL_XIAO_ESP32S3 // Has PSRAM
// ** Espressif Internal Boards **
//#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_ESP32S2_CAM_BOARD
//#define CAMERA_MODEL_ESP32S3_CAM_LCD
//#define CAMERA_MODEL_DFRobot_FireBeetle2_ESP32S3 // Has PSRAM
//#define CAMERA_MODEL_DFRobot_Romeo_ESP32S3 // Has PSRAM
#include "camera_pins.h"

// ===========================
// Enter your WiFi credentials
// ===========================
const char* ssid = "saythanks";
const char* password = "nunz1234";

void startCameraServer();
void setupLedFlash(int pin);

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.frame_size = FRAMESIZE_UXGA;
  config.pixel_format = PIXFORMAT_JPEG; // for streaming
  //config.pixel_format = PIXFORMAT_RGB565; // for face detection/recognition
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 12;
  config.fb_count = 1;
  
  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if(config.pixel_format == PIXFORMAT_JPEG){
    if(psramFound()){
      config.jpeg_quality = 10;
      config.fb_count = 2;
      config.grab_mode = CAMERA_GRAB_LATEST;
    } else {
      // Limit the frame size when PSRAM is not available
      config.frame_size = FRAMESIZE_SVGA;
      config.fb_location = CAMERA_FB_IN_DRAM;
    }
  } else {
    // Best option for face detection/recognition
    config.frame_size = FRAMESIZE_240X240;
#if CONFIG_IDF_TARGET_ESP32S3
    config.fb_count = 2;
#endif
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  // initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1); // flip it back
    s->set_brightness(s, 1); // up the brightness just a bit
    s->set_saturation(s, -2); // lower the saturation
  }
  // drop down frame size for higher initial frame rate
  if(config.pixel_format == PIXFORMAT_JPEG){
    s->set_framesize(s, FRAMESIZE_QVGA);
  }

#if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

#if defined(CAMERA_MODEL_ESP32S3_EYE)
  s->set_vflip(s, 1);
#endif

// Setup LED FLash if LED pin is defined in camera_pins.h
#if defined(LED_GPIO_NUM)
  setupLedFlash(LED_GPIO_NUM);
#endif

  WiFi.begin(ssid, password);
  WiFi.setSleep(false);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // Do nothing. Everything is done in another task by the web server
  delay(10000);
}
```

### Debug Message

```plain
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x01): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-04:
>I have connected the esp32 Cam with Arduino UNO R3 7 want to upload an code but i am getting an error "A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x01): Possible serial noise or corruption.

How exactly are both connected?
ESP32 UART0 port is used to Upload the firmware, using  U0RX (GPIO 3) / U0TX (GPIO 1) pins of the ESP32 Cam board.

This error seems to be related to some problem with PC USB Serial <---> CP2102 | CH430 UART-USB Chip <--> ESP32 UART0.
It can be the cable, or something in the circuit / board.

If you disconnect the UNO R3, does the sketch upload to the ESP32?

Shinjan-saha, 2024-06-04:
> How exactly are both connected? ESP32 UART0 port is used to Upload the firmware, using U0RX (GPIO 3) / U0TX (GPIO 1) pins of the ESP32 Cam board.
>
> This error seems to be related to some problem with [...]

I am using the UNOR3 to upload the code in ESP32 Cam as i don't have USB to TTL  chip  so how can i upload the code if i disconnect UNO R3 with ESP 32 Cam

SuGlider, 2024-06-04:
>I am using the UNOR3 to upload the code in ESP32 Cam 

I've never tried using an UNO for that...
I've found a video about it: https://www.youtube.com/watch?v=2QLYsIBNvB0
https://iotstarters.com/esp32-cam-code-upload-using-arduino-uno/

**UNO shall have its RESET pin connected to GND** in order to do not run and do not send any noise to its own UART port...
ESP32 shall have GPIO 0 connected to GND in order to enter in Download Mode.

https://easyelectronicsproject.com/esp32-projects/program-esp32cam-arduino/
[image]

SuGlider, 2024-06-04:
@Shinjan-saha - If following the instructions doesn't work... you can try reducing the upload speed to 115200, using the IDE Board Menu.

Shinjan-saha, 2024-06-04:
> @Shinjan-saha - If following the instructions doesn't work... you can try reducing the upload speed to 115200, using the IDE Board Menu.

I have already done all this step & infact  I followed exactly the same yt video which you have referred to but still the error persist

SuGlider, 2024-06-04:
A last observation is that UNO is 5 volts and its UART will generate pulses with 5v amplitude.
ESP32 is 3.3 volts. Maybe using some resistor divider or level converter may be necessary.

[image]

SuGlider, 2024-06-04:
Open the serial monitor, reset the ESP32 and check if you can see the boot message on it. Use 115200 baud for the serial monitor.
If it doesn't display anything, the wires may be damaged, the ESP32 may be damaged or the wires are not correctly connected.

ESP32 TX will produce 3.3v pulses --- in order to the UNO be able to read it shall be converted to 5v level.
ESP32 RX expects 3.3v pulses -- UNO output shall be converted from 5v to 3.3v.

SuGlider, 2024-06-04:
@Shinjan-saha - Please read this... I hope that your ESP32 Cam isn't already demaged.
https://forum.arduino.cc/t/how-do-i-connect-my-esp-32-cam-with-arduino/1193406

lucasssvaz, 2024-06-04:
As @SuGlider suggested, one of these might do the job: https://www.sparkfun.com/products/12009

SuGlider, 2024-06-05:
This issue is related to HW more than to the Arduino Core implementation.
Closing this issue. Please fell free to reopen it, if necessary.

gerhil, 2025-04-20:
I had same problem. Hold the BOOT button (on the shield) and then press the RESET button (on the ESP32 board) and release the BOOT button. The EN button doesn't work. Now I have upload to the ESP32.