espressif/arduino-esp32 #9775: arduino-esp32 v3.0.0 not compatible with latest version of SSLClientESP32 library (v2.0.3)

bonamyr, 2024-06-04:
### Board

ESP32-WROVER-B

### Device Description

I am using the following development board: Lilygo T-SIM7000G.

### Hardware Configuration

I have detached all I2C devices from the Lilygo board and the problem persists.

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

There is a problem with the header file ssl_lib_client.h of the SSLClientESP32 library. It should #include mbedtls/net.h but doesn't find it. I have reduced the code to the bare minimum to show that the problem is related to the SSLClientESP32 library. The code works fine with the earlier version of arduino-esp32 (v2.0.17) and the latest version of SSLClientESP32 library (v.2.0.3), showing that the most recent arduino-esp32 version is not compatible with SSLClientESP32 library anymore.

### Sketch

```cpp
#include <SSLClientESP32.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
In file included from c:\Users\myria\Documents\Arduino\libraries\SSLClientESP32\src/SSLClientESP32.h:25,
                 from C:\Users\myria\AppData\Local\Temp\.arduinoIDE-unsaved202454-21456-hvzkm5.0tuzn\sketch_jun4a\sketch_jun4a.ino:1:
c:\Users\myria\Documents\Arduino\libraries\SSLClientESP32\src/ssl_lib_client.h:11:10: fatal error: mbedtls/net.h: No such file or directory
   11 | #include "mbedtls/net.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-04:
Please file an issue in the library repository. It needs to be updated to support Arduino 3.0.0

closed 2024-06-04 by me-no-dev

BlueAndi, 2024-10-04:
@me-no-dev You mean here? https://github.com/espressif/mbedtls
Because there its not possible to create an issue.

me-no-dev, 2024-10-04:
No. In SSLClientESP32