espressif/arduino-esp32 #9776: ESP IDF, Arduino as a component, doesn't compile (IDFGH-12940)

meakashrao, 2024-06-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2-666-ga5e3e0b90e

### Espressif SoC revision.

ESP32-WROOM-32E-N8

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

I expected that when using Arduino as a component, the Arduino core to compile, I had use one similar some 3 moths ago and it did compile. 

I had to make some changes, 

1) The Arduino Core I downloaded on 6th Feb 2024 did not have this line so I erased it after it threw the error. 
[image]

2) Same goes with this, earlier version does not have this and I had to comment this out. 
[image]

I also tried IDF Version 5.2.1, but it has some other error while downloading mdns as an component, so came back to idf version 5.1.2

### What is the actual behavior?

Even after making those above changes, there are some errors in .h files of the Arduino cores.

[IDF Bug.zip](https://github.com/user-attachments/files/15533415/IDF.Bug.zip)

### Steps to reproduce.

1. IDF Version 5.1.2, download Arduino as a Component
2. Create a component called as OTA
3. Use the below Files 

Its a .c file, for uploading here made .txt file
[OTA.txt](https://github.com/user-attachments/files/15533468/OTA.txt)
[CMakeLists.txt](https://github.com/user-attachments/files/15533426/CMakeLists.txt)

### Debug Logs.

```plain
-- Found Git: C:/Users/Akash/esp/Tools_Directory/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
[... 273 lines cut ...]
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:38:1: error: 
unknown type name 'class'
   38 | class IPAddress : public Printable {
      | ^~~~~
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:38:17: error:
 expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
   38 | class IPAddress : public Printable {
      |                 ^
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:136:14: error: unknown type name 'IPAddress'
  136 | extern const IPAddress IN6ADDR_ANY;
      |              ^~~~~~~~~
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:137:14: error: unknown type name 'IPAddress'
  137 | extern const IPAddress INADDR_NONE;
      |              ^~~~~~~~~
In file included from D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/libraries/WiFi/src/WiFiSTA.h:29,
                 from D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/libraries/WiFi/src/WiFi.h:33:
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/libraries/WiFi/src/WiFiGeneric.h:30:10: fatal error: functional: No such file or directory
   30 | #include <functional>
      |          ^~~~~~~~~~~~
compilation terminated.
[1635/1643] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/STA.cpp.obj       
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\cmd.exe /d /c ninja " terminated with exit code: 1.
```

everslick, 2024-06-04:
This seems to be a mismatch between Arduino Core 3.0.0 and the IDF version it comes with, then. I have the same two issues:
```
'ESP_PARTITION_SUBTYPE_DATA_LITTLEFS' was not declared in this scope
```
and
```
libraries/WiFi/src/WiFiGeneric.cpp:668:3: error: 'wifi_ftm_initiator_cfg_t' has no non-static data member named 'use_get_report_api'
```

The first one is easy to work around by defining `-DESP_PARTITION_SUBTYPE_DATA_LITTLEFS=0x83`
on the command line, but the second one requires patching the source.

I use makeEspArduino, if this makes any difference.

Jason2866, 2024-06-04:
@everslick which exact version or better commit is used for IDF 5.1?

Jason2866, 2024-06-04:
@meakashrao Your IDF version is to old. `ESP-IDF v5.1.2-666-ga5e3e0b90e`

everslick, 2024-06-05:
hmm, I never had to worry about this. I was under the impression, that a specific IDF version **comes** with a specific CORE version (i.e. they are version pinned to each other). Installing IDF was never a separate thing to do. CORE and IDF came as one.

@Jason2866 The contents of the file you posted above is the same as yours, btw.

`"version": "idf-release_v5.1-442a798083"`

maybe something is wrong with `get.py`?

me-no-dev, 2024-06-05:
That is true. Are you sure that when you update the core you also run get.py? You must have the wrong libs to get those errors (or you use as IDF component and have the wrong IDF version)

everslick, 2024-06-05:
I just realized that this might be exactly the issue. I DID NOT run get.py again after `git pull`ing from master...

```
System: Linux, Bits: 64, Info: Linux-6.5.0-28-generic-x86_64-with-glibc2.35
Platform: x86_64-pc-linux-gnu
Downloading 'esp32-arduino-libs-951ade74d7886e1ce931ea46614c4ac47ae3a6c0.zip' ...
Downloading... 194.94 MB downloaded - Elapsed Time: 02:33.53 - Speed: 1.27 MB/s
```

me-no-dev, 2024-06-05:
@everslick with previous versions, the libs used to come inside the core. We changed this for v3.x, so it's usually critical to run get.py when updating the core.

me-no-dev, 2024-06-05:
@meakashrao  can we close this issue?

everslick, 2024-06-05:
Thx, @me-no-dev ! Iwill update my build scripts to handle this!

BTW, good work on the 3.0.0 release, the migration was pretty straight forward and you managed to keep it pretty API compatible. I'm very impressed! :+1:

meakashrao, 2024-06-05:
> @meakashrao can we close this issue?

Hey I have just sometime started using IDF, I use IDF in VS Code, what do I need to do? 

idf.py --version  
ESP-IDF v5.1.2-666-ga5e3e0b90e

me-no-dev, 2024-06-05:
@meakashrao get ESP-IDF v5.1.4 instead (or pull branch `release/v5.1`) and use that

@everslick glad to hear! I hope the new APIs make more sense in Arduino also :)

meakashrao, 2024-06-05:
Sure. Thanks will do that

closed 2024-06-05 by meakashrao

everslick, 2024-06-05:
Confirming that running `get.py` fixes the problem for me. Issue can be closed from my side.