espressif/arduino-esp32 #9780: I want to read the files in the program into the firmware, is there a way?

hzhh110, 2024-06-05:
### Related area

ESP32

### Hardware specification

ESP32

### Is your feature request related to a problem?

I want to read the files in the program into the firmware, is there a way?

### Describe the solution you'd like

I want to read the files in the program into the firmware, is there a way?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

hzhh110, 2024-06-05:
Or can you only directly write the text or data in the file directly into a fixed code segment?

me-no-dev, 2024-06-05:
Hi @hzhh110 this issue tracker is for bugs in the core. Please search google or ask in the PlatformIO tracker.

closed 2024-06-05 by me-no-dev