espressif/arduino-esp32 #9782: Compilation fails for Arduino 3: fatal error: Network.h: No such file or directory

mathieucarbou, 2024-06-05:
### Board

esp32dev

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v3.0.1

### IDE Name

PlatformIO

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

Can be reproduced with Arduino 3.0.0 and 3.0.1

This happens only when these settings are set:

- `lib_ldf_mode = deep+`
- `lib_ldf_mode = chain+`

When `chain` or `deep` is used, compilation succeeds.

Was working with Arduino 2.x

Is it possible that the network layer refactoring impacts the compilation wen [C/C++ Preprocessor conditional syntax](https://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-mode). is processed ?

### Sketch

```cpp
PIO project setup with:

framework = arduino
platform = espressif32@6.7.0
platform_packages=
  platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.1
  platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.1/esp32-arduino-libs-3.0.1.zip
```
```

### Debug Message

```plain
Compilation error:

   39 | #include "Network.h"
      |          ^~~~~~~~~~~
compilation terminated.
In file included from /Users/mat/.platformio/packages/framework-arduinoespressif32@src-c4332d380c103569bb66c530e15d287f/libraries/WiFi/src/WiFiSTA.h:29,
                 from /Users/mat/.platformio/packages/framework-arduinoespressif32@src-c4332d380c103569bb66c530e15d287f/libraries/WiFi/src/WiFi.h:33,
                 from /Users/mat/.platformio/packages/framework-arduinoespressif32@src-c4332d380c103569bb66c530e15d287f/libraries/WiFi/src/WiFiGeneric.cpp:25:
/Users/mat/.platformio/packages/framework-arduinoespressif32@src-c4332d380c103569bb66c530e15d287f/libraries/WiFi/src/WiFiGeneric.h:39:10: fatal error: Network.h: No such file or directory
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-05:
some cache or else wrong on your end. 3.0.1 is fine.

closed 2024-06-05 by me-no-dev

mathieucarbou, 2024-06-05:
> some cache or else wrong on your end. 3.0.1 is fine.

Not a cache issue. I deleted my whole `~/.platformio` folder before running the test, and ran after with `pio run -e <env>`.

Like I said, it's linked to the `lib_ldf_mode` pio flag. 

It fails in 2 different projects the same way. as soon as we activate macro pre-prpcessing (`+`).

me-no-dev, 2024-06-05:
The Network library does not really have anything special. It's not much different than all other Arduino base classes (Print, Stream, Server, etc.). It works on all chips and the only preprocessor conditions are for WiFi support and events (H2 does not have WiFi). I can not speak of what PIO has done to support 3.x and if there are any substantial differences, but on our end, we do still generate all PIO required files. 

Why do you even need those `lib_ldf_mode` flags?

mathieucarbou, 2024-06-05:
> Why do you even need those `lib_ldf_mode` flags?

For the obvious reason cited in the pio doc. 

> In spite of the fact that Library Dependency Finder is written in pure Python, it evaluates [C/C++ Preprocessor conditional syntax](https://gcc.gnu.org/onlinedocs/cpp/Conditional-Syntax.html#Conditional-Syntax) (#ifdef, if, defined, else, and elif) without calling gcc -E

```c++
#ifdef MY_PROJECT_VERSION
// include common file for the project
#include "my_common.h"
#endif

#if MY_PROJECT_VERSION < 10
// this include will be ignored because does not satisfy condition above
#include "my_old.h"
#endif
```

But I agree with you that if the pre-processing is done through their python code, then it is up to them.

me-no-dev, 2024-06-05:
This does not seem to have anything to do with the preprocessor per say. It's looking for a file and it can not find it. Interesting why is that? It is true that the library is new, bot ArduinoOTA and other libs do depend on things in the core, so why?

me-no-dev, 2024-06-05:
So maybe it thinks that the target does not have WiFi? https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.h#L26

me-no-dev, 2024-06-05:
Maybe add `-D SOC_WIFI_SUPPORTED=1` to your build flags, but I am not sure if it will not cause any warnings for re-declaration

mathieucarbou, 2024-06-05:
Oh! i didn't pay attention to this define! Yes, maybe... 
The fat that it works fine in Arduino 2 and not in 3 was confusing to me and I thought maybe there is something to look at.
I will try, other wise switch to `chain` which is the default.

Thanks !!! :-)