espressif/arduino-esp32 #9786: Unable to upload code after update to 3.0.1 (invalid value '460800' for option 'UploadSpeed')

wangzongming, 2024-06-06:
### Board

nodemcu-32s

### Device Description

no

### Hardware Configuration

Uploading any code results in an error

### Version

latest master (checkout manually)

### IDE Name

Arduino IED

### Operating System

Window 11

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

default

### Description

Uploading the code will directly prompt the following error

I can't find a place on the menu bar where I can change the upload speed. Can you tell me how to change the speed?

It's similar to this problem：https://github.com/espressif/arduino-esp32/issues/8320

```
Invalid FQBN: getting build properties for board esp32:esp32:nodemcu-32s: invalid value '460800' for option 'UploadSpeed'

Compilation error: Invalid FQBN: getting build properties for board esp32:esp32:nodemcu-32s: invalid value '460800' for option 'UploadSpeed'
```

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Invalid FQBN: getting build properties for board esp32:esp32:nodemcu-32s: invalid value '460800' for option 'UploadSpeed'

Compilation error: Invalid FQBN: getting build properties for board esp32:esp32:nodemcu-32s: invalid value '460800' for option 'UploadSpeed'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-06:
460800 is marked as compatible only with Linux and Mac. Have you tried the other baudrates?

wangzongming, 2024-06-06:
> 460800 is marked as compatible only with Linux and Mac. Have you tried the other baudrates?

Yes, I don't see 460800 baud rate on the serial port.

I don't see an option to set the speed in the arduino IDE(v2.3.2).
[image]

I tried to modify the contents of this file
` \Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\boards.txt`
But nothing worked.
```
nodemcu-32s.menu.UploadSpeed.460800.linux=115200        <= this
nodemcu-32s.menu.UploadSpeed.460800.macosx=115200        <= this
nodemcu-32s.menu.UploadSpeed.460800.upload.speed=115200  <= this
nodemcu-32s.menu.UploadSpeed.115200=115200
nodemcu-32s.menu.UploadSpeed.115200.upload.speed=115200
nodemcu-32s.menu.UploadSpeed.256000.windows=256000
nodemcu-32s.menu.UploadSpeed.256000.upload.speed=256000
nodemcu-32s.menu.UploadSpeed.230400.windows.upload.speed=256000
nodemcu-32s.menu.UploadSpeed.230400=230400
nodemcu-32s.menu.UploadSpeed.230400.upload.speed=230400
nodemcu-32s.menu.UploadSpeed.512000.windows=512000
nodemcu-32s.menu.UploadSpeed.512000.upload.speed=512000
nodemcu-32s.menu.UploadSpeed.921600=921600
nodemcu-32s.menu.UploadSpeed.921600.upload.speed=921600

```

wangzongming, 2024-06-06:
I can upload the code after I fall back to version 2.0.17.

But I want to use ESP_SR, it seems that need 3.0 to have this library.

wangzongming, 2024-06-06:
Version 3.0.0 also does not report errors.

me-no-dev, 2024-06-06:
You should be able to see menu options there. Not sure why you do not. What happens if you select "ESP32 Dev Module"?

me-no-dev, 2024-06-06:
[image]

This is on Mac

wangzongming, 2024-06-06:
[image]

> You should be able to see menu options there. Not sure why you do not. What happens if you select "ESP32 Dev Module"?

There's no problem choosing this one. Thank you very much.🌹

Then I'll forget about the `nodemcu-32s` 🤣

me-no-dev, 2024-06-06:
@SuGlider can you check what you see about this board in IDE2.3.2 on Windows?

per1234, 2024-06-06:
The issue should be reopened since it is not resolved.

## Background

The platform's board definitions are configured to cause specific [custom board option](https://arduino.github.io/arduino-cli/dev/platform-specification/#custom-board-options) items to be displayed in the **Tools > Upload Speed** menu only when Arduino IDE/Arduino CLI is running on a certain operating system. They are doing this by controlling whether the option name property is defined using the OS-specific override feature:

https://arduino.github.io/arduino-cli/dev/platform-specification/#automatic-property-override-for-specific-os

Since the code in the `nodemcu-32s` board definition is a bit messy, it might be difficult to understand from studying it, so I'll provide a simplified example of the use of this technique. Let's say I have a definition for a board with ID `foo`, which has a custom option menu `bar`, with option items `pippo`, `pluto`, and `paperino`. Normally it would look something like this:

```text
menu.bar=Bar

foo.name=Foo
foo.menu.bar.pippo=Pippo
foo.menu.bar.pippo.asdf=1
foo.menu.bar.pluto=Pluto
foo.menu.bar.pluto.asdf=2
foo.menu.bar.paperino=Paperino
foo.menu.bar.paperino.asdf=3

[...]
```

However, if I changed the board definition to this:

```text
menu.bar=Bar

foo.name=Foo
foo.menu.bar.pippo=Pippo
foo.menu.bar.pippo.asdf=1
foo.menu.bar.pluto.linux=Pluto
foo.menu.bar.pluto.macosx=Pluto
foo.menu.bar.pluto.asdf=2
foo.menu.bar.paperino.macos=Paperino
foo.menu.bar.paperino.asdf=3

[...]
```

Linux users would see the following options in the **Tools > Bar** menu:

- Pippo
- Pluto

macOS users would see these

- Pippo
- Pluto
- Paperino

and Windows users would see this

- Pippo

## Explanation

Prior to the regression in https://github.com/espressif/arduino-esp32/pull/9277, Linux and macOS users  would see the following items in the **Tools > Upload Speed** menu when the "NodeMCU-32S" board was selected:

- 921600
- 115200
- 230400
- 460800

while Windows users would see the following items:

- 921600
- 115200
- 256000
- 230400
- 512000

(note the presence of "256000" and "512000" items and absence of "460800")

That worked fine until https://github.com/espressif/arduino-esp32/pull/9277 made the Linux/macOS-exclusive `460800` option the default by moving it to the top of the list in `boards.txt`. Arduino CLI automatically selects that default option regardless of operating system, which causes the board configuration to be invalid for Windows users since the use of [OS-specific overrides](https://arduino.github.io/arduino-cli/dev/platform-specification/#automatic-property-override-for-specific-os) in the  `460800` option name definition property causes that property to not be defined on Windows machines.

## Solution

A simple solution would be something like this:

```diff
--- a/boards.txt
+++ b/boards.txt
@@ -14434,8 +14434,7 @@ nodemcu-32s.menu.FlashFreq.80.build.flash_freq=80m
 nodemcu-32s.menu.FlashFreq.40=40MHz
 nodemcu-32s.menu.FlashFreq.40.build.flash_freq=40m

-nodemcu-32s.menu.UploadSpeed.460800.linux=460800
-nodemcu-32s.menu.UploadSpeed.460800.macosx=460800
+nodemcu-32s.menu.UploadSpeed.460800=460800
```

However, I guess there is a reason why the board definitions are configured to prevent Windows users from selecting a 460800 upload speed, which the naive solution shown above would allow.

If so, the best solution will be to set the default to the closest cross-platform speed, which is 230400:

```diff
--- a/boards.txt
+++ b/boards.txt
@@ -14434,6 +14434,9 @@ nodemcu-32s.menu.FlashFreq.80.build.flash_freq=80m
 nodemcu-32s.menu.FlashFreq.40=40MHz
 nodemcu-32s.menu.FlashFreq.40.build.flash_freq=40m

+nodemcu-32s.menu.UploadSpeed.230400.windows.upload.speed=256000
+nodemcu-32s.menu.UploadSpeed.230400=230400
+nodemcu-32s.menu.UploadSpeed.230400.upload.speed=230400
 nodemcu-32s.menu.UploadSpeed.460800.linux=460800
 nodemcu-32s.menu.UploadSpeed.460800.macosx=460800
 nodemcu-32s.menu.UploadSpeed.460800.upload.speed=460800
@@ -14441,9 +14444,6 @@ nodemcu-32s.menu.UploadSpeed.115200=115200
 nodemcu-32s.menu.UploadSpeed.115200.upload.speed=115200
 nodemcu-32s.menu.UploadSpeed.256000.windows=256000
 nodemcu-32s.menu.UploadSpeed.256000.upload.speed=256000
-nodemcu-32s.menu.UploadSpeed.230400.windows.upload.speed=256000
-nodemcu-32s.menu.UploadSpeed.230400=230400
-nodemcu-32s.menu.UploadSpeed.230400.upload.speed=230400
 nodemcu-32s.menu.UploadSpeed.512000.windows=512000
 nodemcu-32s.menu.UploadSpeed.512000.upload.speed=512000
 nodemcu-32s.menu.UploadSpeed.921600=921600
```

per1234, 2024-06-06:
Note that the `nodemcu-32s.menu.UploadSpeed.230400.windows.upload.speed` property has absolutely no effect (it defines a property named `windows.upload.speed`, which is not used anywhere). If the intention was to cause the `upload.speed` property value to be set to `256000` for Windows users, then it should have been written like this (the [OS-specific override](https://arduino.github.io/arduino-cli/dev/platform-specification/#automatic-property-override-for-specific-os) suffix must be placed at the end of the property name):

```text
nodemcu-32s.menu.UploadSpeed.230400.upload.speed.windows=256000
```

and that property definition must be placed after the universal definition of the property so that it will work as an override:

```text
nodemcu-32s.menu.UploadSpeed.230400.upload.speed=230400
nodemcu-32s.menu.UploadSpeed.230400.upload.speed.windows=256000
```

per1234, 2024-06-08:
@wangzongming @me-no-dev please reopen this valid, unresolved issue.

If I was confident in the validity of the solution I described in my previous comment, I would simply submit a pull request. However, I don't understand the reason why certain baud rates are only offered for specific operating systems so I don't think it would be responsible for me to do that.

So it will be important for this issue to be open so that the task of fixing the bug can be effectively tracked.

wangzongming, 2024-06-08:
> @wangzongming @me-no-dev please reopen this valid, unresolved issue.
>
> If I was confident in the validity of the solution I described in my previous comment, I would simply submit a pull request. [...]

Okay, I'm sure you can handle it. Thanks.

Jason2866, 2024-06-08:
Since you are at the speed settings. I use 1500000 with all my boards and a Mac M1
Never had issues. Interestingly using 460800 does often fail.

me-no-dev, 2024-06-08:
As far as I remember 460800 was not compatible with Windows before. Limitation was introduced by Arduino and we just kept it the same. I am not a Windows user myself and have used 460800 only with S3 native USB CDC port.

wangzongming, 2024-06-08:
> As far as I remember 460800 was not compatible with Windows before. Limitation was introduced by Arduino and we just kept it the same. I am not a Windows user myself and have used 460800 only with S3 native USB CDC port.

Yes, the latest `Arduino IDE` and the latest `esp32` cannot appear together.
Otherwise selecting 'Nodemocu-32s' will not be available in Windows 11.

dev-dakky, 2024-06-09:
Glad to know this is a current issue and not something I was messing up on my end. Using the esp32 dev module is an easy enough workaround in the meantime haha

Armagedon13, 2024-06-29:
i have the same problem with 3.0.2 in windows 11.
look at the options, it doesn't have any. i reinstalled all, IDE and libraries, and nothing.
[image]

xtremekratos, 2024-06-30:
well i have the exact same problem with v3.0.2 on IDE 2.3.2 in windows 11.
can't find any way to edit the upload speed
[image]

jawadzafar19, 2024-07-10:
same senerio with me too.using window 10 pro

Invalid FQBN: getting build properties for board esp32:esp32:nodemcu-32s: invalid value '460800' for option 'UploadSpeed'

Compilation error: Invalid FQBN: getting build properties for board esp32:esp32:nodemcu-32s: invalid value '460800' for option 'UploadSpeed'

jawadzafar19, 2024-07-10:
@me-no-dev  caan help me out set the upload speed 
[image]

Armagedon13, 2024-07-11:
> well i have the exact same problem with v3.0.2 on IDE 2.3.2 in windows 11. can't find any way to edit the upload speed [...]

try to use the module Node32s

me-no-dev, 2024-07-11:
Fix is coming: https://github.com/espressif/arduino-esp32/pull/10012

closed 2024-07-11 by me-no-dev

WuChungPing, 2024-07-29:
try this~
nodemcu-32s.menu.UploadSpeed.460800=230400
nodemcu-32s.menu.UploadSpeed.460800.linux=460800
nodemcu-32s.menu.UploadSpeed.460800.macosx=460800
nodemcu-32s.menu.UploadSpeed.460800.upload.speed=230400

UltimaBear14, 2024-07-30:
nodemcu-32s.menu.UploadSpeed.460800.upload.speed=230400
Where do we put this line? 

Thanks