espressif/arduino-esp32 #9787: New timer API usage could lead to log flooding

mathieucarbou, 2024-06-06:
### Board

N/A

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v3.0.1

### IDE Name

N/A

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

New timer API can lead to log flooding.

I am currently using a library which is doing something like that:

```c++
void ARDUINO_ISR_ATTR startTimerAndTrigger(uint32_t delay) {
  timerWrite(timer, 0);
#if ESP_IDF_VERSION_MAJOR >= 5
  timerAlarm(timer, delay, false, 0);
  timerStart(timer);
#else
  timerAlarmWrite(timer, delay, false);
  timerAlarmEnable(timer);
  timerStart(timer);
#endif
}
```

Works fine on Arduino 2.

But on Arduino 3, the console is flooding with logs :

```
E (29382) gptimer: gptimer_start(348): timer is not enabled yet
```

This is because the gptimer API is now logging to the console apparently for any not expected state .

The problem is that the Arduino API does not let the user query this state.

In the case of this method, the timer could be either stopped or started before it is called. So if I add `timerStop(timer)`, it does not solve the issue, I get another error log from gptimer.

Solution would be to introduce a `bool timerStarted(timer)` method  to query the state of the timer.

Please let ne know if the solution is acceptable, I could quickly draft a PR.

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-06-07:
@mathieucarbou From what I see, if you are getting this error message, it means that Timer was not initialised. `timer_enable` is called inside `hw_timer_t *timerBegin(uint32_t frequency)` function. Can you send complete sketch to reproduce it? If `timerBegin` returns NULL, it means that timer initialisation failed.

Also the idea of having `timerStarted(timer)`, its not ideal with the new IDF timer driver API, as its not possible to get the state of the timer. We would need to store the states on our side, which will mean more memory usage.

mathieucarbou, 2024-06-07:
@P-R-O-C-H-Y : thank you for your reply!

So here is the PR I did to the project to adapt it to Arduino 3 (a few lines of code change). I tested that successfully, it works now.

https://github.com/fabianoriccardi/dimmable-light/pull/63/files

And here are the latest changes I did to make it work (on the left: before, on the right: work fine).

https://github.com/fabianoriccardi/dimmable-light/compare/15332cc5e536885b9832c6db4f7b6f9b024274d7..db3aaa622773ca683aba070422f9766bfba82c70

So there were 2 things: 
- the dev was doing `timerStop(timer);` just after begin, which is not required, and this is not the code which was causing issue
- the dev was starting a timer already started -> this was the issue

```c+
  timerWrite(timer, 0);
#if ESP_IDF_VERSION_MAJOR >= 5
  timerAlarm(timer, delay, false, 0);
#else
  timerAlarmWrite(timer, delay, false);
  timerAlarmEnable(timer);
#endif  
  timerStart(timer);
```

This sequence was replaced for ESP_IDF >= 5 by:

```c+
  timerRestart(timer);
  timerAlarm(timer, delay, false, 0);
```

P-R-O-C-H-Y, 2024-06-07:
@mathieucarbou Thank you for quick answer. I am glad everything works now correctly. If it's all for this issue, can you please close it? If you face any other issue feel free to open a new ticket :) Thank you

mathieucarbou, 2024-06-07:
FYI, the issue was caused by the `timerStart(timer);` call because the timer **was already started**.

I will close it, yes.

But I must say that overall, I found that migrating timers was not easy because the behaviour changed between 2 and 3 regarding logging for example, and the now non-tolerance for unexpected states. I don't say it is wrong, I prefer the new API,  it is strict, but it is harder to migrate too to avoid these logs.

So the migration doc should IMO explain that and also maybe give some exemples on how to properly migrate.

closed 2024-06-07 by mathieucarbou

P-R-O-C-H-Y, 2024-06-07:
> FYI, the issue was caused by the `timerStart(timer);` call because the timer **was already started**.
>
>
>
> I will close it, yes.
>
>
>
> But I must say that overall, I found that migrating timers was not [...]

Thank you for the feedback. We will try to improve the migration guide for future breaking changes.

GitHubLionel, 2024-10-14:
Hello,

I have exactly the same problem. Many questions and no really answer :
- After Stop timer, should I use Start or Restart ?
- when I initialize timerAlarm, is the timer start immediately or no (in version 2, I use Enable to start the timer) ? Should I call Stop if I don't want to immediately start ?
- when I initialize timerAlarm with autoreload to false, can I use Start or Restart to restart the timer or should I re-initialize timerAlarm ?
Thanks