espressif/arduino-esp32 #9791: WIFI AND BLE NOT WORKING

Lakshmidivija, 2024-06-06:
### Board

ESP32 WROOM32E

### Device Description

my ic is on my customized pcb board . uart code is working on ic but ble and wifi are not working

### Hardware Configuration

do i have to enable any pins inorder to enable wifi or ble . the same code worked fine for development kit but i am facing issue on my pcb

### Version

latest master (checkout manually)

### IDE Name

arduino ide

### Operating System

windows 11 pro

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

i am unable to detect wifi or ble after uploading my code . the same code worked fine on development board. but on my pcb ist not working. i am uploading the code using esp - prog v1.1 capuf programmer.

### Sketch

```cpp
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    // WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
    // it is a good practice to make sure your code sets wifi mode how you want it.

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    // reset settings - wipe stored credentials for testing
    // these are stored by the esp library
    // wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result

    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
    // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:   
}
this is the basic wifi code. do i need to change anything?
```

### Debug Message

```plain
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c i am getting this on my serial monitor
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Lakshmidivija, 2024-06-06:
// This example code is in the Public Domain (or CC0 licensed, at your option.)
// By Evandro Copercini - 2018
//
// This example creates a bridge between Serial and Classical Bluetooth (SPP)
// and also demonstrate that SerialBT have the same functionalities of a normal Serial
// Note: Pairing is authenticated automatically by this device

#include "BluetoothSerial.h"

String device_name = "ESP32-D";

// Check if Bluetooth is available
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

// Check Serial Port Profile
#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Port Profile for Bluetooth is not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin(device_name);  //Bluetooth device name
  //SerialBT.deleteAllBondedDevices(); // Uncomment this to delete paired devices; Must be called after begin
  Serial.printf("The device with name \"%s\" is started.\nNow you can pair it with Bluetooth!\n", device_name.c_str());
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
and this is the ble code i am using to check

me-no-dev, 2024-06-06:
If it works on deb board but not on your custom one, it would mean that you have a hardware issue with your own board. Maybe antenna is disconnected or shielded in some way. At least provide image of the board

Lakshmidivija, 2024-06-06:
[image]
[image]

Lakshmidivija, 2024-06-06:
[image]

me-no-dev, 2024-06-06:
I don't see anything obvious.
Can you try to enable core debug to level Verbose and then the following sketch:
```cpp
#include "WiFi.h"

void setup() {
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    WiFi.begin("AutoConnectAP","password");
    delay(10000);
}

void loop() {
    delay(1000);   
}
```

SuGlider, 2024-06-06:
Converting this issue to Discussion as it is related to HW or User Custom Board.