espressif/arduino-esp32 #9797: error: 'wifi_power_t' does not name a type

enjoyneering, 2024-06-07:
### Board

ESP32 Dev Module

### Device Description

standard dev module

### Hardware Configuration

nothing attached

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi,

I have a header file where I store all my WiFi settings, but after upgrading to version 3.0.1 it no longer compiles. I don’t understand why. I have included the file "WiFi.h" which includes "WiFiGeneric.h" which contains "wifi_power_t". What am I doing wrong?

### Sketch

```cpp
//wifi header file

#ifndef WIFI_STA_CONFIG_h
#define WIFI_STA_CONFIG_h

#include <Arduino.h>
#include <WiFi.h>

typedef struct
{
  wifi_power_t   power;     //station output power, -1dBm(120mA)..+19.5dBm(150mA) 
  wifi_ps_type_t sleep;     //radio sleep WIFI_PS_NONE=0 or WIFI_PS_MIN_MODEM=1 or WIFI_PS_MAX_MODEM=2
}
WIFI_STA_CONF_STRUCT;

endif
```

### Debug Message

```plain
error: 'wifi_power_t' was not declared in this scope; did you mean 'wifi_mode_t'?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-07:
> What am I doing wrong?

Nothing it seems. Can you give a more complete example that we can try to replicate?

SuGlider, 2024-06-07:
Just fixing `endif` to `#endif`, it compiles fine with Arduino IDE and Core 3.0.1.
Therefore, this issue can't be reproduced.

enjoyneering, 2024-06-08:
Yes, the simplified code compiles without problems. Thanks, your comments led me to a solution. My project also has "wifi.h". I separated local and global "wifi.h" like this:

#include <WiFi.h> //global
#include "wifi.h"    //local

And everything worked fine until version 3.0.0. I renamed the local "wifi.h" to "wifi_ap_sta.h" and the problem went away.

closed 2024-06-08 by enjoyneering