espressif/arduino-esp32 #9799: Brownout detected since v2.0.10+

joysfera, 2024-06-07:
### Board

ESP32

### Device Description

plain ESP32 WROOM 32UE module with HT7833 LDO

### Hardware Configuration

nothing else is attached

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Ubuntu 22

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I've got a complex application running web server, web SSL client, MQTT client, OneWire scanning and more. All Arduino cores from 1.0.0 to 2.0.9 work correctly and can run my app stable for hours, days and months.

However, every Arduino core from 2.0.11 up to 2.0.17 causes Brownout detection randomly within seconds after powering the device up (please note that Arduino core 2.0.10 cannot compile so it's not listed). Sometimes it ends up in kind of a boot loop where it brownouts several times in a row, sometimes it runs semi-stable for two-three minutes and then brownouts. 

It seems related to underlying SDK change (4.4.4 for 2.0.9, 4.4.5+ for 2.0.10+).

It looks like a change in SDK caused higher current spikes in Arduino core 2.0.10+ that my LDO does not handle well and the voltage drops below brownout level. So it is partially a hardware fault but can be fixed easily by downgrading the Arduino core to 2.0.9 or below.

I wish to find out what has changed in SDK 4.4.5+ or Arduino core 2.0.10+ and change it back so that the application running on my hardware would be stable again and I could use the latest Arduino core features and bug fixes.

Perhaps WiFi TX Power has changed in the SDK 4.4.5+ that causes it? I've noticed there used to be
**CONFIG_ESP_PHY_REDUCE_TX_POWER** defined in sdkconfig.h in Arduino core 2.0.9 but it is no longer defined in newer cores. Could this be a hint?

### Sketch

```cpp
Unfortunately I don't have a minimal sketch. Even if I had you wouldn't have the hardware with LDO I've got so it wouldn't help, anyway.
```

### Debug Message

```plain
Brownout detection
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-07:
I will suggest that you ask this same thing in the ESP-IDF issue tracker. We are in a sense only consumer of the SDK.

BTW have you tried erasing the flash before uploading newer version? And surely you should think of large cap on the power or changing the LDO to something that can handle larger currents (1A)

joysfera, 2024-06-07:
The hardware cannot be changed (easily), it is deployed in the field in many copies already. BTW, larger caps make it worse (unless the cap is insanely large so it handles the complete WiFi power spike - that requires more than 5000 µF).

Will ask in ESP-IDF, thanks, but perhaps it is not related to SDK - that is just my guess.

joysfera, 2024-06-07:
Just tried erasing the complete flash and that causes kind of a funny issue: since my WiFi credentials are gone the application starts [WiFi Manager](https://github.com/tzapu/WiFiManager) and that causes immediate and 100% reproducible Brownout.

So now I could provide a minimal sketch for you: just enabling the WIFI_AP causes the Brownout as well.

```Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:5b:43:b0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.2s
Hard resetting via RTS pin...
```

joysfera, 2024-06-07:
Tried 3.0.1 yesterday, spent quite some time fixing libraries' incompatibilities but in the end didn't manage to build my app. Will try again in a moment, perhaps the library managers will catch up in the meantime.

me-no-dev, 2024-06-07:
you can at least try to start the WiFi and see if it behaves better. Should have a newer WiFi driver

joysfera, 2024-06-07:
Thanks, will report back if 3.0.1 happened to help and will try asking in ESP-IDF and add a link here just for reference.

JAndrassy, 2024-06-07:
you could try to reduce the WiFi TX power with WiFi.setTxPower(WIFI_POWER_8_5dBm)

Jason2866, 2024-06-07:
@joysfera The solution is to stay at the version you are. Since the part which is responsible for wifi stuff is closed source (in IDF) nothing can be done. Using different older wifi libs will fail since the are not compatible (you get linker errors, or wifi fail silently).

joysfera, 2024-06-07:
@Jason2866 
[image]

Jason2866, 2024-06-07:
@joysfera  "is no less than 500mA"  -> 500mA is everything else than a safe value.

joysfera, 2024-06-07:
@JAndrassy didn't help, unfortunately. I've even tried different values but to no effect.

everslick, 2024-06-14:
Another thing you **could** do (it's risky though!!!) is to disable the brownout handler. This only works, if you have control over the linker flags though:

Add `-Wl,-wrap,esp_brownout_init` to the linker flags, then add

```
void __wrap_esp_brownout_init() {
}
```

to your code.

This might bring you over the WiFi init phase and you can than call `__real_esp_brownout_init();` later on.

joysfera, 2024-06-14:
Disabling the brownout at runtime (before enabling WiFi) stops the boot loop but my OneWire bus stops working.
I suspect the power spikes on SDK 4.4.5+ are real and the only fix is to return back to SDK 4.4.4.

spyder0069, 2024-06-15:
Curious.  If you perform a reset via the EN pin instead of a power cycle does everything work?  In this scenario power is already stable.  Just looking for confirmation that it works and that the issue is only during a power-up.   Did you try additional reset commands on your onewire?  What onewire sensors are you using?

kevinjterry, 2024-06-17:
I'm having a nearly identical issue. Brownouts on an LDL1117S33R LDO. Adding a lot of capacitance (> 440uF) helps, but does not fix the issue. Cold starts runs for a bit longer, then it essentially boot loops at wifi begin once warm. I've tested a handful of other LDO's and all have the same issue. I've also verified there is a short sag in voltage (2.8v- 3.0v) as wifi.begin occurs. 

WiFi.setTxPower(WIFI_POWER_8_5dBm) did not help.

I'm unsure how to downgrade the arduino core below 2.0.9 to test that.

joysfera, 2024-06-17:
@spyder0069 the power spikes repeat every few seconds when WiFi is active so the power is never stable.

@kevinjterry in Arduino IDE in "Board Manager" you'll find "esp32" from Espressif and in a drop down you can pick any version you wish. Then click "Install" and that's it.

kevinjterry, 2024-06-17:
> @kevinjterry in Arduino IDE in "Board Manager" you'll find "esp32" from Espressif and in a drop down you can pick any version you wish. Then click "Install" and that's it.

ah, understood. Thank you. I'm using platformio -- I didn't find anything obvious for swapping core versions when searching. I'm going to swap the regulator for a switching regulator and give that a shot as well.

Jason2866, 2024-06-17:
@kevinjterry No need to install anything else than Platformio. 
Simply do
```
platform = espressif32 @ ^6.7.0
platform_packages = platformio/framework-arduinoespressif32@3.20009.0
```

You find the available versions here https://registry.platformio.org/tools/platformio/framework-arduinoespressif32/versions?

kevinjterry, 2024-06-17:
> @kevinjterry No need to install anything else than Platformio. Simply do
>
> ```
> platform = espressif32 @ ^6.7.0
> platform_packages = platformio/framework-arduinoespressif32@3.20009.0
> ```
>
> You find the [...]

Thank you -- that worked great. 

Sadly, espressif32@^6.7.0 and arduinoespressif32@2.20009.0 was no different and the brownout condition continued.

joysfera, 2024-06-17:
@kevinjterry Could you please, just for sure, include the following line in your program (if it ever gets to printing  something on serial):

`Serial.println(ESP.getSdkVersion());`

kevinjterry, 2024-06-17:
@joysfera Absolutely, I included it and received SDK Version: v4.4.4

JAndrassy, 2024-06-18:
I have an older Wemos D1 ESP32 mini with this problem. It resets on enableSTA(). With ETH it works without problems. I didn't test older versions now but I know it works with 2.0.9.

I tested it with esp hosted fg firmware. It had brown-out too, but setting CONFIG_ESP_PHY_REDUCE_TX_POWER=y in sdkconfig helped.

With Arduino I didn't find a solution. Reducing TX power is only possible after enableSTA() or enableAP().

(my(newer ESP32 D1 mini works ok with 3..0.1)

joysfera, 2024-06-18:
@JAndrassy please clarify it for me as the ethernet confused me: does your Wemos D1 ESP32 mini work with WiFi on stock Arduino core v2.0.9 correctly or not?

JAndrassy, 2024-06-18:
> does your Wemos D1 ESP32 mini work with WiFi on stock Arduino core v2.0.9 correctly or not?

yes it does

>  as the ethernet confused me

it was to clarify that there is no other problem with 3.0.1 with that board. without WiFi everything works.

joysfera, 2024-06-18:
@JAndrassy thanks, so you're confirming this issue is real. Seems like the default TX power has increased in SDK 4.4.5+ and that causes instability on boards with less capable power circuits, that are in fact 100% reliable on SDK 4.4.4 or older.

So in theory rebuilding the current SDK with the CONFIG_ESP_PHY_REDUCE_TX_POWER (as I've pointed out in the original Description as the visible difference between Arduino core 2.0.9 and 2.0.10+) would allow us to use latest Arduino core again? Judging from your test in ESP-IDF where you said setting the CONFIG_ESP_PHY_REDUCE_TX_POWER helped.

Sounds like there's a chance for newer Arduino cores to be less power hungry again? @me-no-dev

me-no-dev, 2024-06-19:
Will be enabled in 3.0.2 https://github.com/espressif/esp32-arduino-lib-builder/pull/185

closed 2024-06-19 by me-no-dev

JAndrassy, 2024-06-25:
CONFIG_ESP_PHY_REDUCE_TX_POWER didn't help for my older ESP32 D1 mini

joysfera, 2024-06-25:
@JAndrassy Last week you wrote literally the opposite:
"setting CONFIG_ESP_PHY_REDUCE_TX_POWER=y in sdkconfig helped."

JAndrassy, 2024-06-26:
> @JAndrassy Last week you wrote literally the opposite: "setting CONFIG_ESP_PHY_REDUCE_TX_POWER=y in sdkconfig helped."
with an IDF 5 project (esp_hosted_fg firmware)