espressif/arduino-esp32 #9802: 3.0.x incompatibilities

habazut, 2024-06-07:
### Board

ALL

### Device Description

Not relevant

### Hardware Configuration

Not relevant

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

ALL

### Flash frequency

not relevant

### PSRAM enabled

yes

### Upload speed

not relevant

### Description

As stated in https://github.com/espressif/arduino-esp32/issues/8796 you want new issue instead of buinding on the existing. Here you go. 

Users have made you aware of the incompatibilities of the 3.0.x version a long time ago. For example here:
https://github.com/espressif/arduino-esp32/issues/8780

With your rollout of the 3.0.0 as an AUTOMATIC upgrade from 2.0.x you did cause a lot of extra work as a lot of code does not compile any more and then the user complains to the project which is only using your API and not to you who are responsible for the change. That is a major pain. You did issue the rollout in spite of the incompatibility. The 3.0.0 HAL is so different it could as well been another CPU architecture.

This above issue from 2023 shows you were very well aware that the 3.0.0 upgrade was a BREAKING change and nevertheless you did roll it out as an automatic upgrade.

It would have given the world much less grief if you would have created a new package (say esp32:idf5) instead of pretending that `esp32:esp32@3.0.x` is an upgrade to  `esp32:esp32@2.0.x`, a fact that you very well were aware of.

To be clear: 
I am NOT complaining about the ESP-IDF API that you do have no control of (as mentioned in #8780)
I am complaining that you have rolled this out as an AUTOMATIC upgrade in spite that as you have admitted you have NO CONTROL OF THE INCOMPATIBILITIES and you were WELL AWARE ABOUT IT.

I would not like want to treat the users of the projetcs where I feel responsible in this way.

### Sketch

```cpp
Any sketch using RMT or Timer or LEDC....
```

### Debug Message

```plain
Well, it does not even compile.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-07:
@habazut - There are many ways to keep using the ESP32 Arduino 2.0.x Core:

3.0.x is a major version change that has breaking changes as already said. This new version is indeed improving the API and preparing space for the new Espressif SoC that were launched.

2.0.x is the former one. We've just released 2.0.17 with bug fixes.
It can still be maintained by the community and Espressif development team.
There is an exclusive branch for it: https://github.com/espressif/arduino-esp32/tree/release/v2.x
New PRs with improvements or bug fixes shall be sent by modifying this specific branch.

We may not release new 2.0.x packages in the future.
But this branch can still receive new fixes and improvements.
In order to use the latest code from this branch, it is necessary to install a `sketchbook board`
This is done by creating `hardware\espressif\arduino-esp32` folder in the `sketchbook` folder of your computer.  
`\arduino-esp32` is obtained by `git clone` the Arduino v2.x branch.
After that, please execute `get.exe` or `get.py` from `arduino-esp32/tools/` folder in order to install all the necessary tools.

[image]

----------------------------------------
> With your rollout of the 3.0.0 as an AUTOMATIC upgrade from 2.0.x 

Please tell us more about the `AUTOMATIC`.
Using the Arduino IDE, it is possible to pick whatever version you want to use.
Select the ESP32, go to the versions, scroll it down, select 2.0.x (2.0.17 is the latest).
Press `Install` ... to install the desired version.
Pressing `Upgrade` will install the latest version (by its "numbers" - 3.x.x at this time).
Pressing `Remove` will remove ESP32 boards from the IDE selection, but not the `sketchbook user boards`.

[image]

habazut, 2024-06-07:
Automatic as in `arduino-cli core upgrade` will bring everything to incompatible latest version. The project does not have a chance to specify somewhere which of the incompatible versions is the right one for the project. How should the project specify that this project is a esp32:idf4 project and not a esp32:idf5 project? For practical reasons (how much code to rewrite) I think using the  idf4 vs idf5 is as differnt as porting from ESP8266 to ESP32. In some aspects even more.

SuGlider, 2024-06-07:
@Lucasczm - Could you please help this issue? @habazut wants to use `arduino-cli` and specify which core (2.0.x or 3.0.x) to use for build the project. Can you confirm that as your final target, @habazut?

Regarding the effort for migrating a code from 2.0.x to 3.0.x, there is a document:
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

Lucasczm, 2024-06-08:
I think what @habazut meant is that when run the `arduino-cli core upgrade`, it automatically updates to version 3.0.x.
You can set version 2.0.x:
```sh 
arduino-cli core uninstall esp32:esp32
arduino-cli core install esp32:esp32@2.0.17
```
You can also use PlatformIO, where you can configure the Arduino core version in each project

@lucasssvaz - I think you were the one who should be mentioned

SuGlider, 2024-06-08:
@Lucasczm - Thanks!! Very kind of you.

VojtechBartoska, 2024-06-11:
Migration guide is linked above, 2.X version is still available. 

We tried to reduce breaking changes for v3.x as much as possible but there are still some which are necessary to support new SoCs and features and this cannot be avoided in general in Software development. Changes are described in Migration Guide and we are also here to help you moving your current projects.

closed 2024-06-11 by VojtechBartoska