espressif/arduino-esp32 #9808: Enable Core dump

michelelenci, 2024-06-08:
### Board

all boards

### Device Description

ESP32 S3

### Hardware Configuration

no

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

Please enable the core dump and core dump to flash (#define CONFIG_ESP_COREDUMP_ENABLE) in the new Arduino SDK.
At the moment the function in #include "esp_core_dump.h" are only place holder. 
Before Arduino edf core 3.0.0 the functionality was enabled.

I think it is very important to have this function enabled by default, so that when there are crashes in the ESP32 it is possible to dump the coredump flash sector and analise it.

The sketch provided below works perfectly with Arduino core 2.0.17, doesnt work at all (aka dump is FF FF FF.. ) on Arduino core 3.0.1

### Sketch

```cpp
#include <esp_task_wdt.h>
#include "Brd_Config.h"
#include "CD.h"
#include "Dbg.h"
#include "esp_partition.h"
#include "esp_log.h"
#include "esp_core_dump.h"
#include "Srv_Config.h"

#define COREDUMP_FILE "/coredump.bin"
#define READ_CHUNKS 4096 /* Core dump partition size must be amultiple of READ_CHUNKS */

static const char* TAG = "CDModule";

bool CD_ReadCoreD(void);

void CD_Init(void) {
  CD_ReadCoreD();
}

bool CD_ReadCoreD(void) {

  const esp_partition_t* pt = NULL;
  size_t bytestoread = READ_CHUNKS;
  size_t bytesread = 0;
  uint8_t str_dst[READ_CHUNKS];
  bool append = false;

#if CONFIG_ESP_COREDUMP_ENABLE
   esp_core_dump_init();

   if (esp_core_dump_image_check() != ESP_OK) {
     Dbg_Printf("CoreDump: not found\n");
     return false;
   }
#endif

  pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");

  if (pt == NULL)
    return false;

  Dbg_Printf("CoreDump: starting partition read\n");
  size_t size = pt->size;
  /* Check that size is a multiple of READ_CHUNKS. We dont wat to read beyond our partition */
  if (size % READ_CHUNKS != 0) {
    Dbg_Printf("CoreDump: ERROR partition not aligned!\n");
    return false;
  }

  while (bytestoread > 0) {
    /* This loop is dumping 64KB, reset the WD... */
    esp_task_wdt_reset();
    //Dbg_Printf("Remaining partition size : %zu \n", size); //Too many printf! this loop is long

    esp_err_t er = esp_partition_read(pt, bytesread, str_dst, bytestoread);
    //Dbg_Printf("Dumping to bin file...\n"); //Too many printf! this loop is long
    if (!FS_WriteBinaryFile(COREDUMP_FILE, str_dst, READ_CHUNKS, append)) {
      Dbg_Printf("CoreDump: cannot write CoreDump...\n");
      return false;
    }
    append = true;

    if (er != ESP_OK) {
      Dbg_Printf("READ IS NOT OK!\n");
      ESP_LOGE(TAG, "CoreDump: partition read error : %s", esp_err_to_name(er));
      return false;
    } else if (er == ESP_OK) {
      bytesread = bytesread + READ_CHUNKS;
      size = size - READ_CHUNKS;

      if (size <= READ_CHUNKS) {
        bytestoread = size;
      }
    }
  }
  Dbg_Printf("CoreDump: written to bin file succesfully.\n");
  //esp_core_dump_image_erase();
  return true;
}
```

### Debug Message

```plain
no debegu massage, just run the snipped on 2.0.17 and run it with 3.0.1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-10:
I have added a PR to enable coredump to flash, but be aware that: `Core dumps to Flash are not available if PSRAM is used for task stacks.`

michelelenci, 2024-06-10:
Thanks a lot @me-no-dev , in my case PSRAM is not used at all.

closed 2024-06-11 by me-no-dev