espressif/arduino-esp32 #9810: Any reason why "esp_sntp_getreachability" and "esp_sntp_getoperatingmode" are not included in 3.0.1?

enjoyneering, 2024-06-09:
### Board

ESP32 Dev Module

### Device Description

plain module

### Hardware Configuration

plain module

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Any reason why "esp_sntp_getreachability" and "esp_sntp_getoperatingmode" are not included in 3.0.1

Old 2.x.x has "sntp_getreachability" and "sntp_getoperatingmode"

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>

#include <esp_sntp.h>
#include <time.h>

uint16_t timeOut = 10000;

void setup()
{
  Serial.begin(115200, SERIAL_8N1);
  Serial.println();

  esp_sntp_setoperatingmode(SNTP_OPMODE_POLL);
  esp_sntp_servermode_dhcp(false);

  configTzTime("GMT0", "time.google.com");

  if (WiFi.status() == WL_CONNECTED)
  {
    while ((esp_sntp_getreachability(0) & 0x01) != 0x01) //check 8-th bit of reachability shift register
    {
      if (timeOut-- < 0) {break;}

      delay(1000);
    }

  if   (timeOut > 0) {Serial.println(F("packet from NTP server received"));}                  
  else               {Serial.println(F("Oops! connection to NTP server failed"));}
  }

  Serial.printf_P(PSTR("operation NTP mode %s\n"), esp_sntp_getoperatingmode() ? "LISTEN" : "POLL");
}

void loop()
{
  //empty
}
```

### Debug Message

```plain
error: 'esp_sntp_getreachability' was not declared in this scope
error: 'esp_sntp_getoperatingmode' was not declared in this scope; did you mean 'esp_sntp_setoperatingmode'?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-06-09:
No Arduino API calls/functions. The underlying IDF has changed to v5.1.

lbernstone, 2024-06-09:
New API in LwIP:
https://github.com/espressif/esp-lwip/blob/542ba2997f6cc14fe9c3d781bf9b0d890cd70bb1/src/apps/sntp/sntp.c#L739
https://github.com/espressif/esp-lwip/blob/542ba2997f6cc14fe9c3d781bf9b0d890cd70bb1/src/apps/sntp/sntp.c#L752

enjoyneering, 2024-06-09:
Hi lbernstone,

I don't think this is a new API. The "sntp_getreachability" and "sntp_getoperatingmode" were present in arduino-esp32 v2.x. For some reason they were not included in version 3.0.1 (see the code below)

```cpp
#include <Arduino.h>
#include <WiFi.h>

#include <esp_sntp.h>
#include <time.h>

int16_t timeOut = 10;

void setup()
{
  Serial.begin(115200, SERIAL_8N1);
  Serial.println();

  esp_sntp_setoperatingmode(SNTP_OPMODE_POLL);
  esp_sntp_servermode_dhcp(false);

  configTzTime("GMT0", "time.google.com");

  if (WiFi.status() == WL_CONNECTED)
  {
  //while ((esp_sntp_getreachability(0) & 0x01) != 0x01) //check 8-th bit of reachability shift register
    while ((sntp_getreachability(0) & 0x01) != 0x01)     //check 8-th bit of reachability shift register
    {
      if (timeOut-- < 0) {break;}

      delay(1000);
    }

  if   (timeOut > 0) {Serial.println(F("packet from NTP server received"));}                  
  else               {Serial.println(F("Oops! connection to NTP server failed"));}
  }

//Serial.printf_P(PSTR("operation NTP mode %s\n"), esp_sntp_getoperatingmode() ? "LISTEN" : "POLL");
  Serial.printf_P(PSTR("operation NTP mode %s\n"), sntp_getoperatingmode() ? "LISTEN" : "POLL");
}

void loop()
{
  //empty
}
```

error: 'sntp_getreachability' was not declared in this scope
error: 'sntp_getoperatingmode' was not declared in this scope; did you mean 'sntp_setoperatingmode'?

enjoyneering, 2024-06-09:
Hi Jason2866,

I understand that arduino-esp32 v3.x started using ESP-IDF v5.1.4. But if you look at the v5.1.4 code, the “esp_sntp_getreachability” and “esp_sntp_getoperatingmode” functions are present.

For some reason, arduino-esp32 v3.x only enabled the functions "esp_sntp_setoperatingmode" (formerly sntp_setoperatingmode in arduino v2.x) and "esp_sntp_servermode_dhcp" (formerly "sntp_servermode_dhcp" in arduino v2.x). and did not move the ESP-IDF v5 functions "esp_sntp_getreachability" and "esp_sntp_getoperatingmode" to arduino-esp32 v3.x

lbernstone, 2024-06-10:
#include <lwip/apps/sntp.h>

enjoyneering, 2024-06-10:
Hi lbernstone,

If I include <lwip/apps/sntp.h> instead of #include <esp_sntp.h>, then the "sntp_set_time_sync_notification_cb()" and "esp_sntp_stop()" functions are lost.

lbernstone, 2024-06-10:
Did I say instead of?  You need to include all the headers from which you will be using functions.

enjoyneering, 2024-06-11:
Hi lbernstone,

If I include both 

#include <lwip/apps/sntp.h>
#include <esp_sntp.h>

I get 
error: 'const char* sntp_getservername(u8_t)' was declared 'extern' and later 'static'

and bunch of notes similar to this:
note: previous declaration of 'const char* sntp_getservername(u8_t)'

lbernstone, 2024-06-11:
Well, it looks like they are deprecating the reachability functions.  I'd post [upstream](https://github.com/espressif/esp-idf/issues) and ask about it.  Maybe it just fell through the cracks.

enjoyneering, 2024-06-11:
Thank you lbernstone. I have the same feeling that this just fell under through the cracks.

enjoyneering, 2024-06-14:
I dug a little deeper and it looks like <sntp.h> has been replaced with [<esp_sntp.h> in ESP-IDF](https://github.com/espressif/esp-idf/blob/master/components/lwip/include/apps/sntp/sntp.h#L9C67-L9C85).

Looks like "sntp_getreachability()" has been replaced "with esp_sntp_get_sync_status()". I have no other explanation why this function is no longer available.

For some reason "esp_sntp_getoperatingmode()" and "esp_sntp_getreachability()" were not included in ESP-IDF v5.1.4. In the latest version 5.5.2 they have been added again ([link1](https://github.com/espressif/esp-idf/blob/master/components/lwip/include/apps/esp_sntp.h#L233), [link2](https://github.com/espressif/esp-idf/blob/master/components/lwip/include/apps/esp_sntp.h#L226))

me-no-dev, 2024-06-14:
They now are included so in next release you will have them again

enjoyneering, 2024-06-14:
Thank you me-no-dev.

closed 2024-06-14 by enjoyneering