espressif/arduino-esp32 #9812: What causes the serial monitor to print

RF91, 2024-06-09:
### Related area

Serial Monitor with Esp32

### Hardware specification

Esp32 devkit, Pzemoo4tv30 

### Is your feature request related to a problem?

My serial monitor is not displaying the values measured by pzem004tv30 in the serial, Instead it's reading 
Ox8 (TGIWDT SYS RESET), boot:0x13 (SPI FAST_FLASH_BOOT)

Figsip: 0, SPIWP: Oxee

drv:0x00,q drv:0x00,d_drv:0x00,cn0_drv:0x00,hd_drv:0xC

DIO, clock div:l

0x3rfr0030, len: 1416

0x40078000, len: 14804

0x40080400, len: 4

0x40080404, len: 3356

y 0x4008059c

Jul 29 2019 12:21:46

### Describe the solution you'd like

I want to see the values of current, voltage, power and power factor on the serial measured Pzemoo4tv30 using Esp32 devkit board

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

vvs551, 2024-06-09:
There is not correct speed of Serial in this version
In my case if select in Serial monitor the speed to one step lower the out is restoring

SuGlider, 2024-06-09:
@RF91 

if it reads `TG1WDT SYS RESET`, that means that the application is reseting by Watchdog Timeout.
Something may have changed in the application that is casing a continuous reboot.

Try to find the part of the code that causes this issue.

Possible root causes:
- Flash access not available: some functions shall be in IRAM, specially ISR.
- PSRAM problems causing timeout.
- Tasks that have high priority and run for long time not allowing other tasks to run.

SuGlider, 2024-06-09:
>I want to see the values of current, voltage, power and power factor on the serial measured Pzemoo4tv30 using Esp32 devkit board

This is not an Arduino Software requirement for a new feature. Sorry, but we can't help with it.
In order to get such information, you need an oscilloscope or a good multimeter.

VojtechBartoska, 2024-06-11:
As mentioned above, here is no action point. I am closing this issue as wontfix, if needed you can reopen it.