espressif/arduino-esp32 #9813: Build is broken when `-DUSE_ESP_IDF_LOG` is used

yaqwsx, 2024-06-09:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module

### Hardware Configuration

Not relevant, build failure

### Version

v2.0.17

### IDE Name

Not relevant, build failure

### Operating System

Ubuntu 22.04

### Flash frequency

Not relevant, build failure

### PSRAM enabled

no

### Upload speed

Not relevant, build failure

### Description

I want to redirect logs to a custom function that decides where the logs should end up. Based on my understanding of source code and #4346 this is not possible.

It seems that `esp32-hal-log.h` allows for using `-DUSE_ESP_IDF_LOG` in order to use ESP-IDF logging facilities that support redirection of logs. However, once this option is specified, the compilation fails as many sources do not specify ESP-IDF logging tag (e.g., `WString.cpp`, `HardwareSerial.cpp`)

### Sketch

```cpp
#include <Arduino.h>

void loop() {}
void setup() {}

// Compile with -DUSE_ESP_IDF_LOG
```

### Debug Message

```plain
cores/esp32/esp32-hal-log.h:163:67: error: 'TAG' was not declared in this scope
 #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                                                   ^~~
tools/sdk/esp32/include/log/include/esp_log.h:403:81: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag __VA_OPT__(,) __VA_ARGS__); } \
                                                                                 ^~~
cores/esp32/esp32-hal-log.h:163:32: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                ^~~~~~~~~~~~~~~~~~~
cores/esp32/Tone.cpp:82:7: note: in expansion of macro 'log_e'
       log_e("Could not create tone task");
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-10:
Fixed in 3.0.1

VojtechBartoska, 2024-06-12:
it is supposed to be fixed here: https://github.com/espressif/arduino-esp32/pull/9686

Closing this ticket, if needed or you still face issue, please reopen it @yaqwsx. Thanks

closed 2024-06-12 by VojtechBartoska

yaqwsx, 2024-06-12:
Thank you for the responses. Does it mean this won't be cherry-picked into v2.0.x and only v3.x.x is supported?

VojtechBartoska, 2024-06-13:
We do not plan to release other version for 2.X branch, only if any crucial issue will occur.

Please consider migrating to version 3.X, there is a migration guide and we hope it will not cause you a lot of problems.

I am adding a label "backport needed" to this issue for tracking.