espressif/arduino-esp32 #9814: Support log redirection

yaqwsx, 2024-06-09:
### Related area

Core of the framework

### Hardware specification

Not relevant

### Is your feature request related to a problem?

The logging facilities provided by the Arduino framework are hard-coded to the main serial output. This is often unwanted (e.g. when the persistence of logs is required or the serial interface needs to be free for other purposes).

### Describe the solution you'd like

Allow the user to specify a custom sink function for the logs.

### Describe alternatives you've considered

I wanted to use the built-in ESP-IDF logging facilities; however, I ran into #9813 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-06-10:
```cpp
static void custom_debug_write_char(char c) {
  //send the char to wherever you want, but be aware that this code might be called from interrupt
}

ets_install_putc1((void (*)(char)) & custom_debug_write_char);
ets_install_putc2(NULL);
```

yaqwsx, 2024-06-12:
Thank you for the proposed solution. However,  I am afraid of unwanted consequences of the proposed solution - it doesn't redirect the output of logging, but the output of the underlying layer. Thus any code using the `ets_` function for output will be affected. 

A quick grep of the Arduino framework showed that `ets_printf` isn't used extensively (just to output the reboot message), however, I have already seen 3rd party out in the wild using it. Thus, I don't find the proposed solution to be exactly clean. On top of that, if there was an interface to supply custom `log_v` and `isr_log_v` functions, the logging facility could store the format string and arguments separately (e.g., to save bandwidth or storage). This is not possible in the proposed solution.

everslick, 2024-06-14:
`esp_log_set_vprintf(...);` in combination of setting `-DUSE_ESP_IDF_LOG -DTAG=\"CORE\"` in CFLAGS works for most of the logs for me.

igrr, 2024-06-14:
> logging facility could store the format string and arguments separately (e.g., to save bandwidth or storage). This is not possible in the proposed solution. On top of that, we can't supply separate function to handle isr and non-isr context in you proposed solution

The new logging library is in development in ESP-IDF and it should address both points. I think the initial version will appear in ESP-IDF before the end of the year.

Muskanjhawar, 2025-02-28:
> `esp_log_set_vprintf(...);` in combination of setting `-DUSE_ESP_IDF_LOG -DTAG=\"CORE\"` in CFLAGS works for most of the logs for me

where should i add this  `-DUSE_ESP_IDF_LOG -DTAG=\"CORE\"` ??

kaerbr, 2025-03-09:
@Muskanjhawar in the `platformio.ini `  
`build_flags = -DUSE_ESP_IDF_LOG -DTAG=\"CORE\"`

Muskanjhawar, 2025-03-10:
Can you please help me with some example code or some reference like how
can i use this?

Jason2866, 2025-03-13:
Moving to discussions since it is integrated and possible to use.