espressif/arduino-esp32 #9816: ESP::getFlashChipMode crashes on ESP32S3 boards

craiglink, 2024-06-09:
### Board

ESP32-S3-Matrix X1

### Device Description

Waveshared ESP32-S3-Matrix X1

### Hardware Configuration

no other devices

### Version

v2.0.16

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Function crashes.  Perhaps needs special handling like the H2 and C chips do in the main branch?

platform.ini
extends = env:esp32-s3-devkitc-1
platform = espressif32@6.7.0
framework = arduino
board_upload.flash_size = 4MB
board_build.flash_mode = qio
monitor_filters = esp32_exception_decoder
board_build.partitions = tools/4MB.csv

### Sketch

```cpp
#include <Arduino.h>

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  delay(2500);
  Serial.println(ESP.getFlashChipMode());
}
```

### Debug Message

```plain
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420208f4  PS      : 0x00060630  A0      : 0x8200174f  A1      : 0x3fcebf90  
A2      : 0x00000008  A3      : 0xfff80000  A4      : 0x00000000  A5      : 0x80000000  
A6      : 0x00000000  A7      : 0x00ffffff  A8      : 0x820025b4  A9      : 0x3fcebf70  
A10     : 0x000009c4  A11     : 0x3fc93c70  A12     : 0x42001928  A13     : 0x00000000  
A14     : 0x3fc9598c  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x420208f1:0x3fcebf90 0x4200174c:0x3fcebfb0 0x42002d29:0x3fcebfd0
  #0  0x420208f1 in EspClass::getFlashChipMode() at C:/Users/CraigLink/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:356
  #1  0x4200174c in loop() at src/Blink.cpp:11
  #2  0x42002d29 in loopTask(void*) at C:/Users/CraigLink/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 79de2a338ffa7305

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42020f7a
  #0  0x42020f7a in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_pm/pm_impl.c:855

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

craiglink, 2024-06-09:
cross-posting details provided elsewhere 

ESP.getFlashChipMode();   // Would crash in the new version at execution.
For esp32s3 this would contain:
     uint32_t spi_ctrl = REG_READ(SPI_CTRL_REG(0));
     To see, what SPI_CTRL_REG(0) returns in the new version:

  Serial.printf("Incorrect return value = 0x%X\n", SPI_CTRL_REG(0));
   This prints: "Incorrect return value = 0x8". esp32s3 cannot read from address 0x8.

   SPI_CTRL_REG is defined as:

     #define SPI_CTRL_REG(i) (REG_SPI_BASE(i) + 0x8)

   In the new version we have:

     #define REG_SPI_BASE(i) (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))
       (at .....sdk/esp32s3/include/soc/esp32s3/include/soc/spi_reg.h: line 122)

   before it was:

     #define REG_SPI_BASE(i) (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
       (at ...../sdk/esp32/include/soc/esp32/include/soc/spi_reg.h: line 19)

   So, go back to the old version:

  #undef REG_SPI_BASE
  #define REG_SPI_BASE(i) (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
  Serial.printf("Correct return value = 0x%X", SPI_CTRL_REG(0));

   The NEWLY INTRODUCED FILE .....sdk/esp32s3/include/soc/esp32s3/include/soc/spi_reg.h contains at line 122:

     #define REG_SPI_BASE(i) (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))
     For i == 0 result is 0!
     which delivers the wrong address in this case.

Jason2866, 2024-06-09:
Same issue in master is fixed with https://github.com/espressif/arduino-esp32/pull/9758
BUT the real issue is still there in IDF in ALL versions.

VojtechBartoska, 2024-06-10:
Hello, this have been fixed and is already released in version 3.0.1.

Link for the release: https://github.com/espressif/arduino-esp32/releases/tag/3.0.1

For migration from v2.x to v3.x, you can refer to migration guide: https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

I am closing this issue, if you have any problems we can reopen it.

Jason2866, 2024-06-10:
@igrr Can you help here? The issue is in IDF. The definitions for `REG_SPI_BASE(i)` are wrong. In earlier version of IDF 4.4 it was not defined at all for the S3.
There where PRs from @TD-er and me to correct this (https://github.com/espressif/esp-idf/pull/12559 and https://github.com/espressif/esp-idf/pull/1256)
The suggested changes from the PRs are not committed. The where replaced by a different approach. At least for the S3 the are wrong and are applied in branch `release/v4.4` and `release/v5.1` (maybe in the other branches too, not checked)

Jason2866, 2024-06-10:
@VojtechBartoska The issue is in branch release/v2.x and it is not solved there.
Please reopen.

me-no-dev, 2024-06-10:
@Jason2866 feel free to open a backport :) We do not guarantee that another 2.0.x will be released though. As you know, best case would be to be fixed in IDF. I have asked about it and I'm waiting a response.

Jason2866, 2024-06-10:
@me-no-dev Lets hope the issue will be fixed in IDF!

mathertel, 2024-08-17:
see also https://github.com/espressif/arduino-esp32/issues/9748

Parsaabasi, 2025-01-09:
Hello, 

I close this report Since it contains an older version that is no longer supported.  Please try newer versions and if the problem reoccurs, feel free to reopen it. 

Thanks.

Jason2866, 2025-01-09:
@Parsaabasi There is a workaround to get the OP issues solved. BUT the general underlying bug in IDF is still present in actual latest Arduino Core.
https://github.com/espressif/arduino-esp32/issues/9816#issuecomment-2156778320
Since this is not corrected.

closed 2025-01-09 by Parsaabasi