espressif/arduino-esp32 #9820: ESP-SR Basic.ino sketch - Wakeword? Special partition? One microphone or need two?

prothan, 2024-06-10:
### Board

Adafruit Metro ESP32-S3 16 MB Flash 8 MB PSRAM

### Device Description

Adafruit Metro ESP32-S3 with ONE IMP441, terminated as shown. Metro connected via usb c. 
 
LRCK       GPIO_NUM_11
SCLK       GPIO_NUM_12
SDIN       GPIO_NUM_10
VDD        POWER
GND        GROUND

### Hardware Configuration

Adafruit Metro ESP32-S3

### Version

v3.0.1

### IDE Name

Arduino 2.3.2

### Operating System

Win 10 Pro

### Flash frequency

40Hz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When I ran this sketch I would receive errors in the serial monitor. It wasn't until I read a forum post about needing to use a special "SR" partition table that I knew what the problem was. This needs to be documented in the sketch or on Git hub. Also, I did not see  documentation on what the actual wake word is, this would be helpful too. 

Once I updated to the special "SR" partition the sketch loaded fine, however, I am unable to detect any commands or wake words(guess Hi Esp?). This might be due to my setup, but it is not clear what is necessary to properly run this sketch. I have a single connection to INMP441. I triple checked the connections they are find. I tried sketch with I2S_SLOT_MODE_STEREO and SR_CHANNELS_STEREO. I also tried with I2S_SLOT_MODE_MONO and SR_CHANNELS_MONO. 

Lastly, it would be nice if there was an example of this functionality, without needing 16MB flash storage. Perhaps a smaller version could be created? 

I am certain there is a lot of work needed to get everything to work together. I appreciate your efforts!

### Sketch

```cpp
See Basic.ino.
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-10:
Check for solution on how to setup this microphone here: https://github.com/espressif/arduino-esp32/issues/9663#issuecomment-2138499486

lbernstone, 2024-06-11:
[This](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP_SR/src/esp32-hal-sr.c#L9) is not exactly self-explanatory.

VojtechBartoska, 2024-06-12:
@prothan Did the guidance mentioned above help?

prothan, 2024-06-13:
No, did not work. I was using esp32s3 Dev Module earlier to program the board in Arduino IDE. I switched over to Adafruit Metro Esp32s3. I added SR partition to the boards.txt file. Debug level set to verbose. Set i2s pin per Adafruit recommendations. PSRAM enabled. Did not see anything at console, thought strange with debug level set to verbose. Put Serial.println("hello") in loop, it keeps repeating, so running. This is with these defines and setup function. 

#define I2S_PIN_BCK 17
#define I2S_PIN_WS  47
#define I2S_PIN_DIN 16

#define LIGHT_PIN 13
//#define FAN_PIN   41

```
void setup() {
  Serial.begin(115200);

  pinMode(LIGHT_PIN, OUTPUT);
  digitalWrite(LIGHT_PIN, LOW);
  //pinMode(FAN_PIN, OUTPUT);
  //digitalWrite(FAN_PIN, LOW);

  i2s.setPins(I2S_PIN_BCK, I2S_PIN_WS, -1, I2S_PIN_DIN);
  i2s.setTimeout(1000);
  i2s.begin(I2S_MODE_STD, 16000, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO);

  ESP_SR.onEvent(onSrEvent);
  //ESP_SR.begin(i2s, sr_commands, sizeof(sr_commands) / sizeof(sr_cmd_t), SR_CHANNELS_STEREO, SR_MODE_WAKEWORD);
  
   if (!i2s.begin(I2S_MODE_STD, 16000, I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_MONO, I2S_STD_SLOT_LEFT)) {
    Serial.println("Failed to initialize I2S bus!");
    return;
  }
  
  // Configure the I2S receiver to transform 32-bit data to 16-bit data
  if (!i2s.configureRX(16000, I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_MONO, I2S_RX_TRANSFORM_32_TO_16)) {
    Serial.println("Failed to configure I2S RX transformation!");
    return;
  }
}
```

me-no-dev, 2024-06-13:
```cpp
#define I2S_PIN_BCK 17
#define I2S_PIN_WS  47
#define I2S_PIN_DIN 16

#define LIGHT_PIN 13
//#define FAN_PIN   41

void setup() {
  Serial.begin(115200);

  pinMode(LIGHT_PIN, OUTPUT);
  digitalWrite(LIGHT_PIN, LOW);
  //pinMode(FAN_PIN, OUTPUT);
  //digitalWrite(FAN_PIN, LOW);

  i2s.setPins(I2S_PIN_BCK, I2S_PIN_WS, -1, I2S_PIN_DIN);
  i2s.setTimeout(1000);

  ESP_SR.onEvent(onSrEvent);
  
   if (!i2s.begin(I2S_MODE_STD, 16000, I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_MONO, I2S_STD_SLOT_LEFT)) {
    Serial.println("Failed to initialize I2S bus!");
    return;
  }
  
  // Configure the I2S receiver to transform 32-bit data to 16-bit data
  if (!i2s.configureRX(16000, I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_MONO, I2S_RX_TRANSFORM_32_TO_16)) {
    Serial.println("Failed to configure I2S RX transformation!");
    return;
  }
  ESP_SR.begin(i2s, sr_commands, sizeof(sr_commands) / sizeof(sr_cmd_t), SR_CHANNELS_STEREO, SR_MODE_WAKEWORD);
}
```

prothan, 2024-06-13:
me-no-dev, ok, yes, should have that last line in. I thought that was specifically for wake word. Looks like it includes commands also, so I will keep that in there and give it a try. What is the wake word, is it documented somewhere?

me-no-dev, 2024-06-13:
> What is the wake word

Hi ESP

prothan, 2024-06-14:
I got it working, thanks. Only had to change the i2s.begin and ESP_SR.begin input values from original Basic.ino sketch to the following: 

```
i2s.begin(I2S_MODE_STD, 16000, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO);

ESP_SR.begin(i2s, sr_commands, sizeof(sr_commands) / sizeof(sr_cmd_t), SR_CHANNELS_MONO, SR_MODE_WAKEWORD);

```

prothan, 2024-06-17:
By the way, I tried your suggestion first. It did not work. The reason I tried below is because the esp skainet   https://github.com/espressif/esp-skainet/tree/master/examples/en_speech_commands_recognition was using same setup.

i2s.begin(I2S_MODE_STD, 16000, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO);

ESP_SR.begin(i2s, sr_commands, sizeof(sr_commands) / sizeof(sr_cmd_t), SR_CHANNELS_MONO, **SR_MODE_WAKEWORD);

VojtechBartoska, 2024-09-16:
Hello @prothan, sorry for late reply, is this issue still valid?

closed 2025-01-15 by Parsaabasi