espressif/arduino-esp32 #9831: Custom Partition in Board Configuration do not take FLASH Mode Size into account

thelastoutpostworkshop, 2024-06-11:
### Board

ESP32S3 Dev Module (ESP32)

### Device Description

ESP32S3-WROOM-1 Development Board

### Hardware Configuration

Problem occurs with no GPIO used

### Version

v3.0.1

### IDE Name

Arduino IDE and VS Code Extension for Arduino

### Operating System

Windows 10

### Flash frequency

QIO 80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I use a custom partition 4MB, no OTA, provided in the example documentation : https://github.com/espressif/arduino-esp32/blob/575a415719d875f88460ec1461408597744f7053/docs/en/tutorials/partition_table.rst#L2

Compiler report a max of 16MB for Flash, which is configured as 4MB in the board configuration in the Arduino IDE.

Le croquis utilise 5258829 octets (31%) de l'espace de stockage de programmes. Le maximum est de 16777216 octets.
Les variables globales utilisent 27712 octets (8%) de mémoire dynamique, ce qui laisse 299968 octets pour les variables locales. Le maximum est de 327680 octets.

### Sketch

```cpp
The problem occur with an empty sketch
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
There is no debug message, compiler report wrong flash size.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

thelastoutpostworkshop, 2024-06-11:
My custom partition in partitions.csv:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     36K,     20K,
factory,  app,  factory, 64K,     4000K,

me-no-dev, 2024-06-12:
The IDE has no way to read your custom partition scheme before running the build and take the partition size into account. That is why you select a larger partition from the menu, so you do not get bothered by "firmware will not fit".

thelastoutpostworkshop, 2024-06-12:
But the esptool read it correctly :
esptool.exe" --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB 
and will not tell me if my code do not fit

me-no-dev, 2024-06-12:
esptool has no way to relay that info back to the IDE

thelastoutpostworkshop, 2024-06-12:
I found the solution, in boads.txt, I just have to specify the max upload size for the custom partition, 4MB in my case:
esp32s3.menu.PartitionScheme.custom.upload.maximum_size=4096000

VojtechBartoska, 2024-09-16:
Hello, I am closing this ticket as asnwered. If needed, you can reopen it.

closed 2024-09-16 by VojtechBartoska