espressif/arduino-esp32 #9833: Send UDP package to server, but server got UDP package with wrong size.

fryefryefrye, 2024-06-12:
### Board

ESP32_S2

### Device Description

An ESP32_S2 Mini board

### Hardware Configuration

N/A

### Version

v2.0.14

### IDE Name

platformio

### Operating System

Win 7

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

While running for days.
The following will happen occasionally

Receive side got UDP package with wrong size.
Got UDP size of 40.
One second later, got UDP size of 60.
One second later, got UDP size of 80.
One second later, got UDP size of 100.
One second later, got UDP size of 120.
................

### Sketch

```cpp
void loop()
{
    delay(1000)
    m_WiFiUDP.beginPacket("xxx.com", 5050);
    m_WiFiUDP.write((const uint8_t *)&SomeData, 20);
    m_WiFiUDP.endPacket();
}
```

### Debug Message

```plain
log_e("could not get host from dns: %d", errno);
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

fryefryefrye, 2024-06-12:
When there is some trouble on DNS lookup, and the  result of "beginPacket" is not checked.
I think it will lead UDP stack with dirt data and cause the DNS lookup always failed.

The following code will fix this problem, but maybe some update is need is better for users. Because I spend month to catch the reason.

```
void loop()
{
    delay(1000)
    if (m_WiFiUDP.beginPacket("xxx.com", 5050))
    {
        m_WiFiUDP.write((const uint8_t *)&SomeData, 20);
        m_WiFiUDP.endPacket();
    }
}
```

The root reason caused this problem.

C:\Users\xxx\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiUdp.cpp
line 168

```
int WiFiUDP::beginPacket(const char *host, uint16_t port){
  struct hostent *server;
  server = gethostbyname(host);
  if (server == NULL){
    log_e("could not get host from dns: %d", errno);
    return 0;
  }
  return beginPacket(IPAddress((const uint8_t *)(server->h_addr_list[0])), port);
}
```

VojtechBartoska, 2024-06-12:
@me-no-dev can you please take a look? thanks

me-no-dev, 2024-06-12:
Please try with 3.0.1

closed 2025-01-15 by Parsaabasi