espressif/arduino-esp32 #9837: v3.0.0 detects / defaults to wrong XTAL frequency on ESP32-D0WDQ6-V3 - millis and UART baud are not correct

PaulZC, 2024-06-12:
### Board

SparkFun ESP32 Thing

### Device Description

The SparkFun ESP32 Thing is a 3rd Party Board based on the ESP32-D0WDQ6-V3 chip with a 26MHz XTAL

https://www.sparkfun.com/products/13907

[image]

### Hardware Configuration

Only USB

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

A full write-up, diagnosis and fix available at: https://github.com/sparkfun/ESP32_Thing_Plus/issues/7

In cores below 3.0.0, the XTAL frequency was detected correctly. millis and UART baud rates were correct.

From 3.0.0, the XTAL frequency is detected as or defaults to 40MHz. millis and UART baud rates are off by a factor of 1.54.

Using the Serial_All_CPU_Freqs example:

```
For the 2.0.17 version it produces ...
CPU Freq = 240 MHz
XTAL Freq = 26 MHz
APB Freq = 80000000 Hz

For the 3.0.1 version it produces ...
CPU Freq = 240 MHz
XTAL Freq = 40 MHz
APB Freq = 80000000 Hz
```

We have a work-around:

```
#include "soc/rtc.h"

  rtc_clk_xtal_freq_update((rtc_xtal_freq_t)26);
  rtc_clk_cpu_freq_set_xtal();
  setCpuFrequencyMhz(240);
```

But it would be really nice if the core could be corrected / updated... Thanks!

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/Serial_All_CPU_Freqs/Serial_All_CPU_Freqs.ino
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-12:
We could not keep the crystal to AUTO, because other things stopped working. I can not recall what was exactly breaking, but it was critical.

PaulZC, 2024-06-12:
That might explain it! ;-) Many thanks Ivan. Best wishes.

me-no-dev, 2024-06-12:
> Maybe it could be possible to specify / override the XTAL frequency through boards.txt?

Unfortunately this would require rework that is just not worth for a single type of board (I also do not know of another board with 26MHz crystal) which was manufactured in the very early ESP32 days. I have a few myself and would rather replace their crystals. Would have been great if we could keep the crystal to AUTO also. I tried to find out what was happening, but the change came in a PR with many, many changes and message was not so clear.

igrr, 2024-06-12:
> I also do not know of another board with 26MHz crystal

In my case it was with Heltec Lora boards, they used to come with 26 MHz XTALs. Unfortunately replacing an XTAL is not an option for me, but I went back to v2.0.x for the time being.

me-no-dev, 2024-06-12:
> I went back to v2.0.x for the time being.

You could try the workaround above

me-no-dev, 2024-06-12:
Created PR with support for SparkFun ESP32 Thing: https://github.com/espressif/arduino-esp32/pull/9844

@igrr similar thing can be done for your board too (or even through menu that adds `-DF_XTAL_MHZ=26`

PaulZC, 2024-06-13:
Excellent - thank you @me-no-dev

dfrodin, 2024-06-24:
I am the developer who originally reported the "ESP32 Thing" board CPU frequency issue to Sparkfun. I've retested with the 3.0.2 version of the ESP32 library and everything is behaving correctly. Thanks for your efforts.

emilfihlman, 2024-09-29:
Is this supposed to be working now?
I'm using the latest nightly and esp32 3.0.5 and running into this issue.

emilfihlman, 2024-09-29:
Initial findings:

So there was a patch, but a) it didn't patch esp32thing**_plus**, just esp32thing, ie esp32thing_plus is missing

`#define F_XTAL_MHZ 26  //SparkFun ESP32 Thing has 26MHz Crystal`
in variants/esp32thing_plus/pins_arduino.h

Secondly it still doesn't work, even though I've closed and opened and cleared /tmp/arduino etc many times for some reason and compiled and upload many times.

emilfihlman, 2024-09-29:
According to the datasheet (ESP32-WROOM-32D), this device should have 40MHz crystal, yet it definitely outputs at 74880 when Serial.begin(115200) is set.

Even stranger: after upload the device outputs at 74880, but then on plug out and back in or pressing reset button results in the device outputting on 115200 :D

Even even stranger, the device also initially outputs on the correct frequency but then switches to the wrong one after _after_

The only code on the device is Arduino RMT.

emilfihlman, 2024-09-29:
I've attempted to change the erase all flash option both on and off for multiple times in a row.

Update on the changing of output: it doesn't change the output on plug out/in, but it does on pressing the reset button.

Could there be an issue with esp32 init code? On fresh boot wrong crystal speed is set, but on reset where there is continuous power we get the correct speed?

emilfihlman, 2024-09-29:
For some, very interesting, reason, the device might have something to do with UART initialisation.

I have managed to narrow the issue down to:
```
#define PIN 16

void setup()
{
	rmt_data_t rawData[]={{.duration0=1, .level0=0, .duration1=1, .level1=1}};
	pinMode(PIN, OUTPUT);
	rmtInit(PIN, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 1000000);
	rmtWriteLooping(PIN, rawData, RMT_SYMBOLS_OF(rawData));
}

void loop(){}
```

When connected to USB, the issue exists. When connected to just a power socket, everything is fine.

emilfihlman, 2024-09-29:
Issue opened at https://github.com/espressif/arduino-esp32/issues/10388

lbernstone, 2025-01-08:
Just an additional final note here.  The easiest way that I see to handle this now is to set the CPU Frequency to 26MHz in the board menu, then add `setCpuFrequencyMhz(240);` at the top of your setup.

ElectronicallyE, 2025-03-31:
I'm having the same issue with the TTGO LoRa32 V1 with a 26 MHz crystal.

Jason2866, 2025-03-31:
Can't be compared to 2.0.17 since the underlying IDF is differently compiled. Auto detecting xtal frequency in 2.0.17 in IDF. This setting isn't possible in newer IDF since it brakes code in other Arduino places.
Have you checked that the board you selected has the xtal clock frequency entry 26Mhz and it is set?

ElectronicallyE, 2025-04-01:
Hi @Jason2866,

Here are the setting I used:

[image]

Jason2866, 2025-04-01:
@me-no-dev Do you think a solution is possible for devices using a 26Mhz crystal?

me-no-dev, 2025-04-01:
Arduino IDE: add a file named `build_opt.h` in the sketch folder containing `-DF_XTAL_MHZ=26`

pioarduino: @Jason2866 can help with the `platformio.ini` line to add that same flag

Let me know if this works

Jason2866, 2025-04-01:
If the compile define `-DF_XTAL_MHZ=26` is working, the setting needs to be added in pioarduino to the platformio.ini. It is added to the build flags with: `build_flags = -DF_XTAL_MHZ=26`

closed 2025-10-02 by Jason2866