espressif/arduino-esp32 #9847: ESP32 Latest 3 series

sagarpateldt, 2024-06-13:
### Board

ESP32 Dev module

### Device Description

[image]

I used ESP32 devkit module

### Hardware Configuration

same circuit.

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Window 11

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

i code upload in esp32 after update board esp32 2.0.17 to 3 series, so problem occurs "task not found" showing in serial print and esp32 also very slow work. i attech in image below. this error show very fast.

[image]

### Sketch

```cpp
I can't give sketch.
```

### Debug Message

```plain
12:05:28.475 -> E (9937) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.475 -> E (9942) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.475 -> E (9947) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.475 -> E (9952) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.475 -> E (9958) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.475 -> E (9963) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.509 -> E (9968) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.509 -> E (9973) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.509 -> E (9979) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.509 -> E (9984) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.509 -> E (9989) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.509 -> E (9994) task_wdt: esp_task_wdt_reset(763): task not found
12:05:28.542 -> E (9999) task_wdt: esp_task_wdt_reset(763): task not found
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-13:
you need to provide minimum sketch that can reproduce the issue. We do not have this problem. Also best if you enable core debug to Verbose in the board menu

VojtechBartoska, 2024-06-13:
Maybe your issue is related to Migration from 2.X to 3.X version, you can also take a look on https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html