espressif/arduino-esp32 #9848: Verify/Uplad throws error if username contains a special character

ChristianPrei, 2024-06-13:
### Board

doesn't matter: no board or e.g. ESP32C3 Dev Module

### Device Description

doesn't matter: no board or e.g. ESP32C3 Dev Module

### Hardware Configuration

doesn't matter: no board or e.g. ESP32C3 Dev Module without anything or ESP32C3 Dev Module with neopixel on PINs 5V, G and GPIO5

### Version

v3.0.1

### IDE Name

Arduiuno IDE 2.3.2

### Operating System

Windows 10

### Flash frequency

80MHz which works on another username

### PSRAM enabled

yes

### Upload speed

921600 which works on another username

### Description

Verify throws error which translates to
")" can syntactically not be used in this place.

```
FQBN: esp32:esp32:esp32c3
Using board 'esp32c3' from platform in folder: C:\Users\(keiner)\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1
Using core 'esp32' from platform in folder: C:\Users\(keiner)\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1

cmd /c if exist "C:\\WinTmp\\.arduinoIDE-unsaved2024513-2940-1jptzcq.4z3el\\BareMinimum\\partitions.csv" COPY /y "C:\\WinTmp\\.arduinoIDE-unsaved2024513-2940-1jptzcq.4z3el\\BareMinimum\\partitions.csv" "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\partitions.csv"
cmd /c if not exist "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\partitions.csv" if exist "C:\\Users\\(keiner)\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\esp32c3\\partitions.csv" COPY "C:\\Users\\(keiner)\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\esp32c3\\partitions.csv" "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\partitions.csv"
")" kann syntaktisch an dieser Stelle nicht verarbeitet werden.
cmd /c if not exist "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\partitions.csv" COPY "C:\\Users\\(keiner)\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\tools\\partitions\\default.csv" "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\partitions.csv"
        1 Datei(en) kopiert.
cmd /c IF EXIST "C:\\WinTmp\\.arduinoIDE-unsaved2024513-2940-1jptzcq.4z3el\\BareMinimum\\bootloader.bin" ( COPY /y "C:\\WinTmp\\.arduinoIDE-unsaved2024513-2940-1jptzcq.4z3el\\BareMinimum\\bootloader.bin" "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\BareMinimum.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\(keiner)\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\esp32c3\\bootloader.bin" ( COPY "C:\\Users\\(keiner)\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\esp32c3\\bootloader.bin" "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\BareMinimum.ino.bootloader.bin" ) ELSE ( "C:\\Users\\(keiner)\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.6\\esptool.exe" --chip esp32c3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\WinTmp\\arduino\\sketches\\D468621BD42A8699DA5F7BF67D705EF5\\BareMinimum.ino.bootloader.bin" "C:\\Users\\(keiner)\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32c3\\bin\\bootloader_qio_80m.elf" ) )
exit status 1

Compilation error: exit status 1
```
My Username is 
`(keiner)
`
Which leads to the environment variables
```
USERPROFILE=C:\Users\(keiner)
LOCALAPPDATA=C:\Users\(keiner)\AppData\Local
```

### Sketch

```cpp
void setup() {
}

void loop() {
}
```

### Debug Message

```plain
none
not getting so far
```

### Other Steps to Reproduce

Created another username without special characters. -> Error gone.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-13:
Not sure if we can do anything here. The username was not escaped by the IDE when it constructed the recipes.

cc: @pillo79

pillo79, 2024-06-13:
Thanks for the ping @me-no-dev, we are investigating. 

@ChristianPrei, it would be helpful if you could repeat the tests using your problematic user and the latest [Arduino CLI](https://arduino.github.io/arduino-cli/1.0/), since the CLI contains a number of fixes not yet in the current IDE. Thanks!

ChristianPrei, 2024-06-14:
Same error with latest Arduino CLI

Update procedure is a little unclear. I replaced C:\Program Files\Arduino IDE\resources\app\lib\backend\resources\arduino-cli.exe and Arduino IDE could not start. Replaced the file after Arduino IDE is started and got the same error.

closed 2025-01-15 by Parsaabasi