espressif/arduino-esp32 #9854: Query regarding esp-modbus.

kabirwbt, 2024-06-13:
### Board

ESP32 Dev Module

### Device Description

Plain module

### Hardware Configuration

Nothing as of yet

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

 Pop!_OS 22.04 LTS

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I wanted to ask how to use esp-modbus(https://github.com/espressif/esp-modbus) with esp32 core 3.0.1. It was earlier available under .arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/freemodbus/.

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-13:
Since IDF 5, modbus is external module. We would need to add it again

kabirwbt, 2024-06-13:
Thank you for your quick response.

closed 2024-06-13 by me-no-dev