espressif/arduino-esp32 #9856: Problem with the HA_temperature_sensor example - can't set a value which is shown in zigbee2mqtt

ruediger218, 2024-06-13:
### Board

esp32c6

### Device Description

esp32c6 dev module

### Hardware Configuration

no

### Version

v3.0.1

### IDE Name

arduino ide

### Operating System

windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Problem with the HA_temperature_sensor example - can't set a value which is shown in zigbee2mqtt as temperature
i'm using in zigbee2mqtt the following definition:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const device = {
    zigbeeModel: ['cc2538.router.v2'],
    model: 'cc2538.router.v2',
    vendor: 'jethome',
    description: 'zigbee router cc2538 with temperature sensor',
    fromZigbee: [fz.ignore_basic_report, fz.device_temperature],
    toZigbee: [],
    exposes: [e.device_temperature(), e.linkquality()],
};

module.exports = device;

### Sketch

```cpp
/*
    test: temperatursensor auf esp32c6

    example sketch
    https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor

    für die device config von zigbee2mqtt
    https://github.com/jethome-ru/zigbee-firmware/tree/master/ti/router/cc2538_cc2592
 */
[... 261 lines cut ...]

void loop() 
{
 static long last_millis=0;
 static int first=1;

  //empty, zigbee running in task
  if ((millis() - last_millis) > 10000){
    if (first) testwerte_update();
    //last_millis = millis();
    first = 0;
  }
}
```

### Debug Message

```plain
no errors - everything seems to work
but do not in zigbee2mqtt
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ruediger218, 2024-06-13:
btw: in zigbee2mqtt i see realistic link quality but no temperature
is my call to **esp_zb_zcl_set_attribute_val** correct for setting the temperature?

P-R-O-C-H-Y, 2024-11-14:
Hello @ruediger218, as your code uses esp-zigbee-sdk APIs, I am closing this issue.
If you want you can try the[ Zigbee Library](https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee) that was recently added to the Arduino core.
If you want to continue on this issue, I suggest to open an issue in the esp-zigbee-sdk repository.
Thanks

closed 2024-11-14 by P-R-O-C-H-Y