espressif/arduino-esp32 #9858: WiFi.setTxPower() always return +19.5dBm

enjoyneering, 2024-06-13:
### Board

ESP32 Dev Module

### Device Description

plain module

### Hardware Configuration

plain module

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

WiFi.setTxPower() always return +19.5dBm in AP or STA mode

I think there is a typo in the [setTxPower()](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L637) and [getTxPower()](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L645). 

### Sketch

```cpp
//in setTxPower() and getTxPower() instead of &&

if (!WiFi.STA.started() && !WiFi.AP.started()) {  //&& is wrong
    log_w("Neither AP or STA has been started");

//should be ||
if (!WiFi.STA.started() || !WiFi.AP.started()) { //|| is correct
    log_w("Neither AP or STA has been started");
```

### Debug Message

```plain
return +19.5dBm
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JAndrassy, 2024-06-14:
"Neither AP or STA has been started" is 
`!(WiFi.STA.started() || WiFi.AP.started())` 
which is same as
`!WiFi.STA.started() && !WiFi.AP.started()`

me-no-dev, 2024-06-14:
The following code resulted in `TX Power: 11.0dBm`: 
```cpp
  WiFi.STA.begin();
  if(!WiFi.STA.waitStatusBits(ESP_NETIF_STARTED_BIT, 10000)){
    Serial.println("Failed to start WiFi!");
  }
  WiFi.setTxPower(WIFI_POWER_11dBm);
  int pwr = WiFi.getTxPower();
  Serial.printf("TX Power: %.1fdBm\n", pwr * 0.25);
  WiFi.STA.connect(SSID, PASSWORD);
```

enjoyneering, 2024-06-14:
Thank you me-no-dev. My bad. It looks like there is already the pull request for this issue. I close mine.

closed 2024-06-14 by enjoyneering