espressif/arduino-esp32 #9861: HTTPClient should allow for setting own Accept-Encoding header

fleinze, 2024-06-14:
### Related area

HTTPClient

### Hardware specification

all ESP32 boards

### Is your feature request related to a problem?

Currently the Library automatically adds the header `"Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0\r\n"`, except when using HTTP1.0. 
This can be a problem in some cases and can not be changed without editing the library source.
Adding a custom Accept-Encoding-header with `addHeader()` will result in two headers beeing sent.

### Describe the solution you'd like

Add a function e.g. `HTTPClient::setAcceptEncoding` similar to `HTTPClient::setUserAgent` for overriding the default Accept-Encoding header.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-06-14:
Done: https://github.com/espressif/arduino-esp32/pull/9863 😄

fleinze, 2024-06-14:
Well, that was fast. Thank you!

closed 2024-06-17 by me-no-dev