espressif/arduino-esp32 #9864: FlashMemory Address for EEPROM

meakashrao, 2024-06-14:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Kit C

### Hardware Configuration

Nothing is connected, just the bare board

### Version

v3.0.0

### IDE Name

Platform IO

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

So, I have an Arduino code that reads from specific EEPROM Library and then when needed reads from it. 

Now, as per requirement everything needs to shifted to ESP-IDF,
So, I can't use EEPROM Library and there are already some machines in the Field and they need to be remotely updated(we have taken care of the remote update) but it's important for device configuration to read from these EEPROM Locations

Now since ESP doesn't have EEPROM and uses NVS(Flash Partition) for its EEPROM Emulation. Please correct me if I am wrong
I need to to what Flash address is the EEPROM Data written and Read from when I do EEPROM.commit() and EEPROM.read(int address).

My EEPROM.begin is 80

Partition Table:
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,0x5000,,
otadata,data,ota,0xe000,0x2000,,
app0,app,ota_0,0x10000,0x1E0000,,
app1,app,ota_1,0x1F0000,0x1E0000,,
spiffs,data,spiffs,0x3D0000,0x30000,,

The Partition table is kept the same for both ESP-IDF and Arduino Code

Please help me with how to find the EEPROM Addresses.

### Sketch

```cpp
for (int i = 0; i < 8; i++)
{
  MACHINEID[i] = byte(EEPROM.read(i + 11));
  MACHINEID[8] = '\0';
}

byte var_OTA = byte(EEPROM.read(0));
if (var_OTA == 255)
{
  OTA_flag = 0;
}
else
{
  OTA_flag = EEPROM.read(0);
}

byte var = byte(EEPROM.read(65));
if(var == 255)
{
  lidlessMode = 0;
}
else
{
 lidlessMode = EEPROM.read(65);
 if(lidlessMode !=1)
 {
   addErrorMessage("EEPROMLidlessNot1");
 }
}

byte encrypt_var = byte(EEPROM.read(encryptionEEPROMAddress));
// Serial.print("encrypt_var: ");
// Serial.println(encrypt_var);
if(encrypt_var == 255)
{
  useEncryption = 0;
}
else
{
  useEncryption = EEPROM.read(encryptionEEPROMAddress);
  //Serial.print("useEncryption: "); Serial.println(useEncryption);
  if(useEncryption >= 1) 
  {
    addErrorMessage("EEPROMEncrpyNot1-0");
  }
}
```

### Debug Message

```plain
The above code is some important configurations that I read from Arduino setup().
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-14:
The EEPROM is saved as a blob in NVS. You can "address" an offset in that blob, but the actual blob address changes when you save the EEPROM. It's not a constant address in flash

meakashrao, 2024-06-14:
> You can "address" an offset in that blob, but the actual blob address changes when you save the EEPROM. It's not a constant address in flash

Is there a way through which we can read the offset and read the data only once and in further power cycles I will use NVS (key value pair) as I will copy to the NVS location.

Out of the 4 mentioned, the Machine ID is the most important one and I need to read.

me-no-dev, 2024-06-14:
I am not aware of such way. You can look into the EEPROM library source and convert that to pure IDF. It's just an abstraction .

lbernstone, 2024-06-14:
Your time would likely be better spent updating whatever legacy code you have using EEPROM to use individual elements or a struct and store that in nvs rather than adding another inefficient compatibility layer.

closed 2025-01-15 by Parsaabasi