espressif/arduino-esp32 #9867: ESP32 WiFiScan: no networks found

yoursunny, 2024-06-15:
### Board

Heltec WiFi Kit 32

### Device Description

plain module on breadboard, powered by USB

### Hardware Configuration

nothing connected

### Version

v3.0.1

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 10

### Flash frequency

240 MHz

### PSRAM enabled

no

### Upload speed

021600

### Description

ESP32 WiFiScan example returns "no networks found".
I expect it returns at least 2.4 GHz WiFi networks at my location.

### Sketch

```cpp
Examples - WiFi - WiFiScan (unchanged)
```

### Debug Message

```plain
Allocated Bytes   :    67716 B (  66.1 KB)
  Minimum Free Bytes:   267388 B ( 261.1 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
Scan start
Scan done
no networks found

Scan start
Scan done
no networks found
```

### Other Steps to Reproduce

I tried v2.0.17 with same ESP32 board, and the WiFiScan example can find networks.
I tried v3.0.1 with an ESP32-C3 board, and the WiFiScan example can find networks.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-16:
I suggest you upload 3.0.1 with erase of flash first, in case some old PHY data is left on the device

yoursunny, 2024-06-18:
I performed **esptool.exe erase_flash** command and re-uploaded the sketch, but the issue persists.

me-no-dev, 2024-06-18:
ok, last suggestion is to call `WiFi.setMinSecurity(WIFI_AUTH_WEP);` before starting the scan.

yoursunny, 2024-06-18:
After reading recent issues, I realized that the Heltec V1 boards I have are using 26MHz crystal.
I have opened PR 9889 to update the board definitions.
With correct crystal frequency set, WiFiScan works on both boards.

PR #9889 (2024-06-19): fix(boards): set XTAL 26MHz for Heltec WiFi & LoRa V1
## Description of Change

For Heltec WiFi Kit 32 V1:

* Change crystal frequency to 26MHz.
* Add partition scheme menu.
* Delete PSRAM menu.

For Heltec WiFi LoRa 32 V1:

* Change crystal frequency to 26MHz.
* Change flash size to 4MB.
* Add partition scheme menu.

## Tests scenarios

The changes were tested on above-mentioned boards.
I have used these examples:

* ESP32/Serial/Serial_All_CPU_Freqs
* FFat/FFat_Test
* WiFi/WiFiScan

## Related links

Heltec WiFi Kit 32 V1 schematic: https://resource.heltec.cn/download/WiFi_Kit_32/WIFI_Kit_32 [... cut]
closed 2024-06-19 by me-no-dev