espressif/arduino-esp32 #9868: BluetoothSerial::flush() takes too long

vincadrn, 2024-06-15:
### Related area

Classic Bluetooth

### Hardware specification

All ESP32 boards

### Is your feature request related to a problem?

For a small but frequent Bluetooth serial data, using delay of 100 ms in `flush()` degrade the performance very much.
https://github.com/espressif/arduino-esp32/blob/cbf1e94dd5f9ac8bdabaae9603611a3b81d8fbe4/libraries/BluetoothSerial/src/BluetoothSerial.cpp#L878-L884

### Describe the solution you'd like

I'd suggest that the delay is reduced to 1 ms. By using much smaller delay, it can accomodate different kinds of serial data.

### Describe alternatives you've considered

Currently I have no alternative approaches in mind that is as efficient and minimal in changes.

### Additional context

When I try without using `flush()` for my project, typically the data is sent in its entirety in about 5 - 20 ms. However, when I try to use `flush()`, the data would be sent entirely in > 100 ms. For a bigger serial data, the loop in the function would cause other 100 ms of delay(s). To avoid the problem altogether, one can avoid using `flush()` at all. But then, there is no way to know and make sure that the transmission buffer is cleared.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-06-18:
As far as I see from the code, the message will not be sent in 100ms, but rather `flush()` will return after 100ms. `delay(2)` would be better. Can you please try?

vincadrn, 2024-06-18:
Yes, sure. I'll raise a PR to address this issue. Thanks.

closed 2024-06-19 by me-no-dev