espressif/arduino-esp32 #9870: printf compile "expects argument of type 'int'" warnings on IDF_VER 5.1

Rob58329, 2024-06-15:
### Board

Any ESP32 board

### Device Description

Board only

### Hardware Configuration

Board only

### Version

latest master (checkout manually)

### IDE Name

Arduno IDE v1.8.19

### Operating System

Windows 10

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

n/a

### Description

The latest "github.com/espressif/arduino-esp32" software (IDF_VER=v5.1) seems to have a NEW "printf" function, which gives compile warning (using Compiler Warnings=more) of:
"`New: format '%i' expects argument of type 'int' (aka int16_t), but argument has type 'int32_t' {aka 'long int'}`"

However I note that it does actually still take "int32_t" numbers.
On ESP32 an "int" is currently the same as "long int" which is "int32_t", so this change in the "printf" warnings does not seem to make sence for the ESP32.

(In general terms I would not be a fan of the re-definition of "int" [so it just becomes the equivalent of int16_t on all architectures] as for architecture-portability this would then necessitate the use of "int_fast16_t" instead of the simpler "int").
 

### Sketch

```cpp
void setup() {
  Serial.begin(115200); delay(3000);
  int32_t x=-1000000;
  Serial.printf("x=%i\n",x);
}

void loop() {}
```

### Debug Message

```plain
only compiler warnings
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-16:
Hi @Rob58329 . Change comes from ESP-IDF and since we are just consumers of it, there is not much we can do. You can file an issue in the ESP-IDF tracker, if you want to get more info on why that happened and how to properly manage it.

Rob58329, 2024-06-17:
@me-no-dev: 
>Change comes from ESP-IDF...  there is not much we can do.

Many thanks for this info - a pity but I suppose makes life "more interesting"...
[Closing issue.]

closed 2024-06-17 by Rob58329