espressif/arduino-esp32 #9876: loop freeze: arduino framework

Red-Owl, 2024-06-17:
### Board

ESP32 wt32-s3-wrover (wt32sc01 plus)

### Device Description

https://en.wireless-tag.com/product-item-26.html

### Hardware Configuration

https://en.wireless-tag.com/product-item-26.html

### Version

v3.0.1

### IDE Name

Platformio

### Operating System

win11

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

loop stop loop (undefined time) freeze and not crash cpu
WORKING: interrupt, pwm display, pwm generator, webserver, wifi, usb msc cdc, serial (debug), sd, temperature internal sensor
**NOT working any loop function**
led is only for testing loop 

### Sketch

```cpp
#define EXT_ICSE                    (GPIO_NUM_21)    

void loop(){
  lv_timer_handler();  //lvgl gui
  info_time_update();// displat current datetime
  geiger_tube_conversion(); // conversione int to double
  cpm_measure();  // update display data
 

       

      if(IS_CHARGE == true){
        battery_charge();// fake battery charge  (no ext battery)
      }

      info_sd_label_update(); // sch 4
      display_count();

         if(COLONANNA==true){
            digitalWrite(EXT_ICSE, HIGH);
          }else{
            digitalWrite(EXT_ICSE, LOW);
          }
          COLONANNA = !COLONANNA;

}
```

### Debug Message

```plain
none, no guru
```

### Other Steps to Reproduce

[platformio]
default_envs = wt32-sc01-plus

[env:wt32-sc01-plus]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
board_build.partitions = default_8MB.csv
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
upload_speed = 115200

lib_deps = 
	mathieucarbou/ESP Async WebServer@^2.10.5
	lvgl/lvgl@^8.3.1
	lovyan03/LovyanGFX@^0.4.18
	bblanchon/ArduinoJson@^6.21.2
	paulstoffregen/Time@^1.6.1
	fbiego/ESP32Time@^2.0.0
lib_extra_dirs =
	lib/async_web_server   ;smarthome asynctcp patch
	lib/ArduinoMD5   ; md5
build_flags = 
	-DNDEBUG
	-D_GNU_SOURCE 
	-D_POSIX_READER_WRITER_LOCKS
	; end
	-DPIO_FRAMEWORK_ARDUINO_ENABLE_EXCEPTIONS
	-DPIO_FRAMEWORK_ARDUINO_ENABLE_RTTI
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-I lib
	-D PLUS=1
	-D LV_LVGL_H_INCLUDE_SIMPLE
	-D LV_MEM_SIZE="(96U * 1024U)"
	-D CORE_DEBUG_LEVEL=0				; 5 verbose 0 none
	-D ARDUINO_USB_MODE=1
monitor_filters  = esp32_exception_decoder
build_type = debug
[env:detect_only_medium_or_high_defects]
check_severity = medium, high

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-06-17:
The Platformio setup you posted does not use Arduino Core 3.0.x.
There are several settings in `build_flags` which are non standard and can end in undefined behaviour or bootloops.

Red-Owl, 2024-06-17:
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries

if this config is not correct please help me to change to correct configuration

Jason2866, 2024-06-17:
Start with something like this. The choosen board fits perfectly to the wt32-sc01-plus
```
[platformio]
default_envs = wt32-sc01-plus

[env:wt32-sc01-plus]
platform = espressif32 @ 6.7.0
board = aventen_s3_sync
framework = arduino
upload_speed = 115200
lib_deps =
   mathieucarbou/ESP Async WebServer@^2.10.5
   lvgl/lvgl@^8.3.1
   lovyan03/LovyanGFX@^0.4.18
   bblanchon/ArduinoJson@^6.21.2
   paulstoffregen/Time@^1.6.1
   fbiego/ESP32Time@^2.0.0
build_flags =
   -DBOARD_HAS_PSRAM
monitor_filters = esp32_exception_decoder
```

Red-Owl, 2024-06-17:
Thank, done, on trial...

Red-Owl, 2024-06-17:
freeze :(

Jason2866, 2024-06-17:
Looks like a bug in the sketch. You have to shrink done the code and provide a simple sketch where the issue happens. Without no further help can be provided

Red-Owl, 2024-06-18:
bug here: lovyan03/LovyanGFX@^0.4.18

gpena208777, 2024-08-31:
This is still present. I am wondering if it from using WIFI and writing/reading inputs/outputs. in your case the LED. 
Replace LED writes with Serial prints and report back please

gpena208777, 2024-08-31:
> bug here: lovyan03/LovyanGFX@^0.4.18

See above

closed 2025-01-15 by Parsaabasi