espressif/arduino-esp32 #9878: ESP32 Arduino BLE Refactoring

SuGlider, 2024-06-17:
### Related area

BLE refactoring using NimBLE

### Hardware specification

Any

### Is your feature request related to a problem?

This is part of an effort to improve ESP32 Arduino.

### Describe the solution you'd like

Starting point for a BLE refactoring Plan.
Targets:
- Move from Blueroid to NimBLE library.
- Redesign the Arduino BLE interface to make it more Arduino API alike.

Suggestion:
- Change the actual BLE API to be similar to https://github.com/arduino-libraries/ArduinoBLE

Examples:
- There many examples in case we decide to go with ArduinoBLE API model: https://github.com/arduino-libraries/ArduinoBLE/tree/master/examples

### Describe alternatives you've considered

None.

### Additional context

None.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

h2zero, 2024-06-17:
ArduinoBLE already supports esp32. Only change to do here would be to configure BLE to be controller only and import the ArduinoBLE library, no Bluedroid or NimBLE required.

SuGlider, 2024-06-17:
> ArduinoBLE already supports esp32. Only change to do here would be to configure BLE to be controller only and import the ArduinoBLE library, no Bluedroid or NimBLE required.

Yes, it supports the ESP32-S3 module from the UNO R4 board and the boards that use Nina-W102 that has an ESP32 module.
But those are used as WiFi/Bluetooth modem using AT commands.

The idea here is to implement the ArduinoBLE API as part of ESP32 Arduino BLE Library as direct ESP-IDF calls, instead of using a UART to send and receive AT commands.

h2zero, 2024-06-17:
It also has the VHCI driver here: https://github.com/arduino-libraries/ArduinoBLE/blob/98ff550988912ffbaeb1d877970e9e05f1de0599/src/utility/HCIVirtualTransport.cpp#L138

Added in https://github.com/arduino-libraries/ArduinoBLE/pull/252

SuGlider, 2024-06-17:
@h2zero  - Thanks for the information!

> It also has the VHCI driver here: https://github.com/arduino-libraries/ArduinoBLE/blob/98ff550988912ffbaeb1d877970e9e05f1de0599/src/utility/HCIVirtualTransport.cpp#L138

I see, this is way to implement it with internal channel AT commands.
I'll try it.

> Added in [arduino-libraries/ArduinoBLE#252](https://github.com/arduino-libraries/ArduinoBLE/pull/25

I like the table showing resource utilization after compiling the different libraries' respective scan example!

VojtechBartoska, 2024-08-05:
related discussion: https://github.com/espressif/arduino-esp32/discussions/9836

VojtechBartoska, 2024-09-24:
Hello @h2zero,

I am contacting you as a Project Manager of Arduino ESP32 project. Our Team have been recently investigating the refactoring of BLE in our core and we also investigated your [NimBle-Arduino library](https://github.com/h2zero/NimBLE-Arduino).

As you know, we need to keep working solution also for Classic BT on ESP32 platform, but in general in case of implementing e.g. Matter, moving to NimBLE based implementation makes sense.

As you proposed before, we would like to open this approach of reusing parts of your library in the core and we would like to ask if it is okay for you and also if you would like to help us with your contribution?

We can share more details about our ideas of implementation and we will be more than happy to hear your thoughts regarding it. Rodrigo (@SuGlider) has his hands on this and has some suggestions already for BLE refactoting.

Thanks in advance

h2zero, 2024-09-24:
@VojtechBartoska sure, let's chat about it.

SuGlider, 2024-09-27:
@h2zero - I have to finish the Arduino Matter Library first and then we will talk here about the nimBLE refactoring. Thank you!

h2zero, 2025-02-06:
So much for talking about this I guess lol.

lucasssvaz, 2025-02-06:
Hi @h2zero. It was an communication issue. In the future we will be more careful and improve in situations like this. Before moving forward with the PR is there anywhere we can chat (like Discord or something similar) to discuss it and align our views ? If there is anything we can do to help you, let us know!

SuGlider, 2025-02-06:
> So much for talking about this I guess lol.

@h2zero, Thanks for the support!

@lucasssvaz is now in charge of this migration project. The general idea is to keep both, Bluedroid and NimBLE back code, guarded by `#ifdef NIMBLE | BLUEDROID` within the Arduino Layers in the ESP32 Arduino Core for the 3.2 version.

NimBLE would be the default option to be compiled within the Core static libraries and Bluedroid could be activated by `sdkconfig` setting when the Arduino project is built as an IDF Component.

At this point, we are analyzing the back compatibility and how it is compiled, and also running, based on all the Bluedroid current examples from ESP32 Arduino Library. Therefore, we are in early stages.

Please feel free to participate directly here or with the open PR.
Your feedback is very welcome! Let us know what you think about it.

h2zero, 2025-02-07:
Thanks @lucasssvaz and @SuGlider, I have some thoughts on this and we should definitely discuss it. You can reach me via the email in my profile and I am also on discord with the same username as here.

lucasssvaz, 2025-02-07:
@h2zero Will send you an email. Thanks!

eMUQI, 2025-06-27:
Thank you so much for your excellent work on this BLE refactoring.

I am very excited about the plan to align the API with the standard `ArduinoBLE` library. As a hobbyist and educator, this compatibility would be incredibly helpful for teaching and for porting projects.

This level of compatibility would be a game-changer for the community, making the ESP32 far more accessible.

Looking forward to this new feature. Keep up the fantastic work!

lucasssvaz, 2026-05-03:
@h2zero Thanks for the feedback, but I think there’s a bit of a mismatch in how this is being interpreted.

To be clear: parts of the current NimBLE backend were indeed originally based on NimBLE-Arduino, with proper attribution. It’s a solid and very flexible library and that work was valuable. That said, for this refactor those pieces have already been heavily modified, and the direction here is quite different overall.

This isn’t meant to be a wrapper or a thin layer on top of NimBLE-Arduino. The API and architecture are being redesigned and modernized around ESP-IDF 6.0, with a unified interface that works across both NimBLE and Bluedroid. That comes with different constraints and goals. You can take a look in https://github.com/espressif/arduino-esp32/pull/12564 (Your review and suggestions are always appreciated 😃)

Pulling NimBLE-Arduino in as a submodule doesn’t really fit what we’re trying to do. We want full control over the API, behavior, and release cycle, especially since we’re making breaking changes and spec-level fixes. Depending on an external project would make that harder.

On `std::function`, the tradeoffs are known. There is some overhead, but we’re prioritizing API clarity and ease of use. Dropping the callback class pattern and allowing lambdas directly makes the API much simpler in practice. Sharing callbacks across multiple clients is still straightforward without going back to the old model.

If someone prefers a different set of tradeoffs, NimBLE-Arduino is still a great option. This refactor is just aiming at a different set of goals.

tanakamasayuki, 2026-09-21:
Just sharing this.
I created a similar library to address a similar challenge.
https://github.com/tanakamasayuki/EspBle

I implemented a dual-host setup using NimBLE and Classic-only Bluedroid.
It is an experimental solution.

lucasssvaz, 2026-09-25:
Done in https://github.com/espressif/arduino-esp32/pull/12564. 4.0.0-RC1 includes the refactor already.

closed 2026-09-25 by lucasssvaz

afpineda, 2026-10-13:
Hi, @lucasssvaz .

I'm testing the 4.0.0-RC1 using an ESP32-DevKit-C board.
The "BLE/README.md" file states:

"Bluedroid will be replaced by NimBLE in version 4.0.0 of the Arduino Core. "

So I expected the new refactored BLE API to be based on NimBLE, but it is still based on Bluedroid.

Am I missing something?

```c++
#include<BLE.h>

void setup() {
  Serial.begin(115200);
  Serial.println(BLE.getStackName());
}

void loop() {
}
```

afpineda, 2026-10-13:
> 19\. The proposal is not set in stone yet and is open for feedback

I don't know if this is the proper place to report this.
I found a mistake in the `BLECharacteristic` class:

`using ReadHandler = std::function<void(const BLECharacteristic &chr, const BLEConnInfo &conn)>;`

should be defined as:

`using ReadHandler = std::function<void(BLECharacteristic &chr, const BLEConnInfo &conn)>;`

Otherwise, you can not write a handler to set the characteristic's value to be sent to the client. For example:

```c++
static void readHandler(
        const BLECharacteristic &chr,
        const BLEConnInfo &conn)
    {
            uint8_t data[dataSize];
            // the `data` array is filled here
            ...
            chr.setValue((const uint8_t *)data, dataSize);
        }
    }
```

**This does not compile** as `chr` is `const` but `setValue()` is not.

lucasssvaz, 2026-10-13:
Hi @afpineda,

The comment is out of date. Bluedroid will still be the default for ESP32 due to the support of BT Classic. NimBLE can still be used in ESP32 by using Arduino as an ESP-IDF component.

Regarding the bug, thanks for the report. I will add the fix in the next days.