espressif/arduino-esp32 #9888: ESP-IDF 5.2.2 Arduino as a component (IDFGH-13064)

meakashrao, 2024-06-18:
### Is your feature request related to a problem?

I was using ESP IDF 5.1.2 and for some BLE and WiFi feature I had to jump to ESP IDF 5.2.2(which is the stable version at this point)

Please provide support for Arduino as a component in IDF Version 5.2.2

I tried playing with the idf_component.yml file but I wasn't able to find a temporary hack to make my things work. 
[image]

### Describe the solution you'd like.

I would Arduino as a component being supported for IDF 5.2.2 

### Describe alternatives you've considered.

I have some legacy code that needs to be ported in ESP IDF, so require Arduino as a component and there is no other option

### Additional context.

I need EEPROM as it is used in Arduino and Arduino-OTA

me-no-dev, 2024-06-18:
Adding support for 5.2 is not a basic task. Also we have decided to skip 5.2 and add support for 5.3 instead, but that is not yet done either.

meakashrao, 2024-06-18:
Thank you for letting me know.

closed 2024-06-18 by meakashrao