espressif/arduino-esp32 #9891: WiFiClient and related macros - troubles when using forward declarations

cziter15, 2024-06-18:
### Board

doesn't matter

### Device Description

-

### Hardware Configuration

-

### Version

v3.0.1

### IDE Name

-

### Operating System

-

### Flash frequency

-

### PSRAM enabled

no

### Upload speed

-

### Description

WiFiClient, WiFiServer and other network-related have been changed to NetworkClient, NetworkServer and so on.
Defines are now used to maintain compatibility, but these defines are included inside class headers.

It will cause problem when you've used forward declarations in your header. This forces the user to include the implementation everywhere.

I suggest to take another approach and simply derive instead of using define.
```
class WiFiClient : public NetworkClient {}
```

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JAndrassy, 2024-06-18:
that would not work for return value of server.accept()

me-no-dev, 2024-06-18:
@cziter15 can you provide a basic test?

Also I suggest you switch to NetworkClient when you detect Arduino 3. As @JAndrassy said, this will not work everywhere either.

cziter15, 2024-06-18:
> that would not work for return value of NetworkServer::accept()

It's easy to trick it using fd() as we can pass fd() to underying implementation, but... there's WiFIClientSecure, which inherits from WiFiClient and this cannot be trivially handled.

Looks like there's no solution to this. Even using #define is not a good idea.

Jason2866, 2024-06-18:
Mhh, core 3.0.0 has breaking changes, why trying to be backwards compatible where it is not possible to do it correctly? Imho this is the wrong way. The code needs to be adopted to the changes.

cziter15, 2024-06-18:
> Mhh, core 3.0.0 has breaking changes, why trying to be backwards compatible where it is not possible to do it correctly? Imho this is the wrong way. The user code needs to be adopted to the changes. Every "trickery" bites back some day...

I think it is possible, but it will require wrapper-like approach. Selective define inside header is bad, because it forces you to include the header - forward declaration is not possible.

JAndrassy, 2024-06-19:
@cziter15 did you see this comment by me-no-dev

> @cziter15 can you provide a basic test?
> 
> Also I suggest you switch to NetworkClient when you detect Arduino 3.

me-no-dev, 2024-06-19:
In your CPP (should include `Arduino.h` or `esp_arduino_version.h`)

```cpp
#if ESP_ARDUINO_VERSION_MAJOR >= 3
class NetworkClient;
#else
class WiFiClient;
#endif
```

Or

```cpp
#if ESP_ARDUINO_VERSION_MAJOR >= 3
#define WiFiClient NetworkClient;
#endif

class WiFiClient;
```

cziter15, 2024-06-19:
> In your CPP (should include `Arduino.h` or `esp_arduino_version.h`)
> 
> ```c++
> #if ESP_ARDUINO_VERSION_MAJOR >= 3
> class NetworkClient;
> #else
> class WiFiClient;
> #endif
> ```
> 
> Or
> 
> ```c++
> #if ESP_ARDUINO_VERSION_MAJOR >= 3
> #define WiFiClient NetworkClient;
> #endif
> 
> class WiFiClient;
> ```

I believe this is more of a consistency issue than a technical one. Current solution, which involves using macros in `WiFiClient` and other files, isn't ideal because it relies on the `#define` directive.

A more effective approach exists, as demonstrated by Microsoft. For instance, in the WinTypes.h file, only declarations and defines are included, without any method definitions. The actual method implementations are placed in separate files.

You can see an example of this in the WinTypes.h file on GitHub: [WinTypes.h](https://github.com/gitpan/FTDI-D2XX/blob/master/WinTypes.h).

I feel that half-measures are being applied. In my opinion, either the definitions should be global (defined using compiler options, similar to the ESP32 or ARDUINO macros), or including the WiFiClient file should result in an error with a message like `#error WiFiClient has been deprecated, please use NetworkClient instead.`

me-no-dev, 2024-06-19:
> including the WiFiClient file should result in an error

How about a warning instead?

cziter15, 2024-06-19:
@me-no-dev Ah, yes, I was thinking of a warning. Busy day... :)

me-no-dev, 2024-06-20:
@cziter15 added PR https://github.com/espressif/arduino-esp32/pull/9909

me-no-dev, 2024-06-20:
@cziter15 try the latest changes please :) typedefs instead of defines

cziter15, 2024-06-20:
I think it's better to leave `#define` until future removal. I've spotted that typedef can generate ambiguity errors in my framework code (on esp8266 WiFiClient is under BearSLL namespace).

PR #9909 (2024-06-24): feat(net): Switch to typedefs for WiFiClient, WiFiServer, WiFiUdp and WiFiClientSecure
Deprecates WiFiClient, WiFiServer, WiFiUdp and WiFiClientSecure

Fixes: #9891
closed 2024-06-24 by me-no-dev