espressif/arduino-esp32 #9896: invert status never reset on serial0 after having been set to invert once

mha1, 2024-06-18:
### Board

esp32-s3-devkitc-1

### Device Description

esp32-s3-devkitc-1

### Hardware Configuration

Pin 3 (UART0 TX) and GND are connected to a logic analyzer.

Please refer to the test code below. Expected output is an alternating output of non-inverted and inverted serial signals.

For serial0 the logic analyzer trace shows `non-inverted, inverted, inverted, ...` which is wrong.

[image]

running the test code on serial1 shows the expected behavior. 

[image]

changing `arduino-esp32/cores/esp32/esp32-hal-uart.c` fixes the problem however it is unclear why the serial0 and serial1 behave differently. https://github.com/espressif/arduino-esp32/blob/99750cd37e4bef0875d36cff7e319f327d8532a2/cores/esp32/esp32-hal-uart.c#L517-L520.

```
if (retCode && inverted) {
  // invert signal for both Rx and Tx
  retCode &= ESP_OK == uart_set_line_inverse(uart_nr, UART_SIGNAL_TXD_INV | UART_SIGNAL_RXD_INV);
}
else
{  
  // reset invert
  retCode &= ESP_OK == uart_set_line_inverse(uart_nr, UART_SIGNAL_INV_DISABLE);
}
```

### Version

latest master (checkout manually)

### IDE Name

VSCode/platformio

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

It should be possible to change serial transmission from non-inverted to inverted and back to non-inverted. This doesn't work on serial0. Please refer to the test code I attached.

### Sketch

```cpp
#include <Arduino.h>
#include <HardwareSerial.h>

HardwareSerial serial(0);   // use serial(1) for comparison

void setup() {}

void loop() {
  serial.begin(115200, SERIAL_8N1, 3, 1, false);
  serial.write("hello");
  serial.flush();
  serial.end();

  serial.begin(115200, SERIAL_8N1, 3, 1, true);
  serial.write("hello");
  serial.flush();
  serial.end();
}
```

### Debug Message

```plain
see attached logic analyzer trace
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-19:
Yes, confirmed. Thank for the issue report, @mha1!
I'll add a PR as suggested.

SuGlider, 2024-06-19:
About why Serial and Serial0 behave differently, I'll try to figure it out.

mha1, 2024-06-28:
@SuGlider any news?

SuGlider, 2024-06-28:
I'll work on it today.

SuGlider, 2024-06-28:
@mha1 - What is the Arduino Core version that you have used to detect that Serial0 can't invert / revert the RX/TX signals?

mha1, 2024-06-28:
framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)

SuGlider, 2024-06-28:
Ok... There are a few bugs with Serial and 2.0.11. Please try 2.0.17 instead. 
Or try 3.0.2.

The PR will fix it for 3.0.3 and I can apply it in the Arduino `release/v2.x` branch.

SuGlider, 2024-06-28:
> framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)

This explains why it works with Serial1 and not with Serial0.
Many bug fixes wihtin HardwareSerial and ESP32 UART HAL were made after 2.0.11
It may be necessary to update your Arduino Core code, manually or using some PlatformIO setup.

mha1, 2024-06-28:
"Ok... There are a few bugs with Serial and 2.0.11. Please try 2.0.17 instead.
Or try 3.0.2."

You mean other bugs besides the invert one, aren't you?
You're fixing the invert bug with 3.0.3, correct?

SuGlider, 2024-06-28:
> "Ok... There are a few bugs with Serial and 2.0.11. Please try 2.0.17 instead. Or try 3.0.2."
> 
> You mean other bugs besides the invert one, aren't you? You're fixing the invert bug with 3.0.3, correct?

yes, correct.

SuGlider, 2024-06-28:
@mha1 - You can copy the HardSerial and HAL from `release/v2.x` into your local Arduino Core files.

https://github.com/espressif/arduino-esp32/blob/release/v2.x/cores/esp32/HardwareSerial.cpp
https://github.com/espressif/arduino-esp32/blob/release/v2.x/cores/esp32/HardwareSerial.h
https://github.com/espressif/arduino-esp32/blob/release/v2.x/cores/esp32/esp32-hal-uart.c
https://github.com/espressif/arduino-esp32/blob/release/v2.x/cores/esp32/esp32-hal-uart.h

After that, fix manually the Inversion...

SuGlider, 2024-06-28:
I'll also apply the same fix you have proposed into `release/v2.x` branch.

mha1, 2024-06-28:
So it'll be a 2.0.18?

SuGlider, 2024-06-28:
> So it'll be a 2.0.18?

I don't think so... New improvements can be done directly into ESP32 Arduino `release/v2.x` branch.
The community can also post PRs for the `release/v2.x` branch.
This will be the way to maintain the v2.0.x 

Therefore, users will be able to git clone/pull the 2.x core into their projects.

SuGlider, 2024-06-28:
@mha1 
I'd appreciate it, if you could confirm that by applying the patch you suggested solves the issue within 2.0.17 for Serial0 and Serial1. Thanks!

SuGlider, 2024-06-28:
I can confirm that the PR #9950 solves it for 3.0.3.

SuGlider, 2024-06-28:
> ~I can confirm that the PR #9950 solves it for 3.0.3.~ Serial0 still an issue...

Actually, yes, the PR #9950  works correctly with Serial0, Serial1 under 3.0.2.
It was my fault... I was using the wrong sketch board for testing.

mha1, 2024-06-28:
The fix works on framework-arduinoespressif32 @ 3.20016.0 (2.0.16). I don't know how to switch to 2.0.17 on PlatformIO

SuGlider, 2024-06-28:
No problem, 2.0.16 is good enough. Thank you!!

SuGlider, 2024-06-28:
I'll also add a PR for 2.x branch.

closed 2024-07-01 by me-no-dev

mha1, 2024-07-01:
Now how do I make use of the fix in PlatformIO?

me-no-dev, 2024-07-01:
> Now how do I make use of the fix in PlatformIO?

Wait until 3.0.3 is released

mha1, 2024-07-02:
@SuGlider added a PR for 2.x branch. How can I make use of it?

SuGlider, 2024-07-02:
> @SuGlider added a PR for 2.x branch. How can I make use of it?

Yes, but it is necessary to clone the branch into a `Sketch_board`.

There are instruction in the Web.
In a few words, it will create a board based on the cloned branch into `your_sketch_folder/hardware/espressif/esp32_arduino_cloned_folder`.

PR #9952 (2024-07-01): [release/2.x] fix(uart): Set back Pin signal polarity
## Description of Change
Backport to `release/v2.x` branch

Fixes a problem related to inverting signal polarity back to normal after a previous inversion.
This shall set the correct polarity in Serial.begin().

## Tests scenarios

``` cpp
#include <Arduino.h>
#include <HardwareSerial.h>

HardwareSerial serial(0);   // use serial(1) for comparison

void setup() {}

void loop() {
  serial.begin(115200, SERIAL_8N1, 3, 1, false);
  serial.write("hello");
  serial.flush();
  serial.end();

  serial.begin(115200, SERIAL_8N1, 3, 1, true);
  seria [... cut]