espressif/arduino-esp32 #9903: NeoPixel can crash system.

Spaceballs3000, 2024-06-19:
### Board

ESP32S3 Dev Module 

### Device Description

ESP32S3 Dev Module 

### Hardware Configuration

ESP32S3 Dev Module 

### Version

v3.0.1

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

System will crash if data is received on port 6656.  If you comment out either the one in main or UDP callback function, it will work fine.

### Sketch

```cpp
#include "WiFi.h"
#include "AsyncUDP.h"

#define RGB_BRIGHTNESS 20 // Change white brightness (max 255)
#define RGB_BUILTIN 21

const char * ssid = "xxxx";
const char * password = "xxxx";

AsyncUDP udp_in;

void setup()
{      
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  WiFi.waitForConnectResult();
  udp_in.listenMulticast(IPAddress(239,1,2,3), 6656);      
  udp_in.onPacket([](AsyncUDPPacket packet) { udpRXCallBack(packet); });    
}

void loop()
{
  neopixelWrite(RGB_BUILTIN,0,0,0); // off
}

void udpRXCallBack(AsyncUDPPacket &packet) 
{
  neopixelWrite(RGB_BUILTIN,0,0,RGB_BRIGHTNESS); // Blue
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327

assert failed: xEventGroupClearBits event_groups.c:495 (xEventGroup)

Backtrace: 0x40376f76:0x3fca5310 0x4037d42d:0x3fca5330 0x40383119:0x3fca5350 0x4037ff5e:0x3fca5480 0x420068c3:0x3fca54a0 0x4200697e:0x3fca54e0 0x42006796:0x3fca5500 0x42002e6a:0x3fca5590 0x42002e8e:0x3fca55b0 0x42005c0d:0x3fca5640 0x42005c62:0x3fca56e0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-19:
you should not just have `neopixelWrite(RGB_BUILTIN,0,0,0);` in the loop. There should be delay or this should be called just once

Spaceballs3000, 2024-06-19:
> you should not just have `neopixelWrite(RGB_BUILTIN,0,0,0);` in the loop. There should be delay or this should be called just once

I still hit crashes with an delay, i.e.
```
void loop()
{
  delay(1);
  neopixelWrite(RGB_BUILTIN,RGB_BRIGHTNESS,RGB_BRIGHTNESS,RGB_BRIGHTNESS); // White
}

void udpRXCallBack(AsyncUDPPacket &packet) 
{
  neopixelWrite(RGB_BUILTIN,0,0,RGB_BRIGHTNESS); // Blue
}
```

SuGlider, 2024-06-19:
Issue confirmed. It is a concurrency problem within `neopixelWrite()` and RMT.

SuGlider, 2024-06-20:
@Spaceballs3000 - The PR #9906 shall solve this issue. Please test it.

Spaceballs3000, 2024-06-20:
> @Spaceballs3000 - The PR #9906 shall solve this issue. Please test it.

Tested fix, I do not get the crashes anymore.
Thanks!

closed 2024-06-20 by Spaceballs3000

Nourbakhsh-Rad, 2024-09-09:
@SuGlider I still have this problem!
I am using the Adafruit_NeoPixel 'simple.ino' sketch for test

Board:                                      ESP32 Dev Module (ESP32-devkitC-v4)
Hardware Configuration:         ESP32 Dev Module
Version:                                    v3.0.4
IDE Name:                                Arduino IDE 2.3.2
Operating System:                   Windows 10
Flash frequency:                       80
PSRAM enabled:                      no
Upload speed:                         921600

SuGlider, 2024-09-09:
>@SuGlider I still have this problem!
> I am using the Adafruit_NeoPixel 'simple.ino' sketch for testing

Do you mean that using this [example](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino) it fails?

This is Adafruit repository. It sounds like some other users have reported similar issues, there.
Please check/post an issue at https://github.com/adafruit/Adafruit_NeoPixel/issues

@ladyada - FYI.

Nourbakhsh-Rad, 2024-09-09:
> Do you mean that using this [example](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino), it fails?
>
> This is Adafruit repository. It sounds like some other users [...]

Yes, this [example](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino)
I am using the latest version of the Adafruit_NeoPixel library and the Arduino ESP32 core

Similar issue reported in [#401](https://github.com/adafruit/Adafruit_NeoPixel/issues/401#issue-2505462349) and [#9903](https://github.com/espressif/arduino-esp32/issues/9903#issue-2362334081) and [#9906](https://github.com/espressif/arduino-esp32/pull/9906#issue-2363288523)

PR #9906 (2024-06-20): feat(rmt): Solve neopixel issue
## Description of Change
If `neopixelWrite()` is used from different tasks  / isr_callbacks, it may result in a concurrency problem because of many detach/attach calls in sequence, but not in synch.

This commit avoids initializing the neopixel GPIO again, after a first use.
It also improves a lot the response time of `neopixelWrite()`.
It still allows to change RMT pin configuration or pin function correctly.

## Tests scenarios
Tested using ESP32-S3. Please check the code from #9903

## Related links
Fix #9903