espressif/arduino-esp32 #9912: LittleFS.format(): task_wdt: delete_entry(233): task not found

yoursunny, 2024-06-20:
### Board

ESP-C3-32S-Kit

### Device Description

Datasheet: https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-32s-kit-v1.0_specification.pdf
In Arduino IDE, I have selected "LOLIN C3 Pico" with "USB CDC On Boot: Disabled".
This board works correctly in other examples under these settings.

### Hardware Configuration

plain module on breadboard, nothing attached

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When `LittleFS.format()` is invoked, the console logs the following error:

```text
E (618) task_wdt: delete_entry(233): task not found
[   404][E][esp32-hal-misc.c:159] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
```

Several seconds later, the board resets due to task watchdog.

### Sketch

```cpp
#include <LittleFS.h>

void setup() {
  Serial.begin(115200);
  Serial.printf("init\n");

  bool ok = LittleFS.begin();
  Serial.printf("begin %d\n", static_cast<int>(ok));

  ok = LittleFS.format();
  Serial.printf("format %d\n", static_cast<int>(ok));

  LittleFS.end();
  Serial.printf("end\n");
}

void loop() {
}
[... 153 lines cut ...]
```text
init
begin 1
E (1117) task_wdt: esp_task_wdt_reset(763): task not found
E (1118) task_wdt: esp_task_wdt_reset(763): task not found
E (1119) task_wdt: esp_task_wdt_reset(763): task not found
E (1214) task_wdt: esp_task_wdt_reset(763): task not found
E (1214) task_wdt: esp_task_wdt_reset(763): task not found
E (1215) task_wdt: esp_task_wdt_reset(763): task not found
E (1219) task_wdt: esp_task_wdt_reset(763): task not found
E (1224) task_wdt: esp_task_wdt_reset(763): task not found
E (1323) task_wdt: esp_task_wdt_reset(763): task not found
E (1324) task_wdt: esp_task_wdt_reset(763): task not found
E (1325) task_wdt: esp_task_wdt_reset(763): task not found
E (1328) task_wdt: esp_task_wdt_reset(763): task not found
E (1333) task_wdt: esp_task_wdt_reset(763): task not found
E (1338) task_wdt: esp_task_wdt_reset(763): task not found
E (1343) task_wdt: esp_task_wdt_reset(763): task not found
format 1
end
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-06-21:
You probably need to put a delay in loop() or else delete it with`vTaskDelete(NULL);`.

yoursunny, 2024-06-21:
Adding `void loop() { delay(1); }` does not change a thing.
The warning on ESP32-C3 comes from `disableCore0WDT` called by `LittleFS.format()` function:
https://github.com/espressif/arduino-esp32/blob/0ef29868749865e754c62595e4b16e7da24608c0/libraries/LittleFS/src/LittleFS.cpp#L106-L109

Jason2866, 2024-06-21:
Beside the fact the function call `format` should do the job, format is not needed when using LittleFS.  It should be possible to use the FS with

```arduino
#include <LittleFS.h>

void setup() {
  Serial.begin(115200);
  Serial.printf("init\n");

  bool ok = LittleFS.begin();
  Serial.printf("begin %d\n", static_cast<int>(ok));

  LittleFS.end();
  Serial.printf("end\n");
}

void loop() {
}
```

yoursunny, 2024-06-21:
My actual program is a unit test suite of my own library.
At the start of the scenario, I need to have a blank filesystem.
Hence, I want to invoke `format()` to ensure there's no file stored.

This was working fine with FFat but I'm trying to move to LittleFS.
https://github.com/yoursunny/esp8266ndn/blob/da3e7dc48c6b3276055156c9719e7f0b9cf3afc7/examples/unittest/unittest.ino#L134-L143

Jason2866, 2024-06-21:
@yoursunny LittleFS is the smallest of your migration challenges. NimBLE Arduino will not work for the new MCUs (C6 H2). See https://github.com/espressif/arduino-esp32/discussions/9836

yoursunny, 2024-06-21:
NimBLE is unrelated to this. 
I can use both NimBLE and Bluedroid.

dirkvranckaert, 2024-11-18:
Any update on this? Running into the exact same issue. I want to be able to remove all local stored files for a factory reset, though failing on ESP32 now...

dirkvranckaert, 2025-01-16:
@Parsaabasi The issue is still persistent for me, I'm on latest 3.1.1 ESP32 Board version on an ESP32-C6.

Whenever I do a format `LittleFS.format();` or the first time deploying to a board and executing `LittleFS.begin(true)` (which internally if I understood correctly) is also doing a format if needed, and because it's my first deploy after getting the board out of the box (or after I deployed to it with erase flash) it's probably needed...

But what happens is that the format (or begin) seem to have worked, although couple of seconds later I run into the below exception.

I've created a simple test sketch to reproduce it:

```
#include <LittleFS.h>

void setup() {
  Serial.begin(115200);
  delay(5000);

  if (!LittleFS.begin(true)) {
    Serial.println("LittleFS Mount Failed");
  } else {
    Serial.println("LittleFS initiated");
  }

  LittleFS.format();

  Serial.println("Storage prepping done");

}

void loop() {
  Serial.print("Working...");
  delay(2000);
}
```

And the output from that test (incl. crash):

```
07:10:07.975 -> ESP-ROM:esp32c6-20220919
07:10:07.975 -> Build:Sep 19 2022
07:10:07.975 -> rst:0x1 (POWERON),boot:0x3c (SPI_FAST_FLASH_BOOT)
07:10:07.975 -> SPIWP:0xee
07:10:07.975 -> mode:DIO, clock div:2
07:10:07.975 -> load:0x40875720,len:0x1228
07:10:07.975 -> load:0x4086c110,len:0xd38
07:10:07.975 -> load:0x4086e610,len:0x2f74
07:10:07.975 -> entry 0x4086c110
07:10:13.086 -> LittleFS initiated
07:10:13.086 -> E (5167) task_wdt: delete_entry(236): task not found
07:10:13.217 -> Storage prepping done
07:10:13.217 -> Working...Working...Working...E (10295) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
07:10:18.230 -> E (10295) task_wdt:  - IDLE (CPU 0)
07:10:18.230 -> E (10295) task_wdt: Tasks currently running:
07:10:18.230 -> E (10295) task_wdt: CPU 0: IDLE
07:10:18.230 -> E (10295) task_wdt: Aborting.
07:10:18.230 -> 
07:10:18.262 -> 
07:10:18.262 -> Core  0 register dump:
07:10:18.262 -> MEPC    : 0x40803830  RA      : 0x4080790e  SP      : 0x40813fd0  GP      : 0x4080c0b4  
07:10:18.296 -> TP      : 0x40814020  T0      : 0x40022494  T1      : 0x4080395a  T2      : 0x00000000  
07:10:18.296 -> S0/FP   : 0x4080f000  S1      : 0x4080d78c  A0      : 0x00000001  A1      : 0xfefefefe  
07:10:18.296 -> A2      : 0x00000000  A3      : 0x00000040  A4      : 0x00000000  A5      : 0x00000000  
07:10:18.296 -> A6      : 0x00000000  A7      : 0x00000000  S2      : 0x4080f000  S3      : 0x408135c0  
07:10:18.296 -> S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
07:10:18.325 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
07:10:18.326 -> T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
07:10:18.326 -> MSTATUS : 0x00000000  MTVEC   : 0x00000000  MCAUSE  : 0x00000000  MTVAL   : 0x00000000  
07:10:18.362 -> MHARTID : 0x00000000  
07:10:18.362 -> 
07:10:18.362 -> Stack memory:
07:10:18.362 -> 40813fd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40806aca
07:10:18.362 -> 40813ff0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[... 23 lines cut ...]
07:10:18.551 -> 408142f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
07:10:18.583 -> 40814310: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
07:10:18.583 -> 40814330: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
07:10:18.583 -> 40814350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
07:10:18.583 -> 40814370: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
07:10:18.615 -> 40814390: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
07:10:18.615 -> 408143b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
07:10:18.615 -> 
07:10:18.615 -> 
07:10:18.615 -> 
07:10:18.615 -> ELF file SHA256: bc5b4ec4c
07:10:18.615 -> 
07:10:18.712 -> Rebooting...
07:10:18.712 -> ESP-ROM:esp32c6-20220919
07:10:18.712 -> Build:Sep 19 2022
07:10:18.712 -> rst:0xc (SW_CPU),boot:0x3c (SPI_FAST_FLASH_BOOT)
07:10:18.712 -> Saved PC:0x4001975a
07:10:18.712 -> SPIWP:0xee
07:10:18.712 -> mode:DIO, clock div:2
07:10:18.712 -> load:0x40875720,len:0x1228
07:10:18.712 -> load:0x4086c110,len:0xd38
07:10:18.712 -> load:0x4086e610,len:0x2f74
07:10:18.712 -> entry 0x4086c110

```

dirkvranckaert, 2025-01-16:
Any idea when this can be looked into? Pretty crucial failure in there for a long time...

enotanet, 2025-01-18:
It seems dead library. To respond for a year...

fracz, 2025-01-23:
I am experiencing the same issue described here.

dirkvranckaert, 2025-01-23:
Great, I see there is progress in fixing this... @lbernstone @me-no-dev by when can we expect this to be merged and have an update available?

lbernstone, 2025-01-23:
Fixes get merged in the next release.  The code is there if you need it now.

dirkvranckaert, 2025-01-27:
Applied the change on my local copy and it seems to do the job perfectly! Any idea when a next version will be released / shipped?

closed 2025-01-27 by me-no-dev