espressif/arduino-esp32 #9913: ESP32-C6 causes connection issues & unexpected reboots when using WiFi + Deep Sleep

QuickJack, 2024-06-20:
### Board

Firebeetle 2 ESP32-C6

### Device Description

Using the Deep Sleep example in Arduino IDE works perfectly. However, after adding code to also create a WiFi connection, the board fails to correctly establish a connection and reboots unexpectedly. A typical part of the log is shown under "Description".

### Hardware Configuration

no

### Version

v3.0.1

### IDE Name

Arduino IDE 2 2.3.3-nightly-20240527

### Operating System

Ubuntu 22.04

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Adding a typical code snippet to connect to an existing WiFi AP causes reboots. At first boot the system always connects without issues. But the following boots following   esp_deep_sleep_start()  are causing issues. After an unexpected reboot, the RTC memory seems to be cleared. This can be seen by looking at the following log output:

> 21:40:00.476 -> Boot number: 1
> 21:40:00.476 -> Wakeup was not caused by deep sleep: 0

Normally, the number after "Boot number" should be incremented with each boot. However, after an unexpected reboot this is not the case. Counting restarts in this case.

In order to verify that this is not a power supply issue, I have checked it using a high quality USB-C cable together with a 65W power supply.

### Sketch

```cpp
/*
Simple Deep Sleep with Timer Wake Up
=====================================
ESP32 offers a deep sleep mode for effective power
saving as power is an important factor for IoT
applications. In this mode CPUs, most of the RAM,
and all the digital peripherals which are clocked
from APB_CLK are powered off. The only parts of
the chip which can still be powered on are:
RTC controller, RTC peripherals ,and RTC memories

This code displays the most basic deep sleep with
a timer to wake it up and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/
[... 169 lines cut ...]
21:39:48.575 -> ------------------------------------------
21:39:48.575 -> Software Info:
21:39:48.575 -> ------------------------------------------
21:39:48.575 ->   Compile Date/Time : Jun 20 2024 21:38:12
21:39:48.575 ->   Compile Host OS   : linux
21:39:48.575 ->   ESP-IDF Version   : v5.1.4-51-g442a798083-dirty
21:39:48.575 ->   Arduino Version   : 3.0.1
21:39:48.575 -> ------------------------------------------
21:39:48.575 -> Board Info:
21:39:48.575 -> ------------------------------------------
21:39:48.575 ->   Arduino Board     : DFROBOT_FIREBEETLE_2_ESP32C6
21:39:48.575 ->   Arduino Variant   : dfrobot_firebeetle2_esp32c6
21:39:48.575 ->   Arduino FQBN      : esp32:esp32:dfrobot_firebeetle2_esp32c6:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=80,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=verbose,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
21:39:48.575 -> ============ Before Setup End ============
21:39:48.673 -> [   704][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x40812e14
21:39:48.673 -> [   705][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x40812e14
21:39:51.793 -> [  3706][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4200776c
21:39:51.793 -> [  3707][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 15 successfully set to type GPIO (1) with bus 0x10
21:39:51.793 -> Boot number: 1
21:39:51.793 -> Wakeup was not caused by deep sleep: 0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-06-21:
You should be sure to disconnect and turn off the radio before deep sleep.
```
#include <WiFi.h>
#include <esp_wifi.h>
...
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  esp_wifi_deinit();
  esp_deep_sleep_start();
```

QuickJack, 2024-06-21:
I added those 3 lines of code. However, there are still connection issues and the board reboots after about 5 s.

```
13:04:28.880 -> Wakeup was not caused by deep sleep: 0
13:04:28.880 -> [  3709][W][WiFiGeneric.cpp:639] setTxPower(): Neither AP or STA has been started
13:04:28.880 -> [  3726][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY
13:04:28.974 -> [  3809][V][STA.cpp:184] _onStaEvent(): STA Started
13:04:28.974 -> [  3810][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 11 - STA_START
13:04:28.974 -> [  3811][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
13:04:29.488 -> ........[  8180][V][STA.cpp:214] _onStaEvent(): STA Disconnected: SSID: MYSSID, BSSID: 34:81:c4:1c:4b:f3, Reason: 2
13:04:33.330 -> [  8181][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 14 - STA_DISCONNECTED
13:04:33.330 -> [  8182][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 14 - STA_DISCONNECTED
13:04:33.330 -> [  8183][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE
13:04:33.330 -> [  8183][D][STA.cpp:153] _onStaArduinoEvent(): WiFi Reconnect Running
13:04:33.330 -> [  8184][W][STA.cpp:533] disconnect(): STA already disconnected.
13:04:33.491 -> ....ESP-ROM:esp32c6-20220919
13:04:35.085 -> Build:Sep 19 2022
13:04:35.085 -> rst:0xc (SW_CPU),boot:0x8 (SPI_FAST_FLASH_BOOT)
13:04:35.085 -> Saved PC:0x4001975a
13:04:35.118 -> SPIWP:0xee
```

QuickJack, 2024-06-21:
I have added the 3 lines you suggested before calling esp_deep_sleep_start(). My code looks like this:

```
/*
Simple Deep Sleep with Timer Wake Up
=====================================
ESP32 offers a deep sleep mode for effective power
saving as power is an important factor for IoT
applications. In this mode CPUs, most of the RAM,
and all the digital peripherals which are clocked
from APB_CLK are powered off. The only parts of
the chip which can still be powered on are:
RTC controller, RTC peripherals ,and RTC memories

This code displays the most basic deep sleep with
a timer to wake it up and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#include <WiFi.h>
#include <esp_wifi.h>

#define uS_TO_S_FACTOR 1000000ULL /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP 10          /* Time ESP32 will go to sleep (in seconds) */

#define STA_SSID "HEREWEGO"
#define STA_PASS "Test051102"

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0: Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1: Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER: Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD: Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP: Serial.println("Wakeup caused by ULP program"); break;
    default: Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason); break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(3000);  //Take some time to open up the Serial Monitor

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  //WiFi test
  WiFi.setTxPower(WIFI_POWER_MINUS_1dBm);
  WiFi.begin(STA_SSID, STA_PASS);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  /*
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) + " Seconds");

  /*
  Next we decide what all peripherals to shut down/keep on
  By default, ESP32 will automatically power down the peripherals
  not needed by the wakeup source, but if you want to be a poweruser
  this is for you. Read in detail at the API docs
  http://esp-idf.readthedocs.io/en/latest/api-reference/system/deep_sleep.html
  Left the line commented as an example of how to configure peripherals.
  The line below turns off all RTC peripherals in deep sleep.
  */
  //esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  //Serial.println("Configured all RTC Peripherals to be powered down in sleep");

  /*
  Now that we have setup a wake cause and if needed setup the
  peripherals state in deep sleep, we can now start going to
  deep sleep.
  In the case that no wake up sources were provided but deep
  sleep was started, it will sleep forever unless hardware
  reset occurs.
  */
  Serial.println("Going to sleep now");
  Serial.flush();
  digitalWrite(LED_BUILTIN, LOW);

  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  esp_wifi_deinit();

  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop() {
  //This is not going to be called
}
```

Jason2866, 2024-06-21:
After deep sleep wake up wifi is in undefined state. You have to init.

QuickJack, 2024-06-21:
Isn't it enough to call WiFi.begin()?

SuGlider, 2024-06-21:
Possible issue with power after a brownout - right after WiFi is started.,. 
This would explain RTC Memory reset.

Check USB cables, power suply etc.

SuGlider, 2024-06-21:
@QuickJack - I have tested it with a ESP32-C6-DevKitC-1 V1.2 - No issue found. It runs as expected.

#### Log:
```
14:28:47.352 -> ESP-ROM:esp32c6-20220919
14:28:47.352 -> Build:Sep 19 2022
14:28:47.352 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
14:28:47.352 -> SPIWP:0xee
14:28:47.352 -> mode:DIO, clock div:2
14:28:47.352 -> load:0x4086c410,len:0xc64
14:28:47.352 -> load:0x4086e610,len:0x2730
14:28:47.352 -> load:0x40875728,len:0x594
14:28:47.352 -> entry 0x4086c410
14:28:47.352 -> Boot number: 1
14:28:47.352 -> Wakeup was not caused by deep sleep: 0
14:28:47.352 -> ......Setup ESP32 to sleep for every 10 Seconds
14:28:47.352 -> Going to sleep now
14:28:47.352 -> ESP-ROM:esp32c6-20220919
14:28:47.352 -> Build:Sep 19 2022
14:28:47.352 -> rst:0x5 (SLEEP_WAKEUP),boot:0xc (SPI_FAST_FLASH_BOOT)
14:28:47.352 -> SPIWP:0xee
14:28:47.352 -> mode:DIO, clock div:2
14:28:47.352 -> load:0x4086c410,len:0xc64
14:28:47.352 -> load:0x4086e610,len:0x2730
14:28:47.352 -> load:0x40875728,len:0x594
14:28:47.352 -> entry 0x4086c410
14:28:47.352 -> Boot number: 2
14:28:47.352 -> Wakeup caused by timer
14:28:47.352 -> ....Setup ESP32 to sleep for every 10 Seconds
14:28:47.352 -> Going to sleep now
14:28:47.352 -> ESP-ROM:esp32c6-20220919
14:28:47.352 -> Build:Sep 19 2022
14:28:47.352 -> rst:0x5 (SLEEP_WAKEUP),boot:0xc (SPI_FAST_FLASH_BOOT)
14:28:47.352 -> SPIWP:0xee
14:28:47.352 -> mode:DIO, clock div:2
14:28:47.352 -> load:0x4086c410,len:0xc64
14:28:47.352 -> load:0x4086e610,len:0x2730
14:28:47.352 -> load:0x40875728,len:0x594
14:28:47.352 -> entry 0x4086c410
14:28:47.352 -> Boot number: 3
14:28:47.352 -> Wakeup caused by timer
14:28:47.352 -> ....Setup ESP32 to sleep for every 10 Seconds
14:28:47.352 -> Going to sleep now
14:28:47.352 -> ESP-ROM:esp32c6-20220919
14:28:47.352 -> Build:Sep 19 2022
14:28:47.352 -> rst:0x5 (SLEEP_WAKEUP),boot:0xc (SPI_FAST_FLASH_BOOT)
14:28:47.352 -> SPIWP:0xee
14:28:47.352 -> mode:DIO, clock div:2
14:28:47.352 -> load:0x4086c410,len:0xc64
14:28:47.352 -> load:0x4086e610,len:0x2730
14:28:47.352 -> load:0x40875728,len:0x594
14:28:47.352 -> entry 0x4086c410
14:28:47.352 -> Boot number: 4
14:28:47.352 -> Wakeup caused by timer
14:28:47.352 -> ....Setup ESP32 to sleep for every 10 Seconds
14:28:47.352 -> Going to sleep now
14:28:53.510 -> ESP-ROM:esp32c6-20220919
14:28:53.510 -> Build:Sep 19 2022
14:28:53.510 -> rst:0x5 (SLEEP_WAKEUP),boot:0xc (SPI_FAST_FLASH_BOOT)
14:28:53.510 -> SPIWP:0xee
14:28:53.510 -> mode:DIO, clock div:2
14:28:53.510 -> load:0x4086c410,len:0xc64
14:28:53.510 -> load:0x4086e610,len:0x2730
14:28:53.510 -> load:0x40875728,len:0x594
14:28:53.510 -> entry 0x4086c410
14:28:56.573 -> Boot number: 5
14:28:56.573 -> Wakeup caused by timer
14:28:57.140 -> ....Setup ESP32 to sleep for every 10 Seconds
14:28:58.661 -> Going to sleep now
14:29:08.631 -> ESP-ROM:esp32c6-20220919
[... 53 lines cut ...]
14:30:08.155 -> rst:0x5 (SLEEP_WAKEUP),boot:0xc (SPI_FAST_FLASH_BOOT)
14:30:08.155 -> SPIWP:0xee
14:30:08.155 -> mode:DIO, clock div:2
14:30:08.155 -> load:0x4086c410,len:0xc64
14:30:08.155 -> load:0x4086e610,len:0x2730
14:30:08.155 -> load:0x40875728,len:0x594
14:30:08.155 -> entry 0x4086c410
14:30:11.184 -> Boot number: 10
14:30:11.184 -> Wakeup caused by timer
14:30:11.797 -> ...Setup ESP32 to sleep for every 10 Seconds
14:30:12.790 -> Going to sleep now
14:30:22.771 -> ESP-ROM:esp32c6-20220919
14:30:22.771 -> Build:Sep 19 2022
14:30:22.771 -> rst:0x5 (SLEEP_WAKEUP),boot:0xc (SPI_FAST_FLASH_BOOT)
14:30:22.771 -> SPIWP:0xee
14:30:22.771 -> mode:DIO, clock div:2
14:30:22.771 -> load:0x4086c410,len:0xc64
14:30:22.771 -> load:0x4086e610,len:0x2730
14:30:22.771 -> load:0x40875728,len:0x594
14:30:22.771 -> entry 0x4086c410
14:30:25.802 -> Boot number: 11
14:30:25.802 -> Wakeup caused by timer
14:30:26.418 -> ...Setup ESP32 to sleep for every 10 Seconds
14:30:27.411 -> Going to sleep now
```

QuickJack, 2024-06-21:
Thanks for testing this with a different ESP32-C6 HW. It might be a brownout issue. Using a 65W fast charger + a high quality USB cable it happens as well. The LED does not go off for a long time which shows that there are reboots. But I am unable to log the output during this test. 

I am using this board: https://www.dfrobot.com/product-2771.html. It's just a few days old. They claim to use a high quality LDO (https://dfimg.dfrobot.com/5d57611a3416442fa39bffca/wiki/6f630301d84caf0e92266e3c5cf11edc.PDF). According to the datasheet, it supports a maximum current of 1A.

It would be interesting if someone could run the sample code with the same board.

QuickJack, 2024-06-21:
According to https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-reference/kconfig.html#config-esp-brownout-det-lvl-sel there seems to be a way to configure the brownout level for this chip.

Jason2866, 2024-06-21:
Looking in the support section for the board (the page you linked) it looks like the board design has issues.

QuickJack, 2024-06-21:
Where exactly do you see this?

QuickJack, 2024-06-21:
So it's definitely a hw design issue. That's very disappointing.

Thanks for helping to find out the reason behind all these strange problems.

closed 2024-06-24 by SuGlider

XylenC4, 2025-01-19:
I’m using the FireBeetle 2 ESP32-C6 as an SML (mains energy meter) reader with ESPHome and Home Assistant. However, I encountered brownout issues when connecting to Wi-Fi, which was surprising since the device had been working fine for a week with the same firmware.

After investigating with my budget oscilloscope, I noticed significant negative voltage spikes. To address this, I added a 100µF 0805 capacitor to the linear voltage regulator. The size was perfect to skip the middle pin at the linear voltage regulator.  Since then, the device has been working flawlessly without any issues. With my my budget oscilloscope ill did not see any drops anymore.

[image]

TD-er, 2025-01-19:
Between which pins did you see those negative spikes?

XylenC4, 2025-01-19:
3V3 as on the regulator

https://dfimg.dfrobot.com/5d57611a3416442fa39bffca/wiki/65df25004a7d1e8bc128894c75ce4089.pdf