espressif/arduino-esp32 #9920: ESP to ESP transmission error.

Jonmurch, 2024-06-23:
### Board

ESP32 Dev Mod

### Device Description

two Esp32 dev modules. using one as transmitter and one as receiver to transmit 6 Variables. 

### Hardware Configuration

they are both empty boards 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Transmitter Produces this prompt on loop when running.

Sent with success
Last Packet Send Status: Delivery Fail

Receiver Produces this prompt and nothing more

Receiver Setup Started
Receiver MAC Address: 00:00:00:00:00:00
ESP-NOW Initialized
Peer Added

### Sketch

```cpp
My transmitters Code
{
#include <esp_now.h>
#include <WiFi.h>

// Structure to send data
typedef struct struct_message {
    int base;
    int lowest;
    int bottomMiddle;
    int topMiddle;
    int claw;
    int wrist;
} struct_message;

// Create a struct_message to send data
struct_message myData;

// MAC Address of the receiver ESP32
uint8_t broadcastAddress[] = {0x10, 0x06, 0x1C, 0xB5, 0xAF, 0x96};

// Callback when data is sent
void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
    Serial.print("Last Packet Send Status: ");
    Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
}

void setup() {
    // Initialize Serial Monitor
    Serial.begin(115200);
    Serial.println("Transmitter Setup Started");  // Debug message

    // Print own MAC address
    Serial.print("Transmitter MAC Address: ");
    Serial.println(WiFi.macAddress());  // Print the transmitter's MAC address

    // Set device as a Wi-Fi Station
    WiFi.mode(WIFI_STA);

    // Init ESP-NOW
    if (esp_now_init() != ESP_OK) {
        Serial.println("Error initializing ESP-NOW");
        return;
    }
    Serial.println("ESP-NOW Initialized");  // Debug message

    // Register the send callback
    esp_now_register_send_cb(OnDataSent);

    // Register the peer
    esp_now_peer_info_t peerInfo;
    memcpy(peerInfo.peer_addr, broadcastAddress, 6);
    peerInfo.channel = 0;  
    peerInfo.encrypt = false;

    // Add peer        
    if (esp_now_add_peer(&peerInfo) != ESP_OK) {
        Serial.println("Failed to add peer");
        return;
    }
    Serial.println("Peer Added");  // Debug message

    // Initialize servo values
    myData.base = 90;
    myData.lowest = 90;
    myData.bottomMiddle = 90;
    myData.topMiddle = 90;
    myData.claw = 90;
    myData.wrist = 90;
}

void loop() {
    // Send message via ESP-NOW
    esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &myData, sizeof(myData));

    if (result == ESP_OK) {
        Serial.println("Sent with success");
    } else {
        Serial.println("Error sending the data");
    }

    delay(100); // Minimal delay between sets of data
}
}

My receivers code

#include <esp_now.h>
#include <WiFi.h>

// Structure to receive data
typedef struct struct_message {
    int base;
    int lowest;
    int bottomMiddle;
    int topMiddle;
    int claw;
    int wrist;
} struct_message;

// Create a struct_message to hold the incoming data
struct_message incomingData;

// Callback when data is received
void OnDataRecv(const esp_now_recv_info *info, const uint8_t *data, int len) {
    Serial.println("Data received callback triggered");  // Debug message
    memcpy(&incomingData, data, sizeof(incomingData));
    Serial.print("Received data from: ");
    char macStr[18];
    snprintf(macStr, sizeof(macStr), "%02x:%02x:%02x:%02x:%02x:%02x", info->src_addr[0], info->src_addr[1], info->src_addr[2], info->src_addr[3], info->src_addr[4], info->src_addr[5]);
    Serial.println(macStr);
    Serial.print("Base: "); Serial.println(incomingData.base);
    Serial.print("Lowest: "); Serial.println(incomingData.lowest);
    Serial.print("Bottom Middle: "); Serial.println(incomingData.bottomMiddle);
    Serial.print("Top Middle: "); Serial.println(incomingData.topMiddle);
    Serial.print("Claw: "); Serial.println(incomingData.claw);
    Serial.print("Wrist: "); Serial.println(incomingData.wrist);
}

void setup() {
    // Initialize Serial Monitor
    Serial.begin(115200);
    Serial.println("Receiver Setup Started");  // Debug message

    // Print own MAC address
    Serial.print("Receiver MAC Address: ");
    Serial.println(WiFi.macAddress());  // Print the receiver's MAC address

    // Set device as a Wi-Fi Station
    WiFi.mode(WIFI_STA);

    // Init ESP-NOW
    if (esp_now_init() != ESP_OK) {
        Serial.println("Error initializing ESP-NOW");
        return;
    }
    Serial.println("ESP-NOW Initialized");  // Debug message

    // Register the receive callback
    esp_now_register_recv_cb(OnDataRecv);

    // Optionally, you can add the transmitter as a peer
    // This is not required for receiving data, but useful for bidirectional communication
    esp_now_peer_info_t peerInfo;
    uint8_t peerAddress[] = {0xCC, 0x7B, 0x5C, 0x36, 0xD8, 0xFE};
    memcpy(peerInfo.peer_addr, peerAddress, 6);
    peerInfo.channel = 0;
    peerInfo.encrypt = false;

    // Add peer
    if (esp_now_add_peer(&peerInfo) != ESP_OK) {
        Serial.println("Failed to add peer");
        return;
    }
    Serial.println("Peer Added");  // Debug message
}

void loop() {
    // Nothing to do here
    // Data will be received via the callback
}
```

### Debug Message

```plain
It seems that the two boards are connected but the receiver is failing to process anything. Any help would be appreciated.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-23:
@P-R-O-C-H-Y , It is about ESP-NOW, PTAL. Thanks.

Jonmurch, 2024-06-24:
@P-R-O-C-H-Y any input would be appreciated. Never used ESP-NOW so im hoping im on the right track here.

lbernstone, 2024-06-24:
There are Arduino libraries for [ESP-NOW](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP_NOW/examples).  You should try them out.  
The ESP-IDF functions are [documented here](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32s3/api-reference/network/esp_now.html).
The [ESP-IDF example is here](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/wifi/espnow)

P-R-O-C-H-Y, 2024-06-25:
Hi @Jonmurch, as Ibernstone commented you can use our ESP-NOW Arduino library instead of using ESP-IDF APIs.
We also have few ready-to-go [examples](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP_NOW/examples), so you can start with them. 
For more informations about ESP-NOW library APIs you can take a look to the [documentation](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/espnow.html).

closed 2025-01-15 by Parsaabasi