espressif/arduino-esp32 #9924: WifiClientSecure - Connection error after 3600 seconds

flamy0, 2024-06-23:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

Plain module

### Hardware Configuration

External wifi antenna connected

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When the code is running for exactly one hour (3600 sec.) the code is not able to connect to the server anymore.

I've noticed that at the last print out of the received data a single not readable char is added and the heap in that moment jumps from somewhere around 210kB up to 260kB.

The error is reproducable each time the code is running. No exceptions from faulty behaviour.

### Sketch

```cpp
#pragma warning(disable : 4996)

#define REFRESHTIME_SHORT      3
#define REFRESHTIMEUNIT_SHORT  0
#define REFRESHTIME_LONG      10
#define REFRESHTIMEUNIT_LONG   0

#ifndef STASSID
  #define STASSID "SSID"                        // set your SSID
  #define STAPSK  "PASS"                        // set your wifi password
#endif

#include <HTTPClient.h>
#include <SPI.h>
#include <string.h>
#include <time.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>

// Initialize the Wifi client library
WiFiClientSecure client;

enum timeUnits { SEC, MIN, HOUR };

enum timeUnits refreshTimeUnit = (timeUnits)REFRESHTIMEUNIT_SHORT;
uint8_t refreshTime = REFRESHTIME_SHORT;

time_t nextRefresh = 0;

struct tm tm;
time_t now;
uint64_t t_ms_1000 = 0;

// string used for the commands parsing
char* c_clientBufferString = NULL;

// server address:
const char *server = "www.httpvshttps.com";

const char *magicCookie = "https://www.httpvshttps.com/";

const char *test_root_ca = \
"-----BEGIN CERTIFICATE-----\n" \
"-----END CERTIFICATE-----\n";

// this function prints the current time
void printTime(time_t now)
{
  Serial.print("Timestamp:");Serial.println(now);
}

void httpConnect()
{
  // Set Certificate
  //client.setCACert(test_root_ca);
  client.setInsecure();

  if (client.connect(server, 443)) {
      Serial.print("Connected to \""); Serial.print(server); Serial.println("\"");
  }
}

uint8_t getRefreshTime() {
  return refreshTime;
}

enum timeUnits getRefreshTimeUnit() {
  return refreshTimeUnit;
}

void appendCharToBuffer(char** buffer, size_t* length, char c) {
    // Speicher um eins vergrößern
    *buffer = (char *)realloc(*buffer, *length + 1 + 1); // +1 für das neue Zeichen, +1 für Nullterminator
    if (*buffer == NULL) {
        fprintf(stderr, "Fehler bei der Speicherzuweisung\n");
        exit(1);
    }

    // Zeichen anhängen
    (*buffer)[*length] = c;
    (*buffer)[*length + 1] = '\0'; // Nullterminator setzen
    (*length)++;
}

void httpRequest()
{
  size_t buffer_length = 0;

  // if there's a successful connection:
  if (client.connected()) {
      // send the HTTP PUT request:
      client.print("GET ");
      client.print(magicCookie);
      client.println(" HTTP/1.1");
      //client.println("Host: ");
      //client.println(server);
      client.println("Connection: keep-alive");
      client.println();

      while (client.available()) {
          char c = client.read();
          appendCharToBuffer(&c_clientBufferString, &buffer_length, c);
      }

      Serial.println("Content:");Serial.println(c_clientBufferString);Serial.println("...end!");

      setCalendarRefreshRate(REFRESHTIME_LONG, (timeUnits)REFRESHTIMEUNIT_LONG);
  }
  else {
    unsigned long timestamp = millis();

    // if you couldn't make a connection:
    Serial.println("Reconnecting...");

    while (!client.connected()) {
      if (millis() >= timestamp + 5000) {
        Serial.println("Connection timeout!");

        client.clear();
        client.stop();

        httpConnect();

        break;
      }
    }

      setCalendarRefreshRate(REFRESHTIME_SHORT, (timeUnits)REFRESHTIMEUNIT_SHORT);
  }
}

void setCalendarRefreshRate(uint8_t _refreshTime, enum timeUnits _refreshTimeUnit) {
  refreshTime = _refreshTime;
  refreshTimeUnit = _refreshTimeUnit;
}

// this function refreshes the calendar
void calendarRefresh(time_t now)
{
  httpRequest();
  
  if (getRefreshTimeUnit() == HOUR) {
      nextRefresh = now + (getRefreshTime() * 60 * 60);
  }
  else if (getRefreshTimeUnit() == MIN) {
      nextRefresh = now + (getRefreshTime() * 60);
  }
  else if (getRefreshTimeUnit() == SEC) {
      nextRefresh = now + getRefreshTime();
  }

  Serial.print("Next refresh: ");Serial.println(nextRefresh);
}

// this function checks if the calendar has to be refreshed
void checkCalendarRefresh(time_t now)
{
    if (nextRefresh <= now) {
        calendarRefresh(now);
    }
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  time(&now);

  // start network
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(STASSID, STAPSK);
  while (WiFi.status() != WL_CONNECTED) {
    delay(200);
    Serial.print(".");
  }
  Serial.println("\nWiFi connected");

  httpConnect();

  calendarRefresh(now);
}

void loop() {
  if (millis() > t_ms_1000) {
    time(&now);

    printTime(now);

    Serial.print("freeHeap:");Serial.println(ESP.getFreeHeap());

    t_ms_1000 = millis() + 1000;
  }

  checkCalendarRefresh(now);

  if (c_clientBufferString != NULL) {
    free(c_clientBufferString);
    c_clientBufferString = NULL;
  }
}
```

### Debug Message

```plain
Trying to add debug message later.
```

### Other Steps to Reproduce

Ported the code to Visual Studio 2022 C Code using libcurl for https connection and it runs flawless for days.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-23:
Maybe @me-no-dev can quickly spot something in the code...

me-no-dev, 2024-06-23:
Please simplify the code to reproduce and make sure that we can run it. There is nothing i the client source to cause any such thing.

1. Does it do the same thing if you use another host?
2. How about opening a new client for each request?

SuGlider, 2024-06-23:
I think that the problem may related to `Connection: keep-alive`, the server may keep it alive for about 1 hour and after that it shall close the connection, causing the issue you have reported.

Why don't just close it and open it again every 10 seconds as done in the sketch?

JAndrassy, 2024-06-24:
try this quick fix in NetworkClient.cpp in `int NetworkClientSecure::available() {` replace `return peeked ? peeked : res;` with `return peeked`.
it will allow your sketch to detect the disconnection. now it attempts to read a closed connection 

the fix is from my PR https://github.com/espressif/arduino-esp32/pull/9869 and it will be in the next release.

holgerlembke, 2024-06-24:
Made my day, @JAndrassy 

I got:

```
[HTTPS] GET: https://cataas.com/cat httpCode: 200 Size: 737830
[HTTPS] GET: https://cataas.com/cat failed with httpCode -1 = connection refused
```
This error goes away with core debug level: verbose and /me was scratching my head , how to debug that.

Applied the above fix and all good now!

Thanks.

SMi1984DE, 2024-06-24:
It might be the same issue as in #9712, where the hostname couldn't be resolved anymore after automatically switching to IPv6. I experienced a similar problem with connections dropping after about half an hour to two hours, leading to a loop that required a reset. For more details, see my comment in https://github.com/espressif/arduino-esp32/issues/9712#issuecomment-2186724477_

flamy0, 2024-06-28:
That seems to be the case. Meanwhile I was able to identify the problem in the DNS area, too.

I started a different sample code by using the WifiClientSecure exapmle. First notice: It does not use "WifiClientSecure"-class, but "NetworkClientScure"-class instead. Using this shortened code, the errors came much earlier and the debug messages pointed directly to a problem with resolving the DNS name. 

Then, I changed the host name to the assigned IP address and voila, code ran in an endless loop without any problems. Changed my main code to the same approach and it worked flawless.

I'll try to enable IPv6 and see what happens.

SuGlider, 2024-06-28:
@me-no-dev - Please take a look into the previous 2 comments. It may be something to test.

cziter15, 2024-06-30:
I'm experiencing an issue which may be related to this.

I have smarthome infrastructure built on top of esp8266 and esp32 in my parents house and one esp32s3 in my home.
These devices are bridged together with dynamic dns.

When I disable my router/FFTH device, remote devices based on esp8266 are able to instantly reconnect, just after populating DNS record, while for ESP32 device it takes few hours.

flamy0, 2024-06-30:
It seems that "WiFi.enableIPv6(true);" did the trick:
"[7627708][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:40xx:xxx::xxxx"

Running for two hours now, without any loss of connection.

SMi1984DE, 2024-06-30:
Basically, it just circumvents the actual problem. Maybe the underlying error will be found next time. It’s also possible that manufacturers are slowly pushing for IPv6 to drive IoT, of course. ;) Well, for me, this interim solution is workable. I don't plan to produce my project in large quantities and I’m less worried about potential device failures only because customers having IPv6 enabled.

closed 2024-07-01 by flamy0