espressif/arduino-esp32 #9927: Add support for LILYGO T3S3 - ESP32S3 with SX1262 lora RF module

joyel24, 2024-06-24:
### Related area

Board Support

### Hardware specification

 LILYGO T3S3

### Is your feature request related to a problem?

.

### Describe the solution you'd like

.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

joyel24, 2024-06-24:
@Xinyuan-LilyGO @LilyGO

joyel24, 2024-06-24:
> Hello, this is something which needs to be done by Community or board vendor.

Of course, this is why I mentioned them in the comment above 👍

joyel24, 2024-06-28:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/179#issuecomment-2196027777
It's probably coming thanks to @lewisxhe

lyusupov, 2024-07-03:
@joyel24  

https://github.com/espressif/arduino-esp32/pull/9956

[image]

joyel24, 2024-07-03:
@lyusupov 
Yes, I have seen it seems to be fixed but did't test it yet. I wanted to wait for the next release and test it with arduino ide to see if it works well before closing the issue..;

lyusupov, 2024-07-17:
@joyel24 
> ... I wanted to wait for the next release ...  before closing the issue...

[image]

closed 2024-10-21 by joyel24