espressif/arduino-esp32 #9936: SPI Flash Mapping Fails on 3.0.2

brentru, 2024-06-25:
### Board

Adafruit ESP32-S3 Reverse TFT Feather 

### Device Description

Adafruit ESP32-S3 Reverse TFT Feather - 4MB Flash, 2MB PSRAM, STEMMA QT

### Hardware Configuration

Nothing is connected to the board

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

QIO 80MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

Adafruit ESP32-S3 Reverse TFT Feather (w/TinyUF2 (4MB partition scheme)) fails to load the Blink sketch. Instead of loading Blink, it appears as a BOOT drive. The debug logs 

The core issue stems from `esp_mmu_unmap` and `spi_flash_munmap` emitting failures:
```
11:20:25.849 -> [  1087][I][esp32-hal-psram.c:90] psramInit(): PSRAM enabled
11:20:25.849 -> E (1341) mmap: esp_mmu_unmap(595): munmap target pointer is outside external memory regions
11:20:25.849 -> 
11:20:25.883 -> assert failed: spi_flash_munmap flash_mmap.c:241 (0 && "invalid handle, or handle already unmapped")
11:20:25.883 -> 
```

### Sketch

```cpp
Generic Arduino Blink Sketch w/TinyUF2 (4MB partition scheme)

/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
[... 159 lines cut ...]
11:20:27.124 -> E (1179) esp_core_dump_common: Core dump write failed with error=-1
11:20:27.162 -> Rebooting...
```

Backtrace:
```

Decoding stack results
0x40376dce: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 466
0x4037b1dd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 84
0x40380f81: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 81
0x4200b363: spi_flash_munmap at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_mmap.c line 241
0x4200ae99: esp_partition_munmap at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_partition/partition_target.c line 173
0x4200af9d: read_otadata at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/app_update/esp_ota_ops.c line 97
0x4200b237: esp_ota_get_state_partition at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/app_update/esp_ota_ops.c line 836
0x42002699: initArduino at /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/esp32-hal-misc.c line 262
0x42003b63: app_main() at /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/main.cpp line 104
0x4202fb6d: main_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/app_startup.c line 208
0x4037deca: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 162
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-25:
cc @hathach @ladyada

me-no-dev, 2024-06-25:
@brentru @ladyada is this one particular board or all such boards? OTA partition is not overwritten?

Code that breaks is part of OTA Rollback support and is trying to get the OTA Data partition to read the state of the partition that is trying to boot. This is nothing special and has worked fine since was added. Can you try with earlier core version? We merged some IDF changes in 3.0.2 that were not in previous 3.0.x

brentru, 2024-06-25:
> is this one particular board or all such boards?

This is only on the Feather ESP32-S3 Reverse TFT. 

> OTA partition is not overwritten?

I am unsure if the OTA partition is not overwritten. Could you suggest a method of verifying this?

> Can you try with earlier core version? We merged some IDF changes in 3.0.2 that were not in previous 3.0.x

Sure, I'll try with 3.0.0 and 3.0.1 and provide info from trying this in a follow-up comment.

me-no-dev, 2024-06-25:
I'll try to replicate tomorrow somehow. Any pointers on using a regular S3 would be welcome. I also notice that the bootloader is still from 2.0.x (IDF 4.4). Not sure if this is connected, but a thing to look into.

me-no-dev, 2024-06-27:
I can confirm the issue. It's not just that board, but any S3 with enabled QSPI PSRAM. No PSRAM or OPI PSRAM works fine. 3.0.1 works also as expected, so issue was recently introduced into IDF release/v5.1 branch. Will track this to be fixed before we release another version

me-no-dev, 2024-06-27:
We triggered this bug by implementing [this feature request](https://github.com/espressif/arduino-esp32/issues/9808).

I have found that it's necessary to have all of the following conditions to trigger it:
1. ESP32-S3 NxR2
2. OTA partition scheme used
3. COREDUMP to Flash to be enabled
4. QPI SPIRAM to be initialized in app_main
5. App RollBack enabled and checked after PSRAM is initialized

Checking for rollback before initializing PSRAM also works, so that is probably one solution to try. Still unsure if this is not going to cause other problems with OTA.

me-no-dev, 2024-06-28:
Bug has been found by the IDF team and fix has been proposed.
@brentru can you please replace your libs with the ones from [this build](https://github.com/espressif/esp32-arduino-lib-builder/actions/runs/9710204792/artifacts/1648096293) and see if issue is resolved. Would also really appreciate if you test other boards/chips.

brentru, 2024-06-28:
> @brentru can you please replace your libs with the ones from [this build](https://github.com/espressif/esp32-arduino-lib-builder/actions/runs/9710204792/artifacts/1648096293) and see if issue is resolved. Would also really appreciate if you test other boards/chips.

Sure - where should I put these libraries?

me-no-dev, 2024-07-01:
> where should I put these libraries?

Depends how you use the core. If it's from Git, then it will be in `../Arduino/hardware/espressif/esp32/tools/esp32-arduino-libs` folder. If you are using the boards manager, then it will be in `../Arduino15/packages/esp32/tools/esp32-arduino-libs` folder. Where the dots are depends on which OS you are running on.

brentru, 2024-07-01:
@me-no-dev  I re-tested the example sketch (Blink) with the same hardware and inspected the IDF output. It's functioning correctly and the PSRAM is initializing successfully, too:
`[  1087][I][esp32-hal-psram.c:90] psramInit(): PSRAM enabled`

> Would also really appreciate if you test other boards/chips.
Okay! I tested Blink on the following
* Adafruit Feather ESP32-S3 Rev TFT Feather
* Adafruit QT Py ESP32-S3
* Adafruit QT Py ESP32

me-no-dev, 2024-07-01:
Thanks @brentru! will merge the fix then :)

UnexpectedMaker, 2024-07-04:
I stumbled on this the other day. Glad it's fixed... any chance we can get a 3.0.3 out asap?
It effects every one of my S3 boards :(

me-no-dev, 2024-07-05:
3.0.3 will be out soon. we are looking in other issues too.

UnexpectedMaker, 2024-07-05:
> 3.0.3 will be out soon. we are looking in other issues too.

I know - but 2 years is a long time to wait for no progress ;)

I do appreciate the teams work, and I know the work on 3.x with such large IDF changes under the hood has been a `challenge` but this issue is really painful and trips so many people over.

me-no-dev, 2024-07-05:
I understand this might look trivial to you, but it's not. (the reboot issue, not what's this thread about). There are some changes in the USB api that we can try and see if change any behavior, but you need to understand that we do not say that the outcome will be what you expect.

VojtechBartoska, 2024-07-17:
will be fixed when releasing 3.0.3 version.

Closed by https://github.com/espressif/esp32-arduino-lib-builder/pull/188

VojtechBartoska, 2024-07-18:
3.0.3 was released, closing this ticket, if needed you can reopen.

closed 2024-07-18 by VojtechBartoska