espressif/arduino-esp32 #9937: Some Weird Issues with ESP32-S3-WROOM-1-N8R8 on a Customized Board. Error Messages Popped Up After Unplugging and Plugging the USB Cord

zyan003, 2024-06-25:
### Board

Wareshare ESP32-S3-Touch-LCD-7

### Device Description

Wareshare ESP32-S3-Touch-LCD-7 with ESP32-S3-WROOM-1-N8R8 on board.
[image]

### Hardware Configuration

```
/*
 * SPDX-FileCopyrightText: 2023-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

// *INDENT-OFF*

/* Set to 1 if using a custom board */
#define ESP_PANEL_USE_CUSTOM_BOARD       (1)         // 0/1

#if ESP_PANEL_USE_CUSTOM_BOARD

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////////////// Please update the following macros to configure the LCD panel /////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 1 when using an LCD panel */
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#if ESP_PANEL_USE_LCD
/**
 * LCD Controller Name. Choose one of the following:
 *      - GC9A01, GC9B71, GC9503
 *      - ILI9341
 *      - NV3022B
 *      - SH8601
 *      - SPD2010
 *      - ST7262, ST7701, ST7789, ST7796, ST77916, ST77922
 */
#define ESP_PANEL_LCD_NAME          ST7262

/* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (800)
#define ESP_PANEL_LCD_HEIGHT        (480)

/* LCD Bus Settings */
/**
 * If set to 1, the bus will skip to initialize the corresponding host. Users need to initialize the host in advance.
 * It is useful if other devices use the same host. Please ensure that the host is initialized only once.
 */
#define ESP_PANEL_LCD_BUS_SKIP_INIT_HOST    (1)     // 0/1
/**
 * LCD Bus Type. Choose one of the following:
 *      - ESP_PANEL_BUS_TYPE_I2C (not ready)
 *      - ESP_PANEL_BUS_TYPE_SPI
 *      - ESP_PANEL_BUS_TYPE_QSPI
 *      - ESP_PANEL_BUS_TYPE_I80 (not ready)
 *      - ESP_PANEL_BUS_TYPE_RGB (only supported for ESP32-S3)
 */
#define ESP_PANEL_LCD_BUS_TYPE      (ESP_PANEL_BUS_TYPE_RGB)
/**
 * LCD Bus Parameters.
 *
 * Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html and
 * https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/index.html for more details.
 *
[... 437 lines cut ...]
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
LVGL porting example start
Initialize panel device
E (30) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (30) GT911: touch_gt911_read_cfg(362): GT911 read error!
E (31) GT911: esp_lcd_touch_new_i2c_gt911(128): GT911 init failed
E (37) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
Initialize LVGL
Create UI
LVGL porting example end
IDLE loop
IDLE loop
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-25:
>E (30) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (30) GT911: touch_gt911_read_cfg(362): GT911 read error!
E (31) GT911: esp_lcd_touch_new_i2c_gt911(128): GT911 init failed
E (37) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!

This is about the Touch Screen. It seems like the I2C pins are not being set correctly.

SuGlider, 2024-06-25:
>#define ESP_PANEL_USE_BACKLIGHT         (0)         // 0/1

Is it correct? The display may fail 9 of 10 times, after first boot, because there is no Back Light on in order to see the display....

zyan003, 2024-06-25:
@SuGlider, thanks for the quick response. I will check `#define ESP_PANEL_USE_BACKLIGHT` to see if that is the issue.

When I received the board, I used esptool to backup the factory firmware with the LVGL demo. With that "factory" firmware, I can upload the customized LVGL project via Arduino and the LCD display works flawlessly with customized LVGL content. Then if I unplug the USB and reconnect the board to either PC or external power source, the display shows nothing and those error messages pop up. I think I see some backlight from the display.

SuGlider, 2024-06-25:
If I understand it correctly, the original firmware that came installed in the display works fine.

But when you upload the Arduino Sketch presented in this issue, it doesn't work, with a back display and error messages.

If this is correct, the problem is in the Arduino Sketch code.

zyan003, 2024-06-25:
No. If I restore the original firmware to the board and then upload my Arduino sketch, the display works fine. However, if I unplug the USB and reconnect it, the board stops working.

I understand this issue might seem unusual, and I apologize if it's difficult to explain.

SuGlider, 2024-06-25:
There is a documentatio and instructions in https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-7

It says that Flash is QIO 120MHz to align with the PSRAM. Your setup says QIO 80MHz.

SuGlider, 2024-06-25:
Check the Notes with the IDF configuration. Maybe Arduino can't be used as written there:
CONFIG_IDF_EXPERIMENTAL_FEATURES=y and CONFIG_SPIRAM_SPEED_120M=y [should align with PSRAM]

```
CONFIG_FREERTOS_HZ=1000
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_120M=y [should align with PSRAM]
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_IDF_EXPERIMENTAL_FEATURES=y and CONFIG_SPIRAM_SPEED_120M=y [should align with PSRAM]
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
CONFIG_ESP32S3_DATA_CACHE_LINE_64B=y
CONFIG_COMPILER_OPTIMIZATION_PERF=y
#The following LVGL configuration can improve the frame rate (LVGL v8.3):
#define LV_MEM_CUSTOM 1 or CONFIG_LV_MEM_CUSTOM=y
#define LV_MEMCPY_MEMSET_STD 1 or CONFIG_LV_MEMCPY_MEMSET_STD=y
#define LV_ATTRIBUTE_FAST_MEM IRAM_ATTR or CONFIG_LV_ATTRIBUTE_FAST_MEM=y
```

SuGlider, 2024-06-25:
@zyan003 - I think that, in order to use Arduino, you will have to build using Arduino as IDF Component, instead of using the Arduino IDE with specific/fixed IDF setup.

zyan003, 2024-06-26:
Now, with all parameters remain unchanged, if I hold the "RESET" button when powering up, then open serial monitor, then press "RESET" again, LCD starts showing content and the touch works fine as well. The program works now, but it is not yet a flawless solution.

Jason2866, 2024-06-27:
The Device has an UART port too. Changing the setup will probably solve the strange behaviour.

SuGlider, 2024-06-27:
What Arduino Core version is being used?

SuGlider, 2024-06-27:
You can also try building it with the Arduino Core 2.0.17, just in case.

zyan003, 2024-06-27:
I found out that the issue is caused by GT911 power up timing and delay. Now, after adding
```
    ESP_IOExpander *expander = new ESP_IOExpander_CH422G((i2c_port_t)I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS_000, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);
    expander->init();
    expander->begin();
    expander->multiPinMode(TP_RST | LCD_BL | LCD_RST | SD_CS | USB_SEL, OUTPUT);
    expander->multiDigitalWrite(TP_RST | LCD_BL | LCD_RST, HIGH);
    delay(100);
    expander->multiDigitalWrite(TP_RST | LCD_RST, LOW);
    delay(100);
    digitalWrite(GPIO_INPUT_IO_4, LOW);
    delay(100);
    expander->multiDigitalWrite(TP_RST | LCD_RST, HIGH);
    delay(200);
```
in `void main()`, the display works flawlessly.

closed 2024-06-27 by zyan003