espressif/arduino-esp32 #9943: WifiProv BLE doesn't auto connect Wifi after initial provisioning

mhendriks, 2024-06-27:
### Board

ESP32-S3-DEV-Module

### Device Description

Start at #8760  the WifiProv example doesn't work properly.
After the initial BLE provisioning the Wifi isn't setup and the proces stops.

Before this commit (BIG Network refactoring) it is working fine.

### Hardware Configuration

-

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

macos latest

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

After successful provisioning the wifi  connection isn't started. 
Only a reboot is or disconnect+start will setup the Wifi connection. 

### Sketch

```cpp
// simplified example

#include "WiFiProv.h"
#include "WiFi.h"

// #define USE_SOFT_AP // Uncomment if you want to enforce using the Soft AP method instead of BLE
const char *pop = "abcd1234";           // Proof of possession - otherwise called a PIN - string provided by the device, entered by the user in the phone app
const char *service_name = "PROV_TEST";  // Name of your device (the Espressif apps expects by default device name starting with "Prov_")
const char *service_key = NULL;         // Password used for SofAP method (NULL = no password needed)
bool reset_provisioned = false;          // When true the library will automatically delete previously provisioned data.

// WARNING: SysProvEvent is called from a separate FreeRTOS task (thread)!
void SysProvEvent(arduino_event_t *sys_event) {
  switch (sys_event->event_id) {
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
      Serial.print("\nConnected IP address : ");
      Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
      break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED: Serial.println("\nDisconnected. Connecting to the AP again... "); break;
    case ARDUINO_EVENT_PROV_START:            Serial.println("\nProvisioning started\nGive Credentials of your access point using smartphone app"); break;
    case ARDUINO_EVENT_PROV_CRED_RECV:
    {
      Serial.println("\nReceived Wi-Fi credentials");
      Serial.print("\tSSID : ");
      Serial.println((const char *)sys_event->event_info.prov_cred_recv.ssid);
      Serial.print("\tPassword : ");
      Serial.println((char const *)sys_event->event_info.prov_cred_recv.password);
      break;
    }
    case ARDUINO_EVENT_PROV_CRED_FAIL:
    {
      Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
      if (sys_event->event_info.prov_fail_reason == WIFI_PROV_STA_AUTH_ERROR) {
        Serial.println("\nWi-Fi AP password incorrect");
      } else {
        Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");
      }
      break;
    }
    case ARDUINO_EVENT_PROV_CRED_SUCCESS: Serial.println("\nProvisioning Successful"); break;
    case ARDUINO_EVENT_PROV_END:          Serial.println("\nProvisioning Ends");       break;
    default:                              break;
  }
}

void setup() {
  Serial.begin(115200);
  WiFi.onEvent(SysProvEvent);

  Serial.println("Begin Provisioning using BLE");
  WiFiProv.beginProvision(
    WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name, service_key, NULL, reset_provisioned
  );

}

void loop() {}
```

### Debug Message

```plain
Received Wi-Fi credentials
	SSID : <ssid>
	Password : <pw>

Provisioning Successful

Provisioning Ends
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-27:
@mhendriks - add this line at the end of `setup()`: `  WiFi.begin();`
It will work.

mhendriks, 2024-06-27:
@SuGlider thanks for the hint. 
I tried with the WiFi.begin() but that didn't always work.
I solved it via the event below.

```
case ARDUINO_EVENT_PROV_END:
        DebugTln("WifiProv: Provisioning Ends");
        ESP.restart();
        break;

```

BTW In 3.0.2 the WifiProv works even less well.

SuGlider, 2024-06-27:
I'm testing it here with 3.0.2.
The PR #9946  works fine. No need to reset.
Using the Android `ESP BLE Prov` v2.1.4 and it fails sometimes to get the list of WiFi SSIDs.
But if I enter it manually in the APP, everything works fine. 

If there is no `WiFi.begin()` the connection never happens.

You say that it doesn't always work.
Can you, please, detail the step and the issue you find?

mhendriks, 2024-06-27:
thanks for the clarification. 
Hmmm perhaps I didn't test the WiFi.begin() in combination with the 3.0.2.
I will retest it.

SuGlider, 2024-06-27:
I also set the Arduino IDE option `Erase All Flash Before Sketch Upload` to `Enable` in order to make sure that the NVS data will be erased and a enforce a new Provisioning cycle.

mhendriks, 2024-06-27:
We are sharing the same thoughts ... use `esptool.py erase_flash` before a test run ;-)

I did a couple of tests with the 3.0.2. 
Confirm the WiFi.begin() is the fix!!!

The WiFi.begin() should be called before the WifiProv in order to work properly. With the 3.0.1 the WiFi.begin() could be after the WifiProv but with the 3.0.2 this should be before.

Many thanks for the solution / hints and swift reply!

mhendriks, 2024-06-27:
see #9946

PR #9946 (2024-06-28): Fixes WiFiProv.ino
## Description of Change
Fixes WiFi Provisioning example to work with the new Network layer.
Adds a `WiFi.begin()` for automatic connection after provisioning or after reading NVS.
Changed the Provisioning mode to BLE in order to avoid a memory releasing problem with ESP32 after provisioning.

## Tests scenarios
Tested with S3 and ESP32 using the example from 
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino

## Related links
Fix #9943
closed 2024-06-27 by mhendriks