espressif/arduino-esp32 #9945: ESP_12S new library Slave mode clarification.

raprakashvi, 2024-06-27:
### Board

ESP32 Dev

### Device Description

Plain module with I2S elements

### Hardware Configuration

NAN

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi, I am going through documentation to use the two I2S ports available and need to use one of them in Slave and saw this documentation on the new library page:

"I2S Configuration
Master / Slave Mode
In Master mode (default), the device generates clock signal sck and word select signal on ws.

In Slave mode the device listens on attached pins for the clock signal and word select - i.e. unless externally driven the pins will remain LOW. **This mode is not supported yet.**"

Is the Slave mode still being supported in the new library? Thanks.

### Sketch

```cpp
NAN
```

### Debug Message

```plain
NAN
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-27:
It is supported as long as the ADC provides all needed signals (BCK, WS, DIN)

me-no-dev, 2024-06-27:
cracks and pops are probably due to wrong configurations. You need to provide more info on your setup if you want us to be able to further help

closed 2024-06-27 by me-no-dev

raprakashvi, 2024-06-30:
Sure. I am describing the setup below.  The cracking noise has been solved by better wiring but I notice an inconsistency in recording. I am using ESPWroom 32 module with 4 MB flash,

The recordings, though of correct time length differ in file size between iterations (or multiple tests) [they are the same between them everytime though]. This points to data loss. Also one of the mic set is much clear and the other seems a bit damped sound. 

The SCK and WS pins of all mics are shared and each of the SD pin pairs have 100 k resistor as per the schematic. Thanks.

`#include "ESP_I2S.h"
#include "FS.h"
#include "SD_MMC.h"

#define ONE_BIT_MODE true

// Define common I2S pins for both sets of microphones
const uint8_t I2S_SCK = 21;
const uint8_t I2S_WS = 22;

// Define I2S pins for first set of microphones
const uint8_t I2S_DIN1 = 25;

// Define I2S pins for second set of microphones
const uint8_t I2S_DIN2 = 19;

// Define SD card pins
const uint8_t SD_CMD = 15;
const uint8_t SD_CLK = 14;
const uint8_t SD_DATA0 = 2;

#define SAMPLE_RATE 16000
#define BITS_PER_SAMPLE I2S_DATA_BIT_WIDTH_32BIT
#define CHANNELS I2S_SLOT_MODE_STEREO
#define BUFFER_SIZE 1024  // Reduced buffer size to save stack space
#define RECORD_DURATION 10  // Record duration in seconds

I2SClass i2s1;
I2SClass i2s2;
const char *audioFileName1 = "/audio_mics1.wav";
const char *audioFileName2 = "/audio_mics2.wav";

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    delay(10);
  }

  // Initialize SD card
  pinMode(2, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(13, INPUT_PULLUP);
  pinMode(15, INPUT_PULLUP);

  Serial.println("Initializing SD card...");
  if (!SD_MMC.setPins(SD_CLK, SD_CMD, SD_DATA0)) {
    Serial.println("Failed to set SD pins!");
    return;
  }

  if (!SD_MMC.begin("/sdcard", ONE_BIT_MODE)) {
    Serial.println("SD Card initialization failed!");
    return;
  }

  Serial.println("SD card initialized.");

  // Create or open the files for writing audio data
  File audioFile1 = SD_MMC.open(audioFileName1, FILE_WRITE);

  File audioFile2 = SD_MMC.open(audioFileName2, FILE_WRITE);

  // Initilize I2S for first set of microphones
  i2s1.setPins(I2S_SCK, I2S_WS, -1, I2S_DIN1);

  // Initialize I2S for second set of microphones
  i2s2.setPins(I2S_SCK, I2S_WS, -1, I2S_DIN2);
 

  // Configure I2S RX transformation for both sets of microphones
  if (!i2s1.configureRX(SAMPLE_RATE, BITS_PER_SAMPLE, CHANNELS, I2S_RX_TRANSFORM_32_TO_16)) {
    Serial.println("Failed to configure I2S RX transformation for first set of mics!");
    return;
  }
  
  if (!i2s2.configureRX(SAMPLE_RATE, BITS_PER_SAMPLE, CHANNELS, I2S_RX_TRANSFORM_32_TO_16)) {
    Serial.println("Failed to configure I2S RX transformation for second set of mics!");
    return;
  }

  Serial.println("I2S bus initialized for both sets of mics.");

}
`

raprakashvi, 2024-06-30:
I did further testing to isolate the problem. In all the test cases, I keep all the mics wired and powered. The only aspect changing is the DIN pin initialization as that determines which microphone pair (left and right) is capturing data.

The configuration is for all 4 mics to share the SCK and WS while data pins are different.
Keeping the wiring unaltered and all 4 mics connected to power,
- when only 2 mic are initialized (1 pair) - we get proper recording with good sound quality. [stereo mode]
- Same happens when the second mic pair is initialized and we get similar quality of data.
- But when both the pairs are being initialized together, one pair has noise and the recording is 1-1.5 seconds shorter.

I am still trying to figure out why this is happening. I suspect this has to do with either :
- the current library does not give the option of choosing I2S port. So maybe the port assignment is causing the issue.
- or as the data is already read sequentially (one function is after the other), that may be causing some issue.

qniens, 2024-09-20:
I'm facing a similar issue. I'm trying to output 4 channels on both peripherals of the ESP32, but creating two separate I2S objects isn't working. I set the same BCLK, WS, and MCLK pins for both I2S objects, while assigning different pins for SDOUT. The sound output is fine, but it's not the correct data—the second I2S object is just outputting the same data as the first one (the I2S.write() from the first I2S object).

me-no-dev, 2024-09-20:
you can not share the same pins for two separate peripherals