espressif/arduino-esp32 #9947: DNSServer had some major changes, but the version was not incremented

TheBlackEagle198, 2024-06-28:
### Board

ESP32

### Device Description

ttgo-lora32-v1 with on-board tft display

### Hardware Configuration

not relevant

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi,
It seems like there's a commit which migrates DNSServer to a AsyncUDP implementation, however the version number of the package is not incremented: https://github.com/espressif/arduino-esp32/commit/d91271019ce0fb545ed9c2b6f51d27c86520f273.

This ended up being a problem, since the package manager from PlatformIO didn't index these changes (not sure how this works), and version 2.0.0 is still using WiFiUdp.

I came to this discovering while troubleshooting the older version of the library (I had this issue: E][WiFiUdp.cpp:185] endPacket(): could not send data: 12), and the newer version fixed it.

### Sketch

```cpp
#include <WiFiUdp.h>
#pragma once
#include <AsyncUDP.h>
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-06-28:
Apart from the fact that a version change should be done, the issue you encountered was a Platformio "hickup". Since installing a new framework deletes completely the old one including all libraries. So code has changed and the stored checksums from cache does not match anymore and the code will be compiled.

TheBlackEagle198, 2024-06-28:
I agree, I just thought I'd report the version increment as I'm not sure how versioning works on these modules.

me-no-dev, 2024-06-28:
they generally do not matter, because as @Jason2866 said, the core comes already packaged with them and the correct library would be compiled. You got a problem in your build environment.

closed 2024-07-16 by me-no-dev