espressif/arduino-esp32 #9949: ESP 32 DEV ERROR

pillewad, 2024-06-28:
### Board

ESP32 DEV MOUDLE

### Device Description

     void (*)(const uint8_t*, const uint8_t*, int) {aka void (*)(const unsigned char*, const unsigned char*, int)}

In file included from C:\Users\G AshiSh Kumar\Downloads\ST_Servo\ST Servo\ServoDriverST\CONNECT.h:2:

C:\Users\G AshiSh Kumar\Documents\ArduinoData\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/esp_wifi/include/esp_now.h:157:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'

  157 | esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);

      |                                    ~~~~~~~~~~~~~~~~~~^~

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\G AshiSh Kumar\Documents\ArduinoData\packages\esp32\hardware\esp32\3.0.2\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Hardware Configuration

NO IDEA

### Version

latest master (checkout manually)

### IDE Name

Ardiuno ide

### Operating System

windows 10

### Flash frequency

4 mb

### PSRAM enabled

no

### Upload speed

115200

### Description

Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Enabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

Specified folder/zip file does not contain a valid library

In file included from C:\Users\G AshiSh Kumar\Downloads\ST_Servo\ST Servo\ServoDriverST\ServoDriverST.ino:85:

C:\Users\G AshiSh Kumar\Downloads\ST_Servo\ST Servo\ServoDriverST\CONNECT.h: In function 'void espNowInit()':

CONNECT.h:328:28: error: invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]

[... 377 lines cut ...]

C:\Users\G AshiSh Kumar\Documents\ArduinoData\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/esp_wifi/include/esp_now.h:157:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'

  157 | esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);

      |                                    ~~~~~~~~~~~~~~~~~~^~

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\G AshiSh Kumar\Documents\ArduinoData\packages\esp32\hardware\esp32\3.0.2\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-28:
ESP-NOW API has changes in ESP-IDF and we have also rolled out our own [ESP-NOW library](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP_NOW). Adapt your code to the changes and it will work.

closed 2024-06-28 by me-no-dev

pillewad, 2024-06-28:
[image]
[image]
error not resolved please let me know procedure to solve this

pillewad, 2024-06-28:
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Enabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

In file included from C:\Users\G AshiSh Kumar\Downloads\ST_Servo\ST Servo\ServoDriverST\ServoDriverST.ino:85:

C:\Users\G AshiSh Kumar\Downloads\ST_Servo\ST Servo\ServoDriverST\CONNECT.h: In function 'void espNowInit()':

CONNECT.h:328:28: error: invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]

  328 |   esp_now_register_recv_cb(OnDataRecv);

      |                            ^~~~~~~~~~

      |                            |

      |                            void (*)(const uint8_t*, const uint8_t*, int) {aka void (*)(const unsigned char*, const unsigned char*, int)}

In file included from C:\Users\G AshiSh Kumar\Downloads\ST_Servo\ST Servo\ServoDriverST\CONNECT.h:2:

C:\Users\G AshiSh Kumar\Documents\ArduinoData\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/esp_wifi/include/esp_now.h:157:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'

  157 | esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);

      |                                    ~~~~~~~~~~~~~~~~~~^~

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\G AshiSh Kumar\Documents\ArduinoData\packages\esp32\hardware\esp32\3.0.2\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Jason2866, 2024-06-28:
Github issues is **NOT** the place to discuss personal coding problems.
You can ask for example here https://www.esp32.com/