espressif/arduino-esp32 #9951: Zigbee example does not compile for H2

petermeter69, 2024-06-28:
### Board

ESP32-H2 Dev module, SuperMini ESP32-H2

### Device Description

devkit

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino 2.3.2

### Operating System

Win10

### Flash frequency

64MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

On the same machine, I can **compile, upload and run** 3 of 4 built-in examples (Light switch, light bulb, temp sensor), while **unable to compile** the 4th example thermostat, with it being not modified by me at all. **Partition scheme** and **Zigbee mode** are correctly set (as the working examples show).

### Sketch

```cpp
Unmodified Zigbee_Thermostat.ino
```

### Debug Message

```plain
In file included 
from C:\Users\.../include/espressif__esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h:38,
from C:\Users\.../include/esp_zigbee_attribute.h:14,
from C:\Users\.../include/esp_zigbee_core.h:16,
from C:\Users\...\Zigbee_Thermostat.ino:33:
        C:\Users\...\Zigbee_Thermostat.ino: In function 'void esp_zb_task(void*)':
C:\Users\.../include/zcl/esp_zigbee_zcl_thermostat.h:180:63: error: narrowing conversion of '65535' from 'int' to 'int16_t' {aka 'short int'} [-Wnarrowing]
  180 | #define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE 0xffff
      |                                                               ^~~~~~
C:\Users\.../include/ha/esp_zigbee_ha_standard.h:329:38: note: in expansion of macro 'ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE'
  329 |                 .local_temperature = ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE,                         \
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\...\Zigbee_Thermostat.ino:479:44: note: in expansion of macro 'ESP_ZB_DEFAULT_THERMOSTAT_CONFIG'
  479 |   esp_zb_thermostat_cfg_t thermostat_cfg = ESP_ZB_DEFAULT_THERMOSTAT_CONFIG();
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

petermeter69, 2024-06-28:
By changing this line in "_include/zcl/esp_zigbee_zcl_thermostat.h_", it compiles:
From:
` #define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE 0xffff`
To:
` #define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE 0x7fff`

P-R-O-C-H-Y, 2024-06-28:
Hi @petermeter69, thank you for reporting the issue. Can you please open a PR with the proposed fix?

P-R-O-C-H-Y, 2024-06-28:
@petermeter69 This is something that needs to be fixed in [ESP-Zigbee-SDK](https://github.com/espressif/esp-zigbee-sdk). Please open an issue in the repository and also include the proposed fix. If you can also ping me or link this issue there.
As its only warning ([-Wnarrowing]) marked as error, you can disable "Compiler Warnings" in the Arduino IDE settings.

petermeter69, 2024-06-29:
@P-R-O-C-H-Y Thanks for you answer, reported there, see: [https://github.com/espressif/esp-zigbee-sdk/issues/379](url)

VojtechBartoska, 2024-10-21:
I am closing this ticket as there are no action points in Arduino repo.

closed 2024-10-21 by VojtechBartoska